syssi/esphome-jk-bms #520: JKPB2A16SXXP: RCV, RFV, Switch "Float" are not reported

holle75, 2024-07-07:
- RCV (requested charge voltage)
- RFV (requested float voltage)
- SWITCH Float (in settings)

are not reported

txubelaxu, 2024-07-07:
@txubelaxu branch is reporting them peroodically

holle75, 2024-07-07:
nice. good to know. Let´s see how stable the BT connection is for a few weeks. If not sufficient i will try to get a RS485 converter for my ESP32.

holle75, 2024-07-07:
Just do doublecheck  (i´m new to ESPHome)
this is the yaml i used. Are the missing values even in here?

`substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:33:10:33:4a
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.0.0

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: ssidxxx
  password: passowrdxxx

  manual_ip:
    static_ip: 192.168.1.60
    gateway: 192.168.1.1
    subnet: 255.255.255.0

web_server:
  port: 80

ota:
  platform: esphome
  password: passwordxxx
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: INFO

mqtt:
  broker: 192.168.1.55
  port: 13
  username: userxxx
  password: passwordxxx
  id: MQTT2_JK_BMS_Container_ESP32

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0 || x.get_name().rfind("JK_", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
[... 131 lines cut ...]
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"`

syssi, 2024-07-07:
This feature isn't implemented yet (here). This is the dedicated issue: https://github.com/syssi/esphome-jk-bms/issues/422

holle75, 2024-07-07:
it´s working with my setup and the (now) new yaml file.

syssi, 2024-07-08:
The feature is available since #485

closed 2024-07-08 by syssi