syssi/esphome-jk-bms #527: external_components_source: github://syssi/esphome-jk-bms@main

manuelmarin1967, 2024-07-07:
hola buenos dias , estoy intentando instalar y me da el siguiente error:
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:80:11:76:6A
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

este es el log de error que me aparece:

INFO ESPHome 2024.4.2
INFO Reading configuration /config/jk-bms.yaml...
INFO Cloning https://github.com/syssi/esphome-jk-bms.git@main
Failed config

external_components: [source /config/jk-bms.yaml:25]
  - 
    unable to access 'https://github.com/syssi/esphome-jk-bms.git/': Could not resolve host: github.com.
    source: github://syssi/esphome-jk-bms@main
    refresh: 0s

syssi, 2024-07-07:
This is the important part:

> unable to access 'https://github.com/syssi/esphome-jk-bms.git/': Could not resolve host: github.com.

It looks like there is no Internet to retrieve the component.

syssi, 2024-07-08:
Did you resolve the issue?

closed 2024-07-13 by syssi