syssi/esphome-jk-bms #528: JKPBXA16SXXP: charging/discharging temps are not reported

holle75, 2024-07-08:
- charging low temperature protection
- charging low temperature recovery
- charging high temperature protection
- ....

charging/discharging temps are not reported. This is a minor problem, but low temp charging protection might be of interest

holle75, 2024-07-13:
Oh, and if possible in any way, .... you can´t set SOC ... but you can change the capacity (even 1Ah is enough) of the battery and hereby the soc is set to an imaginary value of 74.

This makes no sense, but is extremely helpful when your SOC is calculated/measured wrongly and is higher than "what´s in the battery" . In my case the solar plant switches off charging when soc100 is reached. If soc is upcounted during charging and reaches 100 much too early, you will never charge your battery fully. It´s only getting worse.

when you can set this value, a charge-cycle to the real soc (according to BMS soc reset value) is going to happen.

holle75, 2024-07-13:
OOoooooh, you can do this already. Oh man, newbies shouldn´t be allowed to write ....

I did this in the dashboard of the ESP. How can i set this parameter via MQTT?

syssi, 2024-07-13:
> I did this in the dashboard of the ESP. How can i set this parameter via MQTT?

```
mosquitto_pub -t 'basement-btle-gateway/button/button_entity_id_example/command' -m 'PRESS'
mosquitto_pub -t 'basement-btle-gateway/switch/switch_entity_id_example/command' -m 'ON'    # Supported values: ON, OFF, TOGGLE
mosquitto_pub -t 'basement-btle-gateway/number/number_entity_id_example/command' -m 10
mosquitto_pub -t 'basement-btle-gateway/select/select_entity_id_example/command' -m 'option1'
```

holle75, 2024-07-13:
Mmh, dont get it. Thats my mqtt topic/reading (fhem mqtt2 server, not mosquitto):

jk_bms_2462abc9d910:jk-bms/number/jk-bms_total_battery_capacity/state:.* total_battery_capacity

and i want to set "315" 

would this be the syntax?:

jk_bms_2462abc9d910:jk-bms/number/jk-bms_total_battery_capacity/command 315

probably this is MQTT basics, but it´s freaking complicated.

syssi, 2024-07-13:
The topic is called `jk-bms/number/jk-bms_total_battery_capacity/command`. The message/payload is "315". `mosquitto_pub` is a simple MQTT tool/client to publish messages. The value of parameter `-t` is the topic. The value of `-m` defines the message payload.

holle75, 2024-07-13:
I got it to work. THANK YOU again. 
in fhem (for following readers)  it´s 

`attr MQTT2_JK_BMS setList capacityForLowerSOC:315 jk-bms/number/jk-bms_total_battery_capacity/command $EVTPART1`

syssi, 2024-07-26:
Please subscribe to issue #129. This is a duplicate.

closed 2024-07-26 by syssi