syssi/esphome-jk-bms #529: JK PB2A16S20P (BLE) - cell_request_charge_voltage issue

MacieGx, 2024-07-09:
Hi there 👋 
I really need the ability to control cell_request_charge_voltage from the HA and I noticed that a few days ago a [PR](https://github.com/syssi/esphome-jk-bms/pull/485) was merged that allows it.

Unfortunately, as soon as I add this to `number:` section
```
cell_request_charge_voltage:
      name: "${name} cell request charge voltage"
```
It stopped updating all editable values.

[image]

The set protocol is: JK02_32S. Am I doing something wrong?

syssi, 2024-07-09:
The settings frame isn't retrieved automatically/periodically. Please pree the `retrieve settings` button to retrieve the settings. They will be applied to the entities.

MacieGx, 2024-07-09:
You're right, I totally forgot about it. Everything works as it should. Thank you very much for your help. ❤️

closed 2024-07-09 by MacieGx