syssi/esphome-jk-bms #531: JKPBXA16SXXP: "switch" settings when restart

holle75, 2024-07-12:
Hi, minor one.

It seems when restarting, the switch settings (charging, charging float mode, display always on, ecc) are not read/set. I have to "click" the "retrieve settings" button to get them shown in the ESP dashboard.

Funny thing is that the overhead in my MQTT server is getting higher when those settings are not defined in the ESP.

question is: any way to read those settings during startup?

syssi, 2024-07-12:
This is a duplicate. Please subscribe this issue to track the progress here: #292

closed 2024-07-12 by syssi

syssi, 2024-07-13:
If you want to request the settings frame periodically, you could do something like this:

```
button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button
    retrieve_device_info:
      name: "${name} retrieve device info"

interval:
  - interval: 30min
    then:
      - button.press: retrieve_settings_button
```

Keep in mind: The BMS will beep on every request.

holle75, 2024-07-13:
Would love to read it once during startup. Just to have the actual settings in the broker.

syssi, 2024-07-13:
This should be already the case.

holle75, 2024-07-13:
Let me doublecheck tomorrow. But the last times, when restarted, the switches where not set.

holle75, 2024-07-14:
I doublechecked. It works as you described. I don´t know why (in the past) sometimes settings where missing. Sorry to bother.

syssi, 2024-07-14:
May be the settings frame can get missed because it is requested just once on every reconnect. This behavior could be improved if it's flaky.