syssi/esphome-jk-bms #532: Integration deye and jk BMS

sarzyk, 2024-07-14:
Hi is there any chance to integrate Deye RW-M6.1 battery with seccond diy battery JK BMS 2A16S20P?
I need to charge and discharge Deye battery only 80A but the diy battery (JK) over 150A. Will you board help mi with this?

syssi, 2024-07-14:
IMO the Mosfet of the BMS shouldn't be used to cut off the charging power periodically. It's just for emergency/protection which should happen only occasionally. I therefore doubt that this project will help you. It could just provide some measurements (SoC etc.).

sarzyk, 2024-07-14:
Ok. Thanks

closed 2024-07-14 by syssi

sarzyk, 2024-07-14:
But maybe it can minimize charging when is almost full (battery). Not permanently shutting down charging.