syssi/esphome-jk-bms #534: [Question] How to generate JK-B* V11 UART fake bms ?

Sleeper85, 2024-07-15:
Hi Sebastian,

I would like to generate different fake BMS to test my code, the alarm system and different scenarios.

How can I do it ?

[I'm currently using the example you provided.](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example-faker.yaml)

[This concerns my YamBMS (Yet another multi-BMS Merging Solution) project in the dev branch currently.](https://github.com/Sleeper85/esphome-jk-bms-can/tree/development)

Thanks

syssi, 2024-07-15:
You can change the payload of the faker as you like. What do you miss here?

Sleeper85, 2024-07-15:
> You can change the payload of the faker as you like. What do you miss here?

I don't know how to generate the desired payload.
For example for a 16S LFP battery which would currently have an OTP alarm.
What does each hex value correspond to?
I need to test different scenarios.

Thanks

syssi, 2024-07-15:
This is captured data from real devices. I'm not aware of an easy way to generate synthetic data.

syssi, 2024-07-15:
Perfect! Let's close this issue. :-)

closed 2024-07-15 by syssi