syssi/esphome-jk-bms #535: Inverter BMS battery capacity not reporting correctly

iamvlanone, 2024-07-16:
I connected a 250Ah battery and all data seemed correct between JK BMS App and ESP. Somehow this changed and the ESP sensor is now reporting 161Ah while the BMS still reports 250Ah in the JK BMS App. 

23:05:22][D][sensor:094]: 'inv-batt-1 capacity remaining': Sending state 161.24701 Ah with 3 decimals of accuracy
[23:05:22][D][sensor:094]: 'inv-batt-1 total battery capacity setting': Sending state 161.24701 Ah with 0 decimals of accuracy

Any ideas or guidance on where to look?

syssi, 2024-07-16:
If you increase the log level the raw data of the BMS is printed to the log. This raw data would be helpful to get an idea what's going on here. Do use use a BLE or UART connection? Do you know how to increase the log level?

iamvlanone, 2024-07-16:
Sorry, I am using BLE. I saw the section on increasing the log level. Should I increase all the levels or is there a specific one in order to minimize the amount of output?

syssi, 2024-07-16:
This is the required logger configuration for a BLE setup:

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: VERY_VERBOSE
    jk_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

I'm interested in all `jk_bms_ble` and `sensor` log messages.

iamvlanone, 2024-07-17:
Please see the logs attached and the sensor states in Home Assistant. Please let me know if you need more info?

 
[logs_inv-batt-1 - Logging Enabled v2.txt](https://github.com/user-attachments/files/16262098/logs_inv-batt-1.-.Logging.Enabled.v2.txt)

[image]

iamvlanone, 2024-07-17:
I did some more troubleshooting.

I did a bunch of logging with and without the verbose logging and the results were inconsistent. I have not kept a detailed log of what happened at what time, although I am of a technical background and know better.

I decided to flash another ESP with esp32-ble-v14-example.yaml again, I only added the api encryption key. I saw the same result where it sends the wrong battery capacity. I thought that rules out my code implementation on the original ESP.

I then proceeded to connect both the old and new ESP to a development HA instance and received the same result.

I opened the JK BMS App on my phone again and looked at the status and setting screens and both showed 250Ah. I decided to change the capacity setting in the BMS from 250Ah to 240Ah and then back to 250Ah and it seems to have resolved the issue as I am now receiving the correct values.

Two days ago I had a battery undervolt protection event and had to switch the the BMS into emergency mode. I did not power down the BMS. I am not sure if that might have caused the issue? I will monitor it and feed back.

I assume this could be closed as a non-issue?

syssi, 2024-07-17:
```
# Cell info frame
# 'inv-batt-1 total battery capacity setting': Sending state 161.24701 Ah
# 'inv-batt-1 capacity remaining': Sending state 144.83101 Ah with 3 decimals of accuracy

0x55, 0xAA, 0xEB, 0x90, 0x02, 0xAF, 0xD3, 0x0C, 0xD1, 0x0C, 0xD6, 0x0C, 0xD8, 0x0C, 0xDA, 0x0C, 0xDB, 0x0C, 0xDB, 0x0C, 0xDB, 0x0C, 0xD7, 0x0C, 0xD0, 0x0C, 0xD7, 0x0C, 0xD8, 0x0C, 0xD3, 0x0C, 0xD5, 0x0C, 0xDD, 0x0C, 0xDD, 0x0C, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0xD7, 0x0C, 0x0F, 0x00, 0x0B, 0x06, 0x54, 0x00, 0x52, 0x00, 0x56, 0x00, 0x53, 0x00, 0x55, 0x00, 0x58, 0x00, 0x5D, 0x00, 0x64, 0x00, 0x65, 0x00, 0x54, 0x00, 0x56, 0x00, 0x53, 0x00, 0x55, 0x00, 0x5A, 0x00, 0x5C, 0x00, 0x6C, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xA2, 0x00, 0x00, 0x00, 0x00, 0x00, 0x72, 0xCD, 0x00, 0x00, 0x4D, 0xF2, 0x09, 0x00, 0x96, 0xCF, 0xFF, 0xFF, 0xA0, 0x00, 0xA2, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x5A, 0xBF, 0x35, 0x02, 0x00, 0xDF, 0x75, 0x02, 0x00, 0x02, 0x00, 0x00, 0x00, 0x8A, 0x38, 0x05, 0x00, 0x40, 0x00, 0x00, 0x00, 0xC3, 0x47, 0x05, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0x00, 0x01, 0x00, 0x00, 0x00, 0xAB, 0x03, 0x01, 0x00, 0x21, 0x00, 0x3B, 0x9A, 0x3F, 0x40, 0x00, 0x00, 0x00, 0x00, 0x8B, 0x14, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x00, 0x06, 0x00, 0x00, 0x97, 0x40, 0x32, 0x00, 0x00, 0x00, 0x00, 0x00, 0xA2, 0x00, 0x9D, 0x00, 0x9F, 0x00, 0xAB, 0x03, 0x56, 0xB1, 0x8B, 0x08, 0xD9, 0x00, 0x00, 0x00, 0x80, 0x51, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFE, 0xFF, 0x7F, 0xDC, 0x2F, 0x01, 0x01, 0xB0, 0x07, 0x00, 0x00, 0x00, 0x6B
```

syssi, 2024-07-17:
I will think about it later the week. I assume the value of the `cell info frame` should be called `total battery capacity setting`. I guess the capacity setting you are looking for is part of the `settings frame`.

iamvlanone, 2024-07-17:
I am not sure I understand the difference between the frames, but it seems the BMS might have been sending the incorrect battery capacity and remaining capacity values although the App was seeing the correct values for some reason. As stated above as soon as I made a change to the capacity setting in the App the values sent by the ESP was corrected. 

The fact that I received the same incorrect values using different ESPs and on a different HA instances indicated to me that the source of the information might be at fault. I noticed that after I made the change on the BMS, the cycled Ah capacity and the cycle count was reset to 0. This makes me worried as if someone makes a change to the BMS settings 5 years into using the BMS you loose all the history?

b1ggi, 2025-03-18:
Just had the same issue like you. app reported 280ah (correct), yambms (uses syssis repo here) reported 275ah. i entered 280ah in the app again, now the cycle count (67) and cycle capacity (18947.512 Ah) is gone. its sad that my history now is erased. is there a way to manually set cycle count and cycle capacity?

iamvlanone, 2025-03-19:
I have had this issue a second time and still have not touched the BMS config for fear of losing my history. It seems to be a BMS software issue and teaching out to JK yielded no response.

My issue happened when the battery was drained to 0% and then rapidly charged. When one of the cell voltages reached the 100% voltage threshold the BMS set the SOC to 100% and then reports the amount of Ah charged at that moment as the capacity of the battery although the configured capacity is still the original capacity. 

The template sensor in HA is a good idea and I will be sure to note the cycled capacity when I make changes to the BMS.

iamvlanone, 2026-01-06:
> edit: as a workaround for me i will add the last known values to a template sensor in homeassistant to keep track of the real usage

I did this and it worked. after my last issue on FW v15.41 when I changed the capacity from 250Ah to the correct 177Ah the cycle count and cycled capacity did not reset to 0. It seems they have sorted that in their latest FW.