syssi/esphome-jk-bms #537: Try working with JK_BD6A20S6P

parumsancto, 2024-07-19:
My BMS: [image]
Hardware: WeMos D1 mini V4.0.0.
My config: [jk-bms-fridge-orig.txt](https://github.com/user-attachments/files/16314013/jk-bms-fridge-orig.txt)

Controls is working
All sensors is "Unknown".

Any help please.

syssi, 2024-07-19:
Could you increase the log level and provide a log?

https://github.com/syssi/esphome-jk-bms?tab=readme-ov-file#debugging

parumsancto, 2024-07-19:
Thank you for reply. Here is the new config and logs.
[jk-bms-fridge-orig.txt](https://github.com/user-attachments/files/16314646/jk-bms-fridge-orig.txt)
[logs_jk-bms-fridge-orig_logs.txt](https://github.com/user-attachments/files/16314647/logs_jk-bms-fridge-orig_logs.txt)

syssi, 2024-07-19:
```
[19:33:10][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[19:33:15][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[19:33:20][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[19:33:25][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[19:33:30][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[19:33:35][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
```

This is outgoing traffic only. Your BMS doesn't respond. Incoming traffic is prefixed like this: `<<<`

parumsancto, 2024-07-19:
Hmm. But if I turn off Discharging, it actually turns off. Voltage at the BMS output drops to zero. And if I turn it on, the voltage back to battery voltage.  This is in the logs:
```
[20:22:40][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[20:22:43][D][switch:016]: 'jk-bms-fridge-orig discharging' Turning OFF.
[20:22:43][D][switch:055]: 'jk-bms-fridge-orig discharging': Sending state OFF
[20:22:43][W][component:237]: Component api took a long time for an operation (161 ms).
[20:22:43][W][component:238]: Components should block for at most 30 ms.
[20:22:43][D][uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:05:02:00:00:00:00:00:00:00:68:00:00:01:28:4E:57:00:14:00:00:00:00:02:02:00:AC:00:00:00:00:00:68:00:00:01:D1
[20:22:45][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[20:22:50][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[20:22:55][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[20:22:55][D][switch:012]: 'jk-bms-fridge-orig discharging' Turning ON.
[20:22:55][D][switch:055]: 'jk-bms-fridge-orig discharging': Sending state ON
[20:22:55][W][component:237]: Component api took a long time for an operation (162 ms).
[20:22:55][W][component:238]: Components should block for at most 30 ms.
[20:22:55][D][uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:05:02:00:00:00:00:00:00:00:68:00:00:01:28:4E:57:00:14:00:00:00:00:02:02:00:AC:01:00:00:00:00:68:00:00:01:D2
[20:23:00][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[20:23:05][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
```

syssi, 2024-07-19:
In this case the BMS is probably able to receive the instructions but the ESP is unable to receive the response. Could you try to use an ESP32?

syssi, 2024-07-20:
The GPIO1 and GPIO3 of your ESP8266 aren't free floating because a USB-to-TTL chip is attached here. If you use a ESP32 the hardware UART can be routed to any GPIO because of the GPIO matrix.

parumsancto, 2024-07-20:
Unfortunately, I have not ESP32. But I connect a logic analyzer to TX, RX and see a request from ESP exactly like in logs (4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28) but in RX wire no signal. So BMS didn't respond on request. Maybe the request is incorrect for my BMS?

syssi, 2024-07-20:
Did you buy the CAN bus enabled version of the BMS? Do you use the Android app and could you check some settings here?

parumsancto, 2024-07-20:
I have the RS485/BT version. Yes, I am using the Android app, what do I need to check?

syssi, 2024-07-20:
Could you provide a screenshot of the settings view and the controls view?

syssi, 2024-07-21:
At the controls view I'm looking for the state of the "CAN vs. RS485" switch. At the settings view there could be ab item called "UART1 protocol". The `GPS module protocol` should be selected here. These settings are depending on your hardware model and version. If you have no luck then aren't available/ aren't the root cause of the issue.

parumsancto, 2024-07-21:
> Could you provide a screenshot of the settings view and the controls view?

Yes, of course. Here they are
[image]
[image]
[image]
[image]
[image]

parumsancto, 2024-07-21:
> At the controls view I'm looking for the state of the "CAN vs. RS485" switch. At the settings view there could be ab item called "UART1 protocol". The `GPS module protocol` should be selected here. [...]

I see "UART1 protocol". Only one option is available and selected.
[image]
[image]

syssi, 2024-07-21:
The selected protocol of UART1 looks good. The `CAN vs. RS485` doesn't exist. The option `GPS heartbeat` is new to me. You could enable the setting just for testing.

As next step a dissembly of the BMS is required to measure a possible response of the BMS before the probably broken optical isolator.

parumsancto, 2024-07-21:
Thank you for your answer. I will try both solutions.

parumsancto, 2024-07-23:
About **GPS heartbeat** from BMS datasheet: 
"When the GPS heartbeat detection function is turned on, the protection board will detect the connection status of GPS. When the connection between GPS and the protection board is less than 24 hours, the protection board will turn off the charge and discharge switch and generate an alarm of "GPS disconnection" (this function is usually used for GPS disassembly detection)."
So this setting do nothing for me.
Next step **disassembly BMS**. Optical isolator is working. And I fount that RX signal to ESP present on BMS controller. But it suppressed to an amplitude of less than a volt on BMS connector. However, if Idisconnect the RX line from the ESP, the response signal on the BMS connector is normal (3.3 volts amplitude). So ESP pulls the RX line to 3V3.  This is not in the scheme, so it is a program. So I tried to use uart_2 (GPIO13, GPIO15) and ESP started receiving responses.

**So here is the configuration that works for me** 
[jk-bms.txt](https://github.com/user-attachments/files/16344315/jk-bms.txt)
**Logs** [logs_jk-bms.txt](https://github.com/user-attachments/files/16344319/logs_jk-bms.txt)
 @syssi  **thank you very much for the help.**

closed 2024-07-23 by parumsancto

syssi, 2024-07-23:
@parumsancto Okay, in this case the USB-to-TTL chip attached to GPIO1 & GPIO3 caused the issue. Awesome engineering job!

parumsancto, 2024-07-24:
By the way, you can add JK-BD6A20S6P to supported devices.