syssi/esphome-jk-bms #545: Read out two Neey balancers?

mathschut, 2024-07-26:
Hi, can someone tell me what the config should look like if I want to monitor two Neey Balancers?

syssi, 2024-07-26:
I've added an example to the project: https://github.com/syssi/esphome-jk-bms/blob/1896c27fdbc8ccd1f80e84767dedc431d52ad425/esp32-heltec-balancer-ble-example-multiple-devices.yaml

closed 2024-07-26 by syssi

mathschut, 2024-07-26:
Hi, thanks for the file. I have now set up my Mac addresses and I always get data. Unfortunately it always hangs up. Do you have any idea what could be causing this?

Is it possible that the ESP is overloaded with two devices?

syssi, 2024-07-26:
Let's start with a super simple and lightweight setup just for comparison:

```
substitutions:
  name: heltec-balancer
  balancer0: "${name} device0"
  balancer1: "${name} device1"
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  balancer0_mac_address: C8:47:8C:E1:E2:FF
  balancer1_mac_address: C8:47:8C:E1:E2:EE

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - switch.turn_off: ble_client_switch1
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${balancer0_mac_address}
    id: client0
  - mac_address: ${balancer1_mac_address}
    id: client1

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: balancer0
  - ble_client_id: client1
    throttle: 5s
    id: balancer1

sensor:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer0
    total_voltage:
      name: "${balancer0} total voltage"
    temperature_sensor_1:
      name: "${balancer0} temperature sensor 1"
    temperature_sensor_2:
      name: "${balancer0} temperature sensor 2"
    total_runtime:
      name: "${balancer0} total runtime"
    balancing_current:
      name: "${balancer0} balancing current"

  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer1
    total_voltage:
      name: "${balancer1} total voltage"
    temperature_sensor_1:
      name: "${balancer1} temperature sensor 1"
    temperature_sensor_2:
      name: "${balancer1} temperature sensor 2"
    total_runtime:
      name: "${balancer1} total runtime"
    balancing_current:
      name: "${balancer1} balancing current"

switch:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer0
    balancer:
      name: "${balancer0} balancer"

  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer1
    balancer:
      name: "${balancer1} balancer"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${balancer0} enable bluetooth connection"

  - platform: ble_client
    ble_client_id: client1
    id: ble_client_switch1
    name: "${balancer1} enable bluetooth connection"
```

Another important change to free some resources is to decrease the log level:

```
logger:
  level: INFO
```

mathschut, 2024-07-26:
Which esp should I use for 2 devices?

syssi, 2024-07-26:
I have no recommendation.

commit 8aa46152 (2024-07-26): Add multiple heltec balancer example (Closes #545) (#546)