syssi/esphome-jk-bms #548: JK-PBx - Where are the "RCV Time" and "RFV Time" params/values?

dmode9000, 2024-07-27:
Could you please let me know if the "RCV Time" and "RFV Time" parameters are available? I haven't found them in the HA interface yet and haven't found any information on how to add them to the interface.

syssi, 2024-07-29:
How do you talk to your BMS? If you are using BLE the feature was added here: https://github.com/syssi/esphome-jk-bms/issues/422

You can just add these additional `number` entities to your configuration YAML: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml#L132-L135

syssi, 2024-07-29:
Oh. We are talking about the runtimes. This setting isn't implemented yet.

```
# RCV time
aa:55:90:eb: b3 :01: 14 :71:55:0b:ba:97:e6:68:67:fe:71:5b:50:33 # 20 * 0.1f = 2.0h
aa:55:90:eb: b3 :01: 0a :76:55:c2:22:63:4f:02:ce:a8:d8:a7:0a:9a # 10 * 0.1f = 1.0h

# RFV time
aa:55:90:eb: b4 :01: 5a :45:6a:43:1a:4d:85:5f:3e:db:6a:f8:73:b4 # 90 * 0.1f = 9.0h
aa:55:90:eb: b4 :01: 50 :d1:61:5a:d0:53:35:21:c9:8a:63:ec:6d:93 # 80 * 0.1f = 8.0h
```

This issue is the follow up of #422 because two feature was missing.

syssi, 2024-07-29:
I've started to implement the feature but some testing is required to complete the feature. Are you able to talk to your BMS already? Could you offer some testing?

closed 2024-07-29 by syssi

dmode9000, 2024-07-30:
Yes, I can test it. Do I understand correctly that after recompiling and flashing in Homeassisant, these parameters should appear as available?

syssi, 2024-07-30:
You have to extend your YAML configuration be these 4 lines at the `number` section:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml#L136-L139

```
number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    # ...
    cell_request_charge_voltage_time:
      name: "${name} cell request charge voltage time"
    cell_request_float_voltage_time:
      name: "${name} cell request float voltage time"
``` 

And recompile + flash your ESP. The two number entities are populated as soon a `device info`  frame gets received. This frame is retrieved once on an established BLE connection and can be trieved manually using the `retrieve device info` button.

dmode9000, 2024-07-30:
Recompiled and added 4 lines in the number section. 
"RCV Time" and "RFV Time" parameters are visible in the interface but are not retrieved from the BMS, even after manually clicking on "Retrieve Device Info."

New issues::

- RCV and RFV parameters (NOT time but Voltage) started displaying in HA as 1 and 0.4 volts.

[image]

- The heating option on BMS is constantly turning off and on.

[image]

syssi, 2024-07-31:
My fault. I've pushed a fix for the `RCV`, `RCV time`, `RFV` and `RFV time` entities. Please recompile & flash again. Could you create a new issue for the `heating` sensor?

dmode9000, 2024-07-31:
Recompiled, RCV, RCV time, RFV and RFV time entities now Ok.

New heating sensor issue created.