syssi/esphome-jk-bms #552: JK-PB2A16S20P: constant heating on/off switching

dmode9000, 2024-07-31:
The heating option on BMS is constantly turning off and on.

[image]

syssi, 2024-07-31:
Do you have a heating element attached? Are you able to test the feature as soon it's fixed?

syssi, 2024-07-31:
I've pushed a fix into the feature branch `heating-v14`. Please change this line at your YAML configuration to use this branch:

```
substitutions:
  # ...
  external_components_source: github://syssi/esphome-jk-bms@heating-v14
```

Recompile & flash your ESP. Does the binary sensor indicate the truth now?

dmode9000, 2024-07-31:
> Do you have a heating element attached?

No, never was.

> Recompile & flash your ESP.

No constant heating switching now.

> Does the binary sensor indicate the truth now?

Looks like fixed.
[image]

syssi, 2024-07-31:
What happens if you use the `heating switch` to turn heating on?

signaleye, 2024-08-01:
Sorry @dmode9000 

I have same BMS and want to connect to it but no luck.. do. you have the complete yaml code to share?

syssi, 2024-08-01:
@signaleye Could you create a new issue and provide an ESPHome log?

closed 2024-08-01 by syssi

dmode9000, 2024-08-01:
> What happens if you use the `heating switch` to turn heating on?

When the "heating switch" is pressed, it changes its status to pressed and remains in that state. The logs show that the heating button was pressed.

> I have same BMS and want to connect to it but no luck.. do. you have the complete yaml code to share?

https://pastebin.com/gcDZYznW#google_vignette

commit 07d1a4b2 (2024-08-01): Fix binary sensor `heating` (Closes: #552) (#553)