syssi/esphome-jk-bms #557: currently flashing an esp8266 module no longer works.

duymelody9x, 2024-08-03:
1 month ago i flashed an esp8266 and it worked perfectly, but now flashing a new device it doesn't work anymore. Do you have any updates during this time?
[image]

syssi, 2024-08-03:
The implementation haven't changed. Please provide some logs. Do you use another ESP? Did you change the BMS? Please try to identify the changed parts of your setup.

duymelody9x, 2024-08-05:
i sold the old battery pack i bought a new battery pack and a new 8266 module. But they are the same as the last time i used them. i bought another new 8266 module but it still doesn't work. i'm using batmon addons now i bought an esp32 that connects via bluetooth will they work stably?

syssi, 2024-08-05:
To resolve the issue it's important to provide more details.

1. Do you have a new BMS too?
2. Are you sure the new BMS has a working GPS port?
3. Is RS485 enabled and the correct protocol selected?
4. Does Batman use a wired connection or BLE?
5. Which GPIOs did you use for RXD and TXD? Please provide your YAML configuration
6. Enable the `debug` output of the `uart` component and provide some logs

syssi, 2024-08-05:
Some ideas:

1. There are D1 mini boards out there where GPIO1/GPIO3 cannot talk to the BMS because the UART-to-TTL chip attached to the same pins makes strange things. Please try GPIO4, GPIO5 instead.
2. May be the underlying Arduino implementation has changed. I'm using ESP32 boards only without trouble. Please try an ESP32 just for testing.
3. If the GPS port remains silent you have to use BLE unfortunately.

duymelody9x, 2024-08-05:
Which ESP32 boards do you usually use?

syssi, 2024-08-05:
They are called `ESP32 D1 Mini` and are looking very similar to the Wemos D1 Mini board. Be careful to not pick a ESP8266 based board. ;-)

closed 2024-08-09 by syssi