syssi/esphome-jk-bms #559: UART Discharging switch not working ( B2A20S20P HW: V11.XW - SW: V11.285H )

Sleeper85, 2024-08-03:
Hi @syssi 

First of all, thank you very much for all the development you have done to integrate the different brands of BMS into HA.

I'm testing the UART switches and I realize that the `discharging` switch does not work from HA.
I can modify it in the JK application and its status will be updated in HA but I cannot modify it from HA.

The `charging` switch works fine.

Is this a known issue with the `V11.285H` version?

```YAML
switch:
  - platform: jk_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
```

Many thanks

syssi, 2024-08-03:
I'm not aware of any issues here. Which BMS model do you use exactly?

Sleeper85, 2024-08-03:
JK-B2A20S20P 200A Heat + CAN.
HW: V11.XW
SW: V11.285H

syssi, 2024-08-05:
I've tested the discharging switch on a 11.XW_S11.27___ device without trouble.

Sleeper85, 2024-08-05:
Ok I will do a new test only with your code.

For the moment I test it from my git and a [minimal version of the JK-B UART YAML](https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/packages/bms/bms_JK-B_UART_minimal.yaml).

Did I remove sensors necessary for proper operation ?

What is strange is that the charge switch works.

syssi, 2024-08-05:
There are no dependencies. A tiny YAML with two switch entities only should work already.

Sleeper85, 2024-08-06:
It seems that the problem is only there when I do a dual UART + BLE supervision of the same BMS.
I did this as part of my multi-bms to CAN bus testing (YamBMS).

I now work only with UART and I have no more problem.

closed 2024-08-06 by Sleeper85

syssi, 2024-08-06:
I don't think this is the root cause. I did the same yesterday using the Android app for verification. An active BLE connection indicates the changes switch state immediately.

Sleeper85, 2024-08-06:
You are right. I just did some other tests. Disabling the `discharge` switch while the battery is charging is not always immediate. This may be related to the fact that with the mosfet blocking the discharge, the charge current must exceed a certain threshold for the charge to be effective.

syssi, 2024-08-06:
Be careful. :-) The Mosfets can die on high currents.