syssi/esphome-jk-bms #560: 2 BMS CONNECTION PROBLEM

jouking, 2024-08-04:
I have been trying for three days to get an esp32 and a wemo d1 connected via UART-TTL to the bms of my second power bank to work but I am not able to, let's see if someone can help me because I am desperate. 

**I have 2 black jkbms in 2 power banks.** 

### The first one is connected through the rs485 port via UART-TTL to an ESP WT32-ETH01 with the configuration that I indicate below and it works perfectly. 

Now in my second power bank I bought another identical WT32-ETH01 and I have connected it the same as the first and I get this error:

[17:29:37][D][text_sensor:064]: 'w32-banco2 Errors': Sending state 'Offline'
[17:29:37][D][sensor:094]: 'w32-banco2 min cell voltage': Sending state nan V with 3 decimals of accuracy
[17:29:37][D][sensor:094]: 'w32-banco2 max cell voltage': Sending state nan V with 3 decimals of accuracy
[17:29:37][D][sensor:094]: 'w32-banco2 min voltage cell': Sending state nan  with 0 decimals of accuracy
[17:29:37][D][sensor:094]: 'w32-banco2 max voltage cell': Sending state nan  with 0 decimals of accuracy 

When I put the WT32-ETH01 from bank 1 in bank 2 it works correctly, but I cannot get the new ESP WT32-ETH01 to work. 

I have tried connecting bank 2 with a Wemo D1 in case the new WT32-ETH01 did not work, but it does the same. 

What am I doing wrong?

CODE:

substitutions:
  name: w32-banco2

esphome:
  name: ${name}
  friendly_name: w32_banco2
  platform: ESP32
  board: esp-wrover-kit

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "1u757XEy/uOj/WcvTB7RpaSjGJx+PfiE7Mqspw="

ota:
  - platform: esphome
    password: "6299d3d3d0ec7414ff477302d"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Optional manual IP
  manual_ip:
    static_ip: 192.168.1.196
    gateway: 192.168.1.254
    subnet: 255.255.255.0

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: GPIO3
    rx_pin: GPIO1

jk_modbus:
  - id: modbus0
    uart_id: uart_0
#  rx_timeout: 50ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
#  update_interval: 5s

sensor:
  - platform: jk_bms
    jk_bms_id: bms0
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    power_tube_temperature:
      name: "${name} Power tube temperature"
    temperature_sensor_1:
      name: "${name} Temperature sensor 1"
    temperature_sensor_2:
      name: "${name} Temperature sensor 2"
    total_voltage:
      name: "${name} Total voltage"
    current:
      name: "${name} Current"
    capacity_remaining:
      name: "${name} Capacity remaining"
    capacity_remaining_derived:
      name: "${name} capacity remaining derived"
    temperature_sensors:
      name: "${name} Temperature sensors"
    charging_cycles:
      name: "${name} Charging cycles"
    total_charging_cycle_capacity:
      name: "${name} Total charging cycle capacity"
    battery_strings:
      name: "${name} Battery strings"
    errors_bitmask:
      name: "${name} Errors bitmask"
    operation_mode_bitmask:
      name: "${name} Operation mode bitmask"
    total_voltage_overvoltage_protection:
      name: "${name} Total voltage overvoltage protection"
    total_voltage_undervoltage_protection:
      name: "${name} Total voltage undervoltage protection"
    cell_voltage_overvoltage_protection:
      name: "${name} Cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} Cell voltage overvoltage recovery"
    cell_voltage_overvoltage_delay:
      name: "${name} Cell voltage overvoltage delay"
    cell_voltage_undervoltage_protection:
      name: "${name} Cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} Cell voltage undervoltage recovery"
    cell_voltage_undervoltage_delay:
      name: "${name} Cell voltage undervoltage delay"
    cell_pressure_difference_protection:
      name: "${name} Cell pressure difference protection"
    discharging_overcurrent_protection:
      name: "${name} Discharging overcurrent protection"
    discharging_overcurrent_delay:
      name: "${name} Discharging overcurrent delay"
    charging_overcurrent_protection:
      name: "${name} Charging overcurrent protection"
    charging_overcurrent_delay:
      name: "${name} Charging overcurrent delay"
    balance_starting_voltage:
      name: "${name} Balance starting voltage"
    balance_opening_pressure_difference:
      name: "${name} Balance opening pressure difference"
    power_tube_temperature_protection:
      name: "${name} Power tube temperature protection"
    power_tube_temperature_recovery:
      name: "${name} Power tube temperature recovery"
    temperature_sensor_temperature_protection:
      name: "${name} Temperature sensor temperature protection"
    temperature_sensor_temperature_recovery:
      name: "${name} Temperature sensor temperature recovery"
    temperature_sensor_temperature_difference_protection:
      name: "${name} Temperature sensor temperature difference protection"
    charging_high_temperature_protection:
      name: "${name} Charging high temperature protection"
    discharging_high_temperature_protection:
      name: "${name} Discharging high temperature protection"
    charging_low_temperature_protection:
      name: "${name} Charging low temperature protection"
    charging_low_temperature_recovery:
      name: "${name} Charging low temperature recovery"
    discharging_low_temperature_protection:
      name: "${name} Discharging low temperature protection"
    discharging_low_temperature_recovery:
      name: "${name} Discharging low temperature recovery"
    current_calibration:
      name: "${name} Current calibration"
    device_address:
      name: "${name} Device address"
    sleep_wait_time:
      name: "${name} Sleep wait time"
    alarm_low_volume:
      name: "${name} Alarm low volume"
    manufacturing_date:
      name: "${name} Manufacturing date"
    total_runtime:
      name: "${name} Total runtime"
    start_current_calibration:
      name: "${name} start current calibration"
    actual_battery_capacity:
      name: "${name} Actual battery capacity"
    protocol_version:
      name: "${name} Protocol version"
      
text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} Errors"
    operation_mode:
      name: "${name} Operation mode"
    battery_type:
      name: "${name} Battery type"
    password:
      name: "${name} Password"
    device_type:
      name: "${name} Device type"
    software_version:
      name: "${name} Software version"
    manufacturer:
      name: "${name} Manufacturer"

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "${name} balancing"
    balancing_switch:
      name: "${name} balancing switch"
    charging:
      name: "${name} charging"
    charging_switch:
      name: "${name} charging switch"
    discharging:
      name: "${name} discharging"
    discharging_switch:
      name: "${name} discharging switch"
    dedicated_charger_switch:
      name: "${name} dedicated charger switch"

[image]
[image]

syssi, 2024-08-04:
Please don't use the TXD and RXD pin. The USB-to-TTL chip is attached here and the BMS doesn't like this. Please use any other GPIO instead.

syssi, 2024-08-04:
Oh. Is there a USB-to-TTL chip on the WT32-ETH01? Nevertheless. Please try other GPIOs. :-)

jouking, 2024-08-04:
> Please don't use the TXD and RXD pin. The USB-to-TTL chip is attached here and the BMS doesn't like this. Please use any other GPIO instead.

I tried with GPIO14 and GPIO4 and the problem continues.

jouking, 2024-08-04:
Anyway the wt32 of bank 1 works perfectly on the GPIO3 and GPIO1

syssi, 2024-08-04:
Are you sure you haven't swapped the RX/TX lines accidentally?

Another idea is to use the ESP-IDF framework just for testing:

```
esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
```

jouking, 2024-08-04:
> Are you sure you haven't swapped the RX/TX lines accidentally?
> 
> Another idea is to use the ESP-IDF framework just for testing:
> 
> ```
> esp32:
>   board: wemos_d1_mini32
>   framework:
>     type: esp-idf
> ```

It doesn't work either. Still giving the same error.

josef232, 2024-08-04:
Hallo, i use this Pin´s on my ESP32 for 1xCAN and 2xModbus (1x JK-BMS, 1x Goodwe HV Hybrid  Inverter).

You are already using some of these pins, but it might work with the others.

# ESP32 CAN/Serail port pins:
# GPIO pins  CAN bus transceiver(TJA1050) is connected to the ESP, note! TX->TX and RX->RX. 
  can_tx_pin: GPIO23
  can_rx_pin: GPIO22
# GPIO pins  Modbus JK-BMS RS485(TTL) is connected to the ESP TX->RX and RX->TX. 
  tx_pin: GPIO17
  rx_pin: GPIO16
# GPIO pins  Modbus Goodwe RS485(TTL) is connected to the ESP TX->RX and RX->TX.
  tx_pin2: GPIO1
  rx_pin2: GPIO3

esphome:
  name: testbms
  friendly_name: testbms

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  baud_rate: 0

jouking, 2024-08-05:
This is starting to work.

First failure, I had the cables connected incorrectly.

Then I had to change the uart port, modbus and bms number to 1.

Now it starts showing data but cuts data from time to time with this error: 
[W][uart.arduino_esp32:192]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.

After giving this notice it continues to work correctly, but after a while it gives this other error and cuts off the communication: [W][jk_modbus:091]: CRC check failed! 0x54DA != 0x54EE

Now I have the GPIO1 and GPIO3. In the afternoon I will change the GPIOs for the 4 and 14 in case this solves the problem.

syssi, 2024-08-05:
If you are using GPIO1/GPIO3 for TX/RX you have to disable the serial logging by setting the baud rate to `0`: 

```
logger:
  baud_rate: 0
```

jouking, 2024-08-05:
> If you are using GPIO1/GPIO3 for TX/RX you have to disable the serial logging by setting the baud rate to `0`:
> 
> ```
> logger:
>   baud_rate: 0
> ```

ok, that has solved the first warning, but it still gives this error:

[W][jk_modbus:091]: CRC check failed! 0x54CC != 0x54EC

syssi, 2024-08-05:
Which framework (Arduino vs. ESP-IDF) do you use at the moment? How often do you see an CRC error?

jouking, 2024-08-05:
> Which framework (Arduino vs. ESP-IDF) do you use at the moment? How often do you see an CRC error?

if I write
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

gives this error: [W][jk_modbus:063]: Invalid header: 0x38 0x2E

If I put this other:

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

gives this error: [W][jk_modbus:091]: CRC check failed! 0x54EB != 0x54EC

This is how I have it in the other esp32 of power bank 1 that works perfectly:

esphome:
  name: wt32
  platform: ESP32
  board: esp-wrover-kit

syssi, 2024-08-05:
Please don't rush and take your time. Both warnings are about transmission errors. Transmission errors can happen from time to time. This is okay and doesn't harm (you just miss a frame of data from time to time). If this happens once per second this isn't fine anymore and must be resolved.

How often do you see CRC errors? Please check your wires / you need proper connections / double check GND. Avoid long cables etc.

jouking, 2024-08-05:
> Please don't rush and take your time. Both warnings are about transmission errors. Transmission errors can happen from time to time. This is okay and doesn't harm (if you just miss a frame of data from time to time). If this happens once per second this isn't fine anymore and must be resolved.
> 
> How often do you see CRC errors? Please check your wires / you need proper connections / double check GND. Avoid long cables etc.

Finally it seems to work correctly. I have had to change the GPIOs to 14 and 4 and now there seems to be no cuts.
The configuration with this WT32 ETH1 is this way:

substitutions:
  name: w32-banco2

esphome:
  name: ${name}
  friendly_name: w32_banco2
  platform: ESP32
  board: esp-wrover-kit

logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "1u757X/nw4cuSjGJx+PfiE7Mqspw="

ota:
  - platform: esphome
    password: "625dfd314ff4772d"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Optional manual IP
  manual_ip:
    static_ip: 192.168.1.196
    gateway: 192.168.1.254
    subnet: 255.255.255.0

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

uart:
  - id: uart_1
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: GPIO14
    rx_pin: GPIO4

jk_modbus:
  - id: modbus1
    uart_id: uart_1
    rx_timeout: 50ms

jk_bms:
  - id: bms1
    jk_modbus_id: modbus1
    update_interval: 5s

sensor:
  - platform: jk_bms
    jk_bms_id: bms1
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    min_cell_voltage:
      name: "${name} min cell voltage"
.
.
.
.
.
.
.
.

jouking, 2024-08-05:
Thanksss!!!
Gracias!!!!

syssi, 2024-08-05:
You are welcome!

closed 2024-08-05 by syssi