syssi/esphome-jk-bms #564: @txubelaxu Hi! I'm trying to understand your project and if I understood correctly, you connect the inverter to the battery via rs485 via CAN bus and simultaneously receive data from the inverter also via rs485 or is it not quite like that?

txubelaxu, 2024-08-08:
@txubelaxu Hi! I'm trying to understand your project and if I understood correctly, you connect the inverter to the battery via rs485 via CAN bus and simultaneously receive data from the inverter also via rs485 or is it not quite like that?

_Originally posted by @vpritulin in https://github.com/syssi/esphome-jk-bms/issues/464#issuecomment-2276331029_
            

Hi @vpritulin!!

The project was about to add to "esphome-jk-bms" original project the compatibility of JK-PB via internal RS485 ports. At the beginning, I tried to "listen (only) to" the RS485 network to sniffing information across the network. I got it. 
But I saw that some information was missing in the network (device info frame type 03 of the JK-BMS that was acting as MASTER 0x00)
So, I changed the strategy: I tried ESP32 to acting as MASTER. So, I moved my jk-bms to address 0x01 and 0x02.
The state of the project is that I have got READ & WRITE almost all paramethers (missing RCV Time and RFV Time). I have inserted the capacity to "broadcast" to all the BMSs in the RS485 network one setting when you change it. This is a "broadcast" setting that you can dissable in the config, of course.

I prefer RS485 to BT because it uses less resources, BT is free to connect with mobile...

Special mention to "esphome-jk-bms-can" project. Now, in the "development" branch there is an integration to @txubelaxu "esphome-jk-bms" project. So, with this two projects together, yes, we have an ESP32 between the Inverter and BMSs. :-)

Welcome!!

vpritulin, 2024-08-08:
Maybe I didn't fully understand your project, but I have an inverter with RS485 support and I would like to connect the battery to it via CAN bus and connect the ESP to RS485 so that I could immediately receive data from the inverter and battery. So that I don't have to use RS232 to receive information about the inverter and can get by with 1 ESP instead of 2.

txubelaxu, 2024-08-08:
> @txubelaxu <https://github.com/txubelaxu> Maybe I didn't fully understand

BMS1 <---->|
BMS2<----->|
                    |
                    -----------> ESP32 <-----------------> INVERTER
       (RS485)                                  (CAN)

But "esphome-jk-bms-can" project has different posibilities that I do not
know.

vpritulin, 2024-08-08:
I thought to use such a connector, connected to the inverter on the RS485 connector, and then 1 cable to the BMS, where there may be a bus, and the second in parallel to the esp32 through the RS485 to TTL converter
[image]

txubelaxu, 2024-08-10:
So, your inverter have not a CAN bus, have it?

Those connectors can be used, as far as the impedance in the network complies the RS485 specification. That can see when connecting into the system. If the cable lengths are not too long I think it should work. But as said, you only know when connecting...

vpritulin, 2024-08-12:
@txubelaxu Hello! My inverter has a CAN bus, but I would like to connect the inverter to the battery via the CAN bus. With the BMS, receive data on the ESP and in parallel from the inverter via the CAN bus send information about the inverter, using only 1 ESP and not using the RS232 connector

txubelaxu, 2024-08-12:
take a look to the "development" branch of "esphome-jk-bms-can" project. It
uses CAN between Inverter and ESP32, and RS485 between ESP32 and BMSs.

vpritulin, 2024-08-12:
@txubelaxu Maybe another branch? I didn't find the developer

txubelaxu, 2024-08-12:
It is another project. There are 2 projects: esphome-jk-bms and
esphome-jk-bms-can

Here the link to esphome-jk-bms-can -->
https://github.com/txubelaxu/esphome-jk-bms-can/