syssi/esphome-jk-bms #581: Sensor readings constantly do not match

man55, 2024-08-18:
There is something strange here, how these sensors are updated, but their readings constantly do not match. How can this be fixed?
3.285 - 3.281 = 0.004 but not a 0.006
The coloring of cells with minimum and maximum voltages does not coincide.
```
sensor.jk_bms_delta_cell_voltage
sensor.jk_bms_max_cell_voltage
sensor.jk_bms_min_cell_voltage
sensor.jk_bms_max_voltage_cell
sensor.jk_bms_min_voltage_cell
...
```

[image]

syssi, 2024-08-18:
Could you provide your YAML? I would like to look up some important details f.e. the used log level. You are using the native `api` and no `mqtt`, correct?

syssi, 2024-08-19:
It looks like the sensors aren't updated properly if the debug lovel isn't `DEBUG`. Please change the log level from (probably?) `INFO` to `DEBUG` and do another comparison / provide another screenshot. Does it fix the issue?

man55, 2024-08-19:
> Could you provide your YAML? I would like to look up some important details f.e. the used log level. You are using the native `api` and no `mqtt`, correct?
[esphome-web-516dac.yaml.txt](https://github.com/user-attachments/files/16657238/esphome-web-516dac.yaml.txt)
Attached my yaml. Yes, I'm using `api`.
As I see, there is already a `DEBUG` level.

syssi, 2024-08-19:
I've pushed a feature branch to debug the issue on your site / make the problem visible at the ESPHome log probably. Please change the `external_components_source` to `@debug-delta`:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS v11 via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@debug-delta
```

Recompile + flash your ESP and provide some logs. I'm interested in these log messages:

``` 
[11:36:09][E][jk_bms_ble:391]: Computed min_cell_voltage: 3.615 V
[11:36:09][E][jk_bms_ble:392]: Computed max_cell_voltage: 3.762 V
[11:36:09][E][jk_bms_ble:393]: Computed min_voltage_cell: 1
[11:36:09][E][jk_bms_ble:394]: Computed max_voltage_cell: 13
[11:36:09][E][jk_bms_ble:395]: Computed delta_cell_voltage: 0.14700 V
[11:36:09][E][jk_bms_ble:397]: Reported min_voltage_cell: 1
[11:36:09][E][jk_bms_ble:398]: Reported max_voltage_cell: 13
[11:36:09][E][jk_bms_ble:399]: Reported delta_cell_voltage: 0.147 V
```

Thanks in advance!

man55, 2024-08-19:
Logs: 
[logs_esphome-web-516dac_run.txt](https://github.com/user-attachments/files/16658684/logs_esphome-web-516dac_run.txt)

ps. At the moment there is 90% charge and the voltage of the cells is almost equal. A little later, when there is a charge or discharge, I will take the log again.

syssi, 2024-08-19:
@man55 Friendly reminder.

man55, 2024-08-19:
New log on discharging.
[logs_esphome-web-516dac_logs.txt](https://github.com/user-attachments/files/16663552/logs_esphome-web-516dac_logs.txt)

```
[20:37:33][E][jk_bms_ble:391]: Computed min_cell_voltage: 3.300 V
[20:37:33][E][jk_bms_ble:392]: Computed max_cell_voltage: 3.306 V
[20:37:33][E][jk_bms_ble:393]: Computed min_voltage_cell: 9
[20:37:33][E][jk_bms_ble:394]: Computed max_voltage_cell: 15
[20:37:33][E][jk_bms_ble:395]: Computed delta_cell_voltage: 0.0060 V
[20:37:33][E][jk_bms_ble:397]: Reported min_voltage_cell: 11
[20:37:33][E][jk_bms_ble:398]: Reported max_voltage_cell: 4
[20:37:33][E][jk_bms_ble:399]: Reported delta_cell_voltage: 0.003 V
```

[image]

syssi, 2024-08-19:
I've pushed some changes. The measurements are calculated now instead of using the reported ones of the BMS firmware. Please flash your ESP another time and give it a try! Does it look better now?

man55, 2024-08-19:
I watched for a while. Oh yeah, it looks correct now. Thanks.

So the problem was incorrect BMS reported values?
[image]

syssi, 2024-08-19:
> So the problem was incorrect BMS reported values?

Yes. This payload isn't used anymore. You can switch back to the `@main` branch again. The feature branch will be deleted soon.

closed 2024-08-19 by syssi