syssi/esphome-jk-bms #586: Add Inverter CAN Support

JohnGalt1717, 2024-08-29:
Now that the OSS project is dead, it would be nice to put back in to this at least the JK CAN protocol(s) that the new bms boards support so that we could connect this wired, and then output CAN to inverters etc.

It looks like this could be reintegrated.

https://github.com/Uksa007/esphome-jk-bms-can/tree/deprecated/components/jk_can

It seems very cost effective to build your own board:

[ESP32 with 60V input] (https://www.amazon.com/gp/product/B0CK3KMCYF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
[4 Pin Connectors](https://www.amazon.com/Connector-1-25mm-Adapter-Extension-Helicopter/dp/B0C77GS8PP)
[CAN Transceiver](https://www.amazon.com/NOYITO-TJA1051-CAN-Transceiver-Consumption/dp/B07DMTFHS7)
[Ethernet Jacks](https://www.amazon.com/ATMLCOER-Ethernet-Terminal-Breakout-Connector/dp/B0CW1ML5C3)

~$35 per

I'll chip in a 3d model of a case that will hold everything nicely and schematics to wire it up, if someone wants to take a go at integrating the code back.

syssi, 2024-08-30:
Nice to see you John! :-) Please keep in mind the project isn't about using the CAN bus port of the JK-BMS. It's more or less retrieving the data via UART-TTL and pass specific measurements via CAN to an Goodwe inverter:

https://github.com/Uksa007/esphome-jk-bms-can?tab=readme-ov-file#schematics

These two files look like a draft because the implementation cannot work:

https://github.com/Uksa007/esphome-jk-bms-can/blob/deprecated/components/jk_can/jk_can.cpp
https://github.com/Uksa007/esphome-jk-bms-can/blob/deprecated/components/jk_can/jk_can.h

The most important parts of the CAN communication between the ESP32 and the inverter can be found here:

https://github.com/Uksa007/esphome-jk-bms-can/blob/deprecated/esp32-example-can.yaml

These CAN IDs are supported:

```
          can_id: 0x359 # Warning, Alarms
          can_id: 0x351 # BMS instructs inverter: Charge Volts, Charge Amps, Discharge Amps, Min voltage
          can_id: 0x355 # Actual SOC, SOH
          can_id: 0x356 # Actual Volts, Amps, Temp
          can_id: 0x35C # Request flag to Enable/Disable: Charge, Discharge
          can_id: 0x70  # Actual Max Cell Temp, Min Cell Temp, Max Cell V, Min Cell V
          can_id: 0x371 # Actual Max Cell Temp ID, Min Cell Temp ID, Max Cell V ID, Min Cell ID
          can_id: 0x35E # GOODWE
```

I've started project with a similar idea long time ago here: https://github.com/syssi/esphome-virtual-can-bms

The idea is the same (1. Consume the measurements of any BMS / Monitor the state of ESHome sensor entities + 2. Pass the measurements using the `virtual_can_bms` component via CAN to an supported inverter.). It's still a draft / proof-of-concept because there is a lack of beta testers.

syssi, 2024-08-30:
Let's focus on a different use case: Do you want to talk to your BMS using CAN bus? :-)

JohnGalt1717, 2024-08-30:
Well, I have a ton of JK BMSes that don't have CAN that can talk to EG4 Inverters (some don't have can full stop). So I want to create a CAN network for that purpose.

But I could also see using The above, or even simpler boards with like 8266 or even lower than that, to translate the UART-TTL to the CAN network, then hook up a single ESP32 with ESPHome to that CAN Network and have it listen to everything on the CAN network including the inverter stuff and feed it into Home Assistant normalized.  I think this use case of a single device on a solar CAN Network could be huge, and expanding it out to more and more supported devices would be a major win.  And being able to have a device that can take 5-60V input and can connect to devices that don't speak CAN, convert them to CAN and put that device on that CAN network would be part of that.

Thoughts?