syssi/esphome-jk-bms #588: JK-B2A8S20P HW V11.XA SW V11.48 is supporting?

BytEvil, 2024-09-02:
Hi, friends! JK-B2A8S20P HW V11.XA SW **V11.48** is supporting?  Can i connect this BMS directly over GPS port to Wemos D1 RX-TX pins, do I need to install any converter? Or can this BMS **only** be connected via BlueTooth?

Thanks in advance!

syssi, 2024-09-02:
Both is supported in general. Some BMS models doesn't want to talk to a ESP8266. If it doesn't work out of the box please try a ESP32 for comparison.

closed 2024-09-02 by syssi

BytEvil, 2024-09-02:
@syssi, thanks! Wemos D1 not communicate with BMS at all. I`ll try ESP32-Wroom later...

BytEvil, 2024-09-03:
@syssi Worked as it should be with ESP32-Wroom over GPS port. Big thanks for your great work. ;)

syssi, 2024-09-03:
Thanks for the feedback! I will add your device (firmware) to the list of supported devices.