syssi/esphome-jk-bms #589: Bluetooth PIN support

HansWegman, 2024-09-02:
Hello Sebastian,

I am trying to get your code to work with my unit which has a bluetooth pin set which cannot be removed. I found some code which might work but I have no idea where to implement this. Could you help out here maybe?

`
ble_client:
- mac_address: "A4:C1:37:04:2D:BE"
  id: hans_battery
  on_passkey_request:
    then:
      - logger.log: "Authenticating with passkey"
      - ble_client.passkey_reply:
          id: hans_battery
          passkey: 000000
`

syssi, 2024-09-02:
It looks like you aren't aware of this important detail: https://github.com/syssi/esphome-jk-bms/issues/139

The BMS firmware never verifies the password.

HansWegman, 2024-09-02:
Thanks Sebastian, I am not worried about security though and it is the existence of the bluetooth pin which seems to be a problem if interpret my google searches correctly. They come up with similar problems for the Overkill Solar software.

The problem is that my unit responds data perfectly in the app but not with your code. I implemented the on_passkey_request which does not seem to make a difference so you are probably right that it is not requested but still it messes up the communication somewhere as you can see in the attached log which shows no data received from the unit.

Is there anyway I can debug this further to see how it is solvable?

[logs_test_logs-5.txt](https://github.com/user-attachments/files/16837922/logs_test_logs-5.txt)

syssi, 2024-09-02:
This project is about the JK-BMS. Your log is about a different BMS (JBD-BMS). The JK-BMS firmware doesn't provide any real security measurements.

HansWegman, 2024-09-02:
Oh I am so sorry! I have been trying for over a week with all sorts of packages without success though so I did not look close enough. I am really sorry to have bothered you.

syssi, 2024-09-02:
Let's discuss the issue at the JBD-BMS project! :-)

closed 2024-09-02 by syssi