syssi/esphome-jk-bms #591: JK B2A24S20P voltage error

Ghilm1986, 2024-09-04:
Hi, I was using your jk project since the last year.
Because i changed the ssid I needed to modify the file, but when I tried to save it I had an error about JK02 unkown value when i was creating the bin file.
I suddenly tried the other option about jk02_32s but I had completely wrong voltages.
My hw version is 10.0.
[image]
[image]
The voltages are wrong but the bms is working perfectly if i use its app.

syssi, 2024-09-04:
Please  use `JK02_24S`.

Ghilm1986, 2024-09-05:
Perfect..it works now.thanks a lot

closed 2024-09-05 by syssi