syssi/esphome-jk-bms #598: give error when compile

oguzatagan, 2024-09-14:
hi

Failed config

external_components: [source /config/esphome/powerwall-bms.yaml:21]
  - 
    fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
    Use '--' to separate paths from revisions, like this:
    'git <command> [<revision>...] -- [<file>...]'.
    source: github://syssi/esphome-jk-bms@main
    refresh: 0s
  - source: github://syssi/esphome-ant-bms@main
    refresh: 0s

my code is here please helpp

'''substitutions:
  name: powerwall-bms
  name1: ant-bms
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 
  external_components_source1: github://syssi/esphome-ant-bms@main
  mac_address1: 

esphome:
  name: powerwall-bms
  project:
    name: "syssi.esphome-ant-bms"
    version: 2.0.0

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s
  - source: ${external_components_source1}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  - platform: esphome
    password: ""

logger:
  level: DEBUG

api:
  encryption:
    key: ""

      
ble_client:
  - mac_address: ${mac_address}
    id: client0
  - id: client1
    mac_address: ${mac_address1}

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0

ant_bms_old_ble:
  - id: bms1
    ble_client_id: client1
    update_interval: 5s
    password: "12345678"

binary_sensor:
  - platform: heltec_balancer_ble
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"
  - platform: ant_bms_old_ble
    ant_bms_old_ble_id: bms1
    online_status:
      name: "${name1} online status"

button:
  - platform: heltec_balancer_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"
  - platform: ant_bms_old_ble
    ant_bms_old_ble_id: bms1
    restart:
      name: "${name1} restart"
    shutdown:
      name: "${name1} shutdown"
    clear_counter:
      name: "${name1} clear counter"
    balancer:
      name: "${name1} balancer"
    factory_reset:
      name: "${name1} factory reset"
  - platform: restart
    name: "Soft Restart"

number:
  - platform: heltec_balancer_ble
    cell_count:
      name: "${name} cell count"
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    max_balance_current:
      name: "${name} max balance current"
    balance_sleep_voltage:
      name: "${name} balance sleep voltage"
    balance_start_voltage:
      name: "${name} balance start voltage"
    nominal_battery_capacity:
      name: "${name} nominal battery capacity"

sensor:
  - platform: heltec_balancer_ble
    min_cell_voltage:
      name: "${name} min cell voltage"
      on_value_range:
        - above: 3.600
          then:
            - switch.turn_off: heltec_balancer
        - below: 3.550
          then:
            - switch.turn_on: heltec_balancer
    max_cell_voltage:
      name: "${name} max cell voltage"
      on_value_range:
        - above: 4.140
          then:
            - switch.turn_on: heltec_balancer
        - below: 4.135
          then:
            - switch.turn_off: heltec_balancer
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    total_voltage:
      name: "${name} total voltage"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    cell_detection_failed_bitmask:
      name: "${name} cell detection failed bitmask"
    cell_overvoltage_bitmask:
      name: "${name} cell overvoltage bitmask"
    cell_undervoltage_bitmask:
      name: "${name} cell undervoltage bitmask"
    cell_polarity_error_bitmask:
      name: "${name} cell polarity error bitmask"
    cell_excessive_line_resistance_bitmask:
      name: "${name} cell excessive line resistance bitmask"

  - platform: ant_bms_old_ble
    ant_bms_old_ble_id: bms1
    current:
      name: "${name1} current"
    soc:
      name: "${name1} soc"
    total_battery_capacity_setting:
      name: "${name1} total battery capacity setting"
    capacity_remaining:
      name: "${name1} capacity remaining"
    battery_cycle_capacity:
      name: "${name1} battery cycle capacity"
    power:
      name: "${name1} power"
    temperature_1:
      name: "${name1} temperature 1"
    temperature_2:
      name: "${name1} temperature 2"
    temperature_3:
      name: "${name1} temperature 3"
    temperature_4:
      name: "${name1} temperature 4"
    temperature_5:
      name: "${name1} temperature 5"
    temperature_6:
      name: "${name1} temperature 6"
    charge_mosfet_status_code:
      name: "${name1} charge mosfet status code"
    discharge_mosfet_status_code:
      name: "${name1} discharge mosfet status code"
    balancer_status_code:
      name: "${name1} balancer status code"

switch:
  - platform: heltec_balancer_ble
    balancer:
      name: "Heltec balancer"
      id: heltec_balancer
  - platform: ant_bms_old_ble
    ant_bms_old_ble_id: bms1
    charging:
      name: "${name1} charging"
    discharging:
      name: "${name1} discharging"

text_sensor:
  - platform: heltec_balancer_ble
    operation_status:
      name: "${name} operation status"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
    buzzer_mode:
      name: "${name} buzzer mode"
    battery_type:
      name: "${name} battery type"
  - platform: ant_bms_old_ble
    ant_bms_old_ble_id: bms1
    charge_mosfet_status:
      name: "${name1} charge mosfet status"
    discharge_mosfet_status:
      name: "${name1} discharge mosfet status"
    balancer_status:
      name: "${name1} balancer status"
    total_runtime_formatted:
      name: "${name1} total runtime formatted"

interval:
  - interval: 5min
    then:
      - button.press: retrieve_settings_button '''

syssi, 2024-09-14:
```
$ esphome config test.yaml >/dev/null
INFO ESPHome 2024.8.3
INFO Reading configuration a.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Updating https://github.com/syssi/esphome-ant-bms.git@main
INFO Configuration is valid!
$
``` 

Your YAML compiles fine here.

oguzatagan, 2024-09-14:
> ```
> $ esphome config test.yaml >/dev/null
> INFO ESPHome 2024.8.3
> INFO Reading configuration a.yaml...
> INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
> INFO Updating https://github.com/syssi/esphome-ant-bms.git@main
> INFO Configuration is valid!
> $
> ```
> 
> Your YAML compiles fine here.

i didnt change anythink. but i was update 2024.8.3 then give this error. red line under external components source. what can i do according to you

syssi, 2024-09-15:
Please try a clean build: `esphome clean yourconfig.yaml`. Do you use a Linux or Windows host system?

alex-dokienko, 2024-10-06:
experiencing same issue on home assistant when trying to build via HA esphome addon
cleaning build files didn't work.

Is there anything else I can do?
[image]

syssi, 2024-10-21:
This looks like an ESPHome issue in general and isn't related to this project. Could you ask the ESPHome community (discord) for help? You could use another host to install esphome and flash the node manually.

closed 2024-10-21 by syssi