syssi/esphome-jk-bms #599: Parallel comms lost when polling master BMS

moto501, 2024-09-15:
When more than one battery is connected in parallel and the JK-BMS (Inverter type) is connected to Venus via BMS.Can, polling the master BMS for data causes the comms between the batteries to be lost for a moment. The result is the combined DCL/CCL values as reported by VenusOS (Cerbo) is lost. For example with two BMS in parallel configured for a DCL/CCL of 150amp each (total 300amp) the cerbo will report 300amp normally. However randomly or when the master is polled via BLE, the CCL/DCL is halved to 150amp. The attached image shows this issue as reported by VRM.
[image]

If the device polling the master BMS is disconnected, the issue resolves.

Handy-Harry, 2024-09-17:
I have the same issue. I have updated the firmware from the BMS from 15.17 to 15.30. Now I have this problem. I don't know about the other firmwares

moto501, 2024-09-18:
Did you see this prior to 15.30?

Handy-Harry, 2024-09-18:
No, I even downgraded beck to 15.17 then it's gone

mapchen, 2024-10-06:
running some batteries on <15.30 with no issue but I can confirm the issue on 15.30

ErJeeBee, 2024-10-06:
Same problem here.
Two batteries running 15.30. Polling data via bluetooth and via rs485.
Polling the master(0000) via bluetooth results in more disturbance then polling the slave(0001).
For now disabled polling via bluetooth.
Polling data via rs485(read only) seems not to be a problem.  
Maybe it has something to do with the added function to enable/diable bluetooth with the powerbutton in firmware 15.30.

mapchen, 2024-10-07:
by the way, I have the same problem while using the iphone app.

syssi, 2024-10-21:
This looks like a firmware issue and cannot be resolved here.

mapchen, 2024-10-31:
does anyone know if this still exists with 15.32?

moto501, 2024-10-31:
Yes, it's a 15.32 issue we think.  Only with Bluetooth that is

closed 2025-07-25 by syssi