syssi/esphome-jk-bms #600: manual update external plugin, why doesn't the automatic update

mirasu, 2024-09-16:
Hello. I have a problem with my bms jk,
I have the Bms connected via bluetooth to home assistant. but since the last update it loses the connection half the time.

I have compiled the Yaml of esphome with version 2.0.0 from its repository. but I see that there is a version 2.1.0, I can't get it to update the version. Could you tell me how to force it to update manually?

I use the external plugin. Will the update improve the connection with my bms JK?

Thanks

syssi, 2024-09-17:
If you use the `@main`  branch at the external component substitution you are always using the newest version of the component. I tag the different versions to be able to use a specific version instead of the `main` branch.

This project haven't changed much. I don't expect stability issues caused by the component.

mirasu, 2024-09-17:
Ok, I changed the version in the yaml like this:
project:
name: "syssi.esphome-jk-bms"
version: 2.1.0
I can now see the new version in the log. I hope it works fine, like before I updated Home Assistant to 24.9.

Thanks

closed 2024-10-21 by syssi