syssi/esphome-jk-bms #601: Fake battery / soc manipulation for sofar inverter charge

git-greatguy, 2024-09-18:
Hi,

Not an issue at all but a question related to my setup.

Long story short, my inverter has the "feature" to stop charge as soon as bms report 100% soc.
To make rid of that situation and saturate cells if soc is off, i put an esp32 on uart with a small piece of code i grab on diysolarforum. ( for reference  : https://diysolarforum.com/threads/jkbms-inverter-bms-soc-fixer-ghost-battery-open-hardware-project.88554/)
This fake a 2AH battery/0% soc on adress 15, so master bms wont never send 100% soc to inverter. It works great... but lack monitoring functionnality. :)

Is it possible to replicate this fake bms behavior with esphome-jkbms ? i see a "fake bms" yaml example in "test" directory but i dont know if it will work to send payload on an interval basis ?

The "ghost battery" only send frames on specific read from the rs485.
ex : "0x10 0x16 0x1E" / send frame1 (settings) or "0x10 0x16 0x20" / send frame 2 (status).

Maybe a regular basis is ok, but i'm not aware of all rs485 things in bms communication.

Any thought ?

thanks a lot !

syssi, 2024-09-18:
In general the behaviour could also be implemented using ESPHome but it means a complete rewrite of the mentioned project.

git-greatguy, 2024-09-18:
Ok, thanks for your reply. 
So, let's wait for next firmware 15.33/.34 for PB hardware. They should release a fix for that 100% soc behavior based on rcv timer.
I will then switch from "fake bms" to monitoring with this esp :)
Have a nice day @syssi !

closed 2024-09-18 by git-greatguy