syssi/esphome-jk-bms #604: Current calibration always same

mevlutsenol, 2024-10-03:
Hi,

I have JK_B1A24S15P. But current calibration field always same value.

[image]

[image]

syssi, 2024-10-03:
There is not much known about the current calibration of the JK-BMS. What do you expect here? Do you know in detail how to calibrate the BMS?

mevlutsenol, 2024-10-03:
In bluetooth application, this field shown current balance.

[image]

syssi, 2024-10-03:
It looks like you are mixing things up here. Are you are of the `balancing_current`?

 https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml#L307-L308

There is a difference between `current calibration` and the `balancing current`.

syssi, 2024-10-04:
No. The UART-TTL protocol doesnt provide this measurement. It's available via BLE only.

closed 2024-10-04 by syssi