syssi/esphome-jk-bms #608: One BMS going off after some time.

LifeofDan-EL, 2024-10-09:
So I currently use the multiple BMS configuration. I have two batteries that I use your integration with, and I really find it useful. However, I discovered that lately, the bms0 is fond of going offline which is weird and it has been recurring much more than I expected. There are also times when both of them go offline but it doesn't happen for too long.

[image]
[image]
[image]

### My config
```
substitutions:
  name: jk-bms
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:80:11:41:7C
  bms1_mac_address: C8:47:80:11:08:1B
 
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: "xxxxxxxxxxxxxxxxxx"
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - switch.turn_off: ble_client_switch1
      - logger.log: "BLE connection suspended for OTA update"

# Enable logging
logger:
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxx"

esp32_ble_tracker:
  scan_parameters:
    active: false
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0 || x.get_name().rfind("JK_", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
[... 371 lines cut ...]
  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${bms0} enable bluetooth connection"
  - platform: ble_client
    ble_client_id: client1
    id: ble_client_switch1
    name: "${bms1} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"
```
Thanks so much for the work you do.

syssi, 2024-10-10:
Is this reproducible? It's always the second BMS? What happens if you flip the initialization order:

```
ble_client:
  - mac_address: ${bms1_mac_address}
    id: client1
  - mac_address: ${bms0_mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client1
    protocol_version: ${bms1_protocol_version}
    throttle: 5s
    id: bms1
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 5s
    id: bms0
```

Which BMS connection stops this time?

LifeofDan-EL, 2024-10-10:
Actually, it's the first BMS that goes off. The first which is bms0. 

I just flipped the initialization order, let me see what happens with time.

vpritulin, 2024-10-20:
@LifeofDan-EL Hi, I have the same problem, but I don't use the BLE connection. On average, it takes about 15 minutes offline. When I tested the BLE, it was the same. So far I haven't managed to solve it or understand why it happens.

syssi, 2024-10-21:
@LifeofDan-EL Sorry for the silence/delay. Is my understanding of your issue correct:

We have proofed it's always the first BLE client connection which fails. It doesn't depend on the connected hardware. The issue is gone if a single BLE connection is used. 

Correct?

LifeofDan-EL, 2024-10-21:
> @LifeofDan-EL Sorry for the silence/delay. Is my understanding of your issue correct:
> 
> We have proofed it's always the first BLE client connection which fails. It doesn't depend on the connected hardware. The issue is gone if a single BLE connection is used.
> 
> Correct?

Yes, if you look at today's data you can see that while bms0 is blanking out, bms1 has been stable. On some days, the bms0 is more sparse on the graph. 
[image]

Christyria, 2025-01-01:
Same here :-)

Have 2 JK-BMSes and the 2 Year older JK-BMS run's great with the identical ESP8266 Hardware Layout. The Second JK-BMS that is about half a Year old shows the same Interruptions.

[image]

[image]

Also it comes and leaves, and when it's no longer sending Data also no Ping from the second BMS. It comes and runs or it don't run.

WiFi Signal is 99-100% and it is 50 cm away.

Strange. Had played with RX Puffer on the 115200 Baud Settings (Doubled it). Same Error.

The other first BMS with exact the same ESP Hardware Layout (DC-DC Converter, Filter Ceramic Capacitors, Fuse and ESP) runs 100% stable without Issues. The second problematic also changed the ESP8266, same behaviour.

Strange enough, it can run 1-2 Weeks flawlessy and then hicking up like above, or when hard reset (Pull it from the Socket) it worked 1-2 Minutes then Stall and second ESP Pull, then it runs and then above Hickups.

syssi, 2025-01-01:
@Christyria This issue is about a BLE connection. Do you use a ESP8266 or ESP32?

Christyria, 2025-01-01:
@syssi  Ouch Sorry. I'm not using an BLE Connection here but the Standard WiFi with Power Saving Mode Off.
It's the ESP8266.

When my Comment is wrong here, cut it and please make a new Thread.

Thanks.

gaitken, 2025-07-21:
I get similar problem over bluetooth.
[12:32:49][I][esp32_ble_client:127]: [0] [C8:47:80:28:93:16] 0x00 Attempting BLE connection
[12:32:49][D][esp-idf:000][BTU_TASK]: W (4841566) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[12:32:49][D][esp-idf:000][BTU_TASK]: W (4841670) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[12:32:50][D][esp-idf:000][BTU_TASK]: W (4842627) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[12:32:51][E][esp32_ble_client:255]: [0] [C8:47:80:28:93:16] Got ESP_GATTC_OPEN_EVT while not in connecting state, status=133

any help would be aprreciated.