syssi/esphome-jk-bms #611: Display protocol doesn't return cell voltages data

bearpawmaxim, 2024-10-13:
Hello!
Using your `jk_bms_display` component to read data from display port of my B2A20S24P bms (because gps port isn't working...).
All the values are read correctly except the `min_cell_voltage` and `max_cell_voltage`.

Here's the logs
[logs_battery-status-display_logs.txt](https://github.com/user-attachments/files/17356341/logs_battery-status-display_logs.txt)

Config can be seen at https://github.com/bearpawmaxim/esphome-configs/blob/master/battery-status-display/jkbms-display-port.yaml

syssi, 2024-10-13:
I've merged a fixed. Please recompile your setup and give it another try!

closed 2024-10-13 by syssi

bearpawmaxim, 2024-10-13:
It works now! Thanks!
[image]

syssi, 2024-10-13:
Nice!

commit 0ef61ba3 (2024-10-13): Register missing sensors (Fixes: #611) (#612)