syssi/esphome-jk-bms #613: JK PB2A16S20P v15a fw15.29 issues with some values

Ghilm1986, 2024-10-15:
Hi, i tried to connect my new bms but I had issues with some values.
I used the protocol you suggest in the discussion.
Thanks in advance 
[image]
[image]
[image]
[image]

syssi, 2024-10-15:
Could you post your YAML configuration too?

Ghilm1986, 2024-10-15:
[jk02-firmware.factory.bin.zip](https://github.com/user-attachments/files/17381278/jk02-firmware.factory.bin.zip)

Ghilm1986, 2024-10-15:
> This is the firmare image which doesn't help. Could you provide your YAML configuration (yourname.yaml).

I don t know.. where is it?
Esp home... device...3 dots... download yaml?
Or the home assistant configuration... (I don t know where download it)

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

syssi, 2024-10-15:
It's the content of the text area. How did you start?

Ghilm1986, 2024-10-15:
Which text area?
I copied your script from the discussion you provide me, modified with my data and flashed esp32.
It s the second jk bms i m using, i had an old 10 version that is working fine with bt.

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

syssi, 2024-10-15:
Could you provide a link to the discussion you are refering? I would like to have a look at the YAML.

Ghilm1986, 2024-10-15:
https://github.com/syssi/esphome-jk-bms/issues/390
U said to use the protocol at comment 390.
Ok i understood u need to check the yaml, but i don't know how I can provide it to you
Pls give me some instructions

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

syssi, 2024-10-15:
What about pasting the YAML here?

syssi, 2024-10-21:
Long story short: This issue is about https://github.com/syssi/esphome-jk-bms/blob/main/esp32-jk-pb-modbus-example.yaml. The Example is incomplete and should be extended using the Modbus register description here:

https://github.com/syssi/esphome-jk-bms/tree/main/docs/pb2a16s20p

I'm looking for some testers. Do you like to do some testing?

This issue is somehow a duplicate of #615.

Didou30, 2024-10-23:
Hello, I can test it, I have 5 to start, which yalm should I use to do it in BLE?

carlosafgraca, 2024-12-09:
Hello, I'm encountering the same issue with incorrect values ("Wrong" values) in my PB2A16S20P. Additionally, my device sometimes reboots when reading via BLE using the ESP32 Dev Board. If you need a tester, please let me know!

Thanks!

> Hi, i tried to connect my new bms but I had issues with some values, capacity and Soc too. I used the protocol you suggest in the discussion. Thanks in advance ![Screenshot_2024-10-15-17-21-55-33_c3a231c25ed346e59462e84656a70e50](https://private-user-images.githubusercontent.com/142755791/376680072-1aaebfba-e268-4778-a9cf-8bf9d12aaaec.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3ODMyNDMsIm5iZiI6MTczMzc4Mjk0MywicGF0aCI6Ii8xNDI3NTU3OTEvMzc2NjgwMDcyLTFhYWViZmJhLWUyNjgtNDc3OC1hOWNmLThiZjlkMTJhYWFlYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQyMjIyMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YzhhOTY3MjE2Y2U0NmQyMWMzNjFlMWZhZjhiZTE3ZWYyMjVlYjE3NjVmNTE4MmIxYWZhMjEzZTg5MzFkNjJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.wNa55YOp6IgM011_MDmWxeB3X1l4t-lF9QKsqUkVGrU) ![Screenshot_2024-10-15-17-22-00-09_c3a231c25ed346e59462e84656a70e50](https://private-user-images.githubusercontent.com/142755791/376680086-21de1c96-a4e8-41a2-a761-1615ca5471be.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3ODMyNDMsIm5iZiI6MTczMzc4Mjk0MywicGF0aCI6Ii8xNDI3NTU3OTEvMzc2NjgwMDg2LTIxZGUxYzk2LWE0ZTgtNDFhMi1hNzYxLTE2MTVjYTU0NzFiZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQyMjIyMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMDU3ZGNkZWMwYzBiMTIyYzAwM2I5MTY5N2Y4NmEwMWZlMDhhNTk1MWY1YjczZTQ3NTFjNTk0OGYxNmFkMmY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5twQ1oWwALHXh2YzKV1jPKOWLAnn2qEWaZ2np-tKmIY) ![Screenshot_2024-10-15-17-22-04-32_c3a231c25ed346e59462e84656a70e50](https://private-user-images.githubusercontent.com/142755791/376680093-88e75108-b483-4670-8236-b6d341934134.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3ODMyNDMsIm5iZiI6MTczMzc4Mjk0MywicGF0aCI6Ii8xNDI3NTU3OTEvMzc2NjgwMDkzLTg4ZTc1MTA4LWI0ODMtNDY3MC04MjM2LWI2ZDM0MTkzNDEzNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQyMjIyMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zODJiYzJjZmYyZjA4NjYxNWFmNDc0ZmJiZTQ4NmRmNWUwMDVlZTk1YmIxN2M5MjkzYWE3ZTM4Mzc2NmE0MTBkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iO3kKZaQkP7aBzj3lRjB2D5JUR8HylmIyHUiWHmVMcs) ![Screenshot_2024-10-15-17-09-31-27_bd59be8c3f820c1e3d6fc2892aab342d](https://private-user-images.githubusercontent.com/142755791/376679091-de397697-dbcd-4756-a96b-9cea8921839e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3ODMyNDMsIm5iZiI6MTczMzc4Mjk0MywicGF0aCI6Ii8xNDI3NTU3OTEvMzc2Njc5MDkxLWRlMzk3Njk3LWRiY2QtNDc1Ni1hOTZiLTljZWE4OTIxODM5ZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQyMjIyMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYTg1ODM3ZTQ2ZDY1Y2IwNGYyYjkwMjNhZDBiZmUyNDJmZjNiZGZhYzA4ZjhmNDRhYmI1MGI5MzZhNDdhMmM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ddcbvAwypwFUFHcFglVcV6D15E6-7vIFl4q96DhHBZ0)

syssi, 2024-12-25:
This is a duplicate of https://github.com/syssi/esphome-jk-bms/issues/615. Let's close this one.

closed 2024-12-25 by syssi