syssi/esphome-jk-bms #619: Help with MQTT publishing from node-red via Mosquito

Phil-b-l, 2024-10-21:
I expect I am being a muppet but I am using this to connect my bluetooth bms via MQTT to node-red and that works fine except I don't know the topic format to change settings and, specifically to turn charging on and off this is what I am trying jk-bms/switch/jk-bms_charging/set with the payload set to ON I have tried it without the /set and without the /switch and every other combination I can think of but no luck. Can anyone help please?

Phil-b-l, 2024-10-21:
Oops tried to delete query as solved it but couldn't so appear to have closed a blank. Sorry

closed 2024-10-21 by Phil-b-l

syssi, 2024-10-21:
Did find all topics and required payloads + did you disable the auto-discovery topics?

Phil-b-l, 2024-10-21:
Hi
I installed MQTT Explorer and that showed me what was expected and I also realised that I had built a version where the yml didn't include those options and, while I had edited it to inlude some stuff that wasn't in the original, that was a switch I hadn't added. I have now rebuilt with the full selection and just deleted what I don't need in an attempt to be kind to my poor little pi and its mosquito which is working rather hard. I might try and turn off the debug as that seems to rather flood it.

Thanks very much for creating this and for taking the time to respond. It is much appreciated

Kind Regards

Phil

syssi, 2024-10-21:
If you don't use Home Assistant you could disable the discovery topic to get rid of some more traffic:

```
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
  discovery: false
```

Phil-b-l, 2024-10-21:
Ok thanks. Would it still discover it ok after a re-build to upload to it? I suppose it would as I wouldn't think it would use info from MQTT for that? Can I turn off the debug info?

syssi, 2024-10-21:
If you use this extension (https://flows.nodered.org/node/node-red-contrib-ha-mqtt) `discovery: true` is recommended. If you wire every sensor, switch etc. manually you can disable the `discovery` feature. Just play around! Flash your device multiple times and give it a try. Everything can be reverted if it's not an improvement.

You can reduce the log level to `INFO` or something else. It's logging related and doesn't change the `mqtt` topics/features.

Phil-b-l, 2024-10-21:
Thank you very much. I will have a play.

Phil-b-l, 2024-11-14:
Dear Sebastian

Than you again so much for your help on the below. I got it all working well with your help and this allowed me to evaluate this BMS for use in an interchangeable battery project. We were wondering if you might be interested in some paid work to tailor this a bit further. While I would like to do it myself and, in doing so, learn more, in the interests of efficiency it makes more sense to employ the help of a professional. As a very brief description of what would be needed to allow you to decide on feasibility we have several small vehicles with 6 interchangeable 12v batteries each. There will be more batteries than vehicles and battery modules will be swapped between them not necessarily as the same group of 6. My most recent idea is a single esp in the vehicle (more if needed) that can get data from these batteries’ BMSs and send it via MQTT with an individual unique battery ID (ser no? MAC address?) As I see it Modbus won’t work as the batteries can’t always have different Modbus addresses as there could be more batteries than addresses. They would however always be connected to different serial pins and my thought was a simple TTL serial connection each using software serial so needing 6 ports / 12 pins.

We could use a single esp per battery slot which would I suppose simplify things and that could be powered by the battery when inserted and in that case it would just need to extract the battery ID and use that as the ID in the MQTT Messages.

The initial idea was a Bluetooth method but the problem there was identifying  the connection to the MQTT broker which would change with vehicles and the risk of crosstalk due to proximity of other similar vehicles.  

I would value your thoughts and if you would be interested in being paid to help implement a solution. 

Kind Regards

Phil

syssi, 2024-11-17:
Hi Phil! Thanks for your offer. I appreciate it very much but I just want to offer/provide a bit support or guidance for free. Your projects sounds exciting and I would like to get it touch with some more details. I already have a day job and don't want to do paid work over and beyond.

Phil-b-l, 2025-10-08:
Hi Sebastian

It has been a while and the project is progressing well. We have a prototype charging station with 6 slots charging our in house designed battery enclosures with the JK 200A BMS inside working on 12v. This communicated via UART to an ESP32 running your software which then goes to an MQTT broker from where it is packed up by node red which monitors the whole station and another ESP controlling and monitoring the individual slot controlling cooling, the power to the charger and safety interlocks along with a neopixl display giving info. At home I use the bluetooth version of your software on my propulsion and domestic battery pack. With these I can change the setting via MQTT but I can't do this on the UART version. Is there a firmware reason for this or is it software. In the bluetooth version I note that the topic has 'number' rather than 'sensor' in it. I have tries adding 'command' rather than 'state' to the end of it but, as expected that doesn't work. There are a couple of reasons for asking, one is for ease of setup of new BMSs and the other is a bit more 'special' as follows. These batteries are for use in Tuk Tuks in Nepal and are interchangeable and I need a way to simply transfer a couple of bits of date from the vehicle to the charging station these being the driver's numerical ID and the ID of the vehicle. I have noticed in the BMSs now there are a couple of fields for 'User Private Data' and 'User Data2'. My thought was when the batteries are in the vehicle to change these to contain the driver and vehicle IDs as this would then use the battery itself to physically move the data which would be a simple neat way to do it without having to faff about with wifi and the like.

Is there any insight or help you could give with this? Needless to say I would be more than happy to pay you although I know you don't do this as a commercial thing, nor do I generally but I seem to have got myself involved with this project and open sourcing is very important as an ethos here because we want the people this is being developed for to be able to take full control of it rather than there being anything proprietary about it.

Any thoughts you can offer would be greatly appreciated as I am rather out of my depth in this side of things.

Kind Regards

Phil

syssi, 2025-10-08:
> Is there a firmware reason for this or is it software

It's a firmware reason because the UART protocol (GPS port protocol). This protocol supports two write instructions (charging mosfet on/off + discharging mosfet on/off). That's all!

If your BMS firmware is pretty new you could change the UART port protocol from "GPS" to "Modbus". The Modbus protocol supports changing settings via a wired connection.

Phil-b-l, 2025-10-08:
> syssi

Thanks for that. I am using the charge and discharge switches which, as you say, work fine. I am using the latest BMSs and so I can switch the protocol. Will the yaml still work as it is? I see it has the modbus protocol included. If this is the case what mqtt commands would I use to change parameters? Asz you can see I am very much feeling my way through this and doing it rather badly I suspect!

Kind regards

Phil