syssi/esphome-jk-bms #621: Update to HA 2024.10.4 - update of SOME values is not wokring

piknew, 2024-10-27:
Firstly - I thought the issue is ESPHome one, so I have reported: https://github.com/esphome/issues/issues/6385

But... I have noticed that for example "cell voltages" in HA are reported "live". The changes on current (outgoing or incoming to battery) - not.

It is difficult to show it but:

[image]
[image]
[image]
[image]
[image]

piknew, 2024-10-27:
It is very strange. I see that logs on "web_server" contain entries about current and power but logs on console (run just after eg. ota - do not):

Web:
[image]

syssi, 2024-10-27:
Do you use the `api` component to pass states to HA?

syssi, 2024-10-27:
This is the root cause probably: https://github.com/esphome/esphome/pull/7321

piknew, 2024-10-27:
> Do you use the `api` component to pass states to HA?

Yes, my yaml is provided in linked issue https://github.com/esphome/issues/issues/6385

> And this is probably a mitigation: [esphome/esphome#7547](https://github.com/esphome/esphome/pull/7547)

If I understand correctly - this shall/will be introduced to code? Simply - I need to wait for esphome fix? Or revert to HA 2024.10.3...

piknew, 2024-10-27:
FYI - when switched to VERY_VERBOSE logging level I have noticed "Cannot send message because of TCP buffer space".

Also please notice something - this is the beginning of transmission. Last "successful" part is pair of (I have reviewed your code, so more or less I know what is happening):

[19:22:38][V][sensor:043]: 'Cell 5 Voltage': Received new state 3.358000
[19:22:38][D][sensor:094]: 'Cell 5 Voltage': Sending state 3.35800 V with 3 decimals of accuracy

But then we have only 1 line of:
[19:22:38][V][sensor:043]: 'Total Voltage': Received new state 26.859999

And there is NO following message about "Sending state". Instead there is new frame received from UART/modbus. Is it interrupting everything? I am not sure if the code is multi-threaded or interrupted (by eg. IRQ of UART?)?

[image]

syssi, 2024-10-27:
If you don't think the `api` component is the root cause please try the `mqtt` component instead. You will see it will resolve the issue and all states are passed to HA without any loss.

piknew, 2024-10-27:
Ok, thanks, I will try it - never used MQTT before, so I need to learn :) Next info will be for MQTT setup.

BTW. To conclude - my setup is:
JK-BD4A8S4P

Screen is connected to dedicated BMS port. ESP32 connected to UART (as suggested on your README file).

Is it possible that one frame is coming as a result of "request" frame and the other is for display?

syssi, 2024-10-27:
I don't know. The preamble of a display frame is 0xA5 0x5A normally.

piknew, 2024-10-27:
With MQTT - everything seems to be OK (I am on "testing phase", so this may not be 100% valid statement).

> I don't know. The preamble of a display frame is 0xA5 0x5A normally.

Or (for this one I was not looking at the code) - may it be some more data read in next loop? I mean - please see 0xA8 0xFF ... 0xA9 ... (this one seems as to be next part or first packet). Event, if it is not used - this data somehow is read later. This is what I mean:

[image]

syssi, 2024-10-27:
I understand your concern now. The UART debug logger cannot simply print log messages longer as ~150 bytes. A JK-BMS UART-TTL frame is 300 bytes long. It's just printed in two parts to the log.

piknew, 2024-10-28:
Ok, thank you for explanation. I believe this issue may be closed (I have "solved" my problem with switching to MQTT). However, the problem with HA API as I understand, is traced separately?

BTW. These pull requests have underlying issues reported (it would be good idea to link it)?

syssi, 2024-10-28:
I think your issue at the ESPHome project is a good start to proceed. I've created the pull request in the past to make the traffic loss more obvious. At the moment it's too silent and hard to track down. I don't have the skills to resolve the issue. It's a ESPHome core issue for sure and cannot be resolved here.

closed 2024-10-28 by syssi