syssi/esphome-jk-bms #622: Stayability - ESP frequently restarts several times a day

peff74, 2024-10-29:
When expanding my system, I noticed that the ESP restarts quite frequently.
As I don't have any experience with ESP home devices... I can't judge whether this is normal.
[image]
The last reboots are mine, I wanted to get some logs. 
```
[16:22:07][D][Avg Temp:432]: Sensor 2: 17.200001
[16:22:07][D][Avg Temp:442]: Sensor 3 ist ungültig
[16:22:07][D][Avg Temp:450]: Sensor 4 ist ungültig
[16:22:07][D][Avg Temp:458]: Durchschnittstemperatur: 17.049999
[16:22:07][D][sensor:094]: 'Avg BMS Temp': Sending state 17.05000  with 1 decimals of accuracy
[16:22:07][D][climate:396]: 'heat_mat' - Sending state:
[16:22:07][D][climate:399]:   Mode: HEAT
[16:22:07][D][climate:401]:   Action: IDLE
[16:22:07][D][climate:419]:   Current Temperature: 17.05°C
[16:22:07][D][climate:425]:   Target Temperature: 16.00°C
[16:22:07][W][component:237]: Component template.sensor took a long time for an operation (52 ms).
[16:22:07][W][component:238]: Components should block for at most 30 ms.
[16:22:10][W][jk_bms_ble:240]: [20:22:08:25:26:8B] Not connected
[16:22:15][W][jk_bms_ble:240]: [20:22:08:25:26:8B] Not connected
[16:22:17][D][Avg Temp:424]: Sensor 1: 16.900000
[16:22:17][D][Avg Temp:432]: Sensor 2: 17.200001
[16:22:17][D][Avg Temp:442]: Sensor 3 ist ungültig
[16:22:17][D][Avg Temp:450]: Sensor 4 ist ungültig
[16:22:17][D][Avg Temp:458]: Durchschnittstemperatur: 17.049999
[16:22:17][D][sensor:094]: 'Avg BMS Temp': Sending state 17.05000  with 1 decimals of accuracy
[16:22:17][D][climate:396]: 'heat_mat' - Sending state:
[16:22:17][D][climate:399]:   Mode: HEAT
-----------------------------REBOOT--------------------------------
[16:22:28][W][component:170]: Component mqtt cleared Warning flag
[16:22:28][I][mqtt:286]: MQTT Connected!
[16:22:29][I][app:062]: setup() finished successfully!
[16:22:29][I][app:100]: ESPHome version 2024.10.2 compiled on Oct 28 2024, 21:54:25
[16:22:29][I][app:102]: Project syssi.esphome-jk-bms version 2.1.0
[16:22:29][C][logger:185]: Logger:
[16:22:29][C][logger:186]:   Level: DEBUG
[16:22:29][C][logger:188]:   Log Baud Rate: 115200
[16:22:29][C][logger:189]:   Hardware UART: UART0
[16:22:29][C][template.sensor:022]: Template Sensor 'Avg BMS Temp'
[16:22:29][C][template.sensor:022]:   State Class: ''
[16:22:29][C][template.sensor:022]:   Unit of Measurement: ''
[16:22:29][C][template.sensor:022]:   Accuracy Decimals: 1
[16:22:29][C][template.sensor:023]:   Update Interval: 10.0s
[16:22:29][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[16:22:29][C][uptime.sensor:033]:   Device Class: 'duration'
[16:22:29][C][uptime.sensor:033]:   State Class: 'total_increasing'
[16:22:29][C][uptime.sensor:033]:   Unit of Measurement: 's'
[16:22:29][C][uptime.sensor:033]:   Accuracy Decimals: 0
[16:22:29][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[16:22:29][C][uptime.sensor:034]:   Type: Seconds
[16:22:29][C][switch.gpio:068]: GPIO Switch 'heat_mat'
[16:22:29][C][switch.gpio:091]:   Restore Mode: always OFF
```

But I can't really see anything in the logs, the device just restarts.
I had already replaced the hardware on suspicion, a WT32-ETH01

here my YAML:

```
substitutions:
  name: jk-bms
  bms0: "${name}"
  bms1: "${name}1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  # Akku gross
  bms0_mac_address: 20:22:09:27:7c:44
  # Akku klein
  # mac_address: 20:22:08:25:0f:a9
  # Akku gross_2
[... 497 lines cut ...]
  - platform: ble_client
    ble_client_id: client1
    id: ble_client_switch1
    name: "${bms1} enable bluetooth connection"
        
  - platform: gpio
    name: "heat_mat"
    pin: GPIO14
    id: heat_mat

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"
  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"      
```
or do I have too many sensors or other errors?

syssi, 2024-10-30:
Could you reduce your YAML configuration step by step just for testing. You are using a lot of heavy components (`web_server`, `mqtt`, two instances of the `jk_bms_ble` component etc.). Does the situation improve if you free some resources? Is you custom code stable / free of memory leaks in general?

peff74, 2024-10-30:
I will follow up on your suggestion.
Firstly, I have now removed all my components and then deactivated the web server.
I now only have a debug component in there to check the heap

[image]

My code used very little heap, the web used much more, now we are at 100KB free heap.
Let's see how it goes
[image]

syssi, 2024-10-30:
You could free a lot of resources by changing the log level to `INFO` too.

peff74, 2024-10-30:
[image]
something eats the heap and sometimes it comes to a bang

syssi, 2024-10-30:
Good job!

peff74, 2024-10-30:
[image]

unfortunately there were more new starts despite loglevel INFO :-(
Where else could I start to find out what is causing the problem?

which is clear, it starts with a significant heap loss

syssi, 2024-10-30:
Please remove one of the BLE client connections as next step.

peff74, 2024-10-30:
done
[image]

Now we got 114kb free heap.
And the largest free block size has increased quite a bit to 64kb

peff74, 2024-10-31:
[image]
the view this morning.
A reboot around 10pm since then it has been running continuously
[image]
Here is a zoom into the crash at 10pm.
Something seems to be stuck, once he makes it, then the second time it's over

syssi, 2024-10-31:
Could you post your current YAML configuration? I would like to get an idea which components are still used.

syssi, 2024-10-31:
Some next steps:

1. Remove all `jk_bms_ble` entities from the YAML but don't remove the `ble_client` connection. We want still be able to connect to the BMS but we want to skip the data decoding.
2. Remove the `mqtt` component and add the `api` or `web_server` component to check the uptime from time to time.
3. Remove the complete BLE communication but add a number of template sensors which publish some random states periodically to generate some work/noise.

The idea is to identify the BLE stack or the network communication as root cause.

peff74, 2024-10-31:
this is then the YAML if I understand Step1 correctly?

```
substitutions:
  name: jk-bms
  bms0: "${name}"
#  bms1: "${name}1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
# Akku gross
  bms0_mac_address: 20:22:09:27:7c:44
# Akku klein
# mac_address: 20:22:08:25:0f:a9
# Akku gross_2
# bms1_mac_address: 20:22:08:25:26:8b
# Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
# Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
# Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
#bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0
#  on_boot:
#    priority: 600
#    then:
#      - climate.control:
#          id: climate_heat_mat
#          mode: HEAT

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

#web_server:
#  local: false

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

ota:
  - platform: esphome
    password: ""
    on_begin:
      then:
        - switch.turn_off: ble_client_switch0
#        - switch.turn_off: ble_client_switch1
        - logger.log: "BLE connection suspended for OTA update"

logger:
#  level: DEBUG
[... 457 lines cut ...]

  - platform: gpio
    name: "heat_mat"
    pin: GPIO14
    id: heat_mat

text_sensor:
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms0
#    errors:
#      name: "${bms0} errors"
#    total_runtime_formatted:
#      name: "${bms0} total runtime formatted"
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms1
#    errors:
#      name: "${bms1} errors"
#    total_runtime_formatted:
#      name: "${bms1} total runtime formatted"
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
```

peff74, 2024-10-31:
uploaded
[image]

now we got 132kb free heap and the biggest block is ~90kb.
And very few changes to the heap.
I therefore now log every value and not just the changes

peff74, 2024-11-01:
[image]
almost 24 hours without anything...

What would be the next steps?

syssi, 2024-11-01:
Re-add a single BMS sensor f.e. total voltage.

peff74, 2024-11-01:
```
substitutions:
  name: jk-bms
  bms0: "${name}"
#  bms1: "${name}1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
# Akku gross
  bms0_mac_address: 20:22:09:27:7c:44
# Akku klein
# mac_address: 20:22:08:25:0f:a9
# Akku gross_2
# bms1_mac_address: 20:22:08:25:26:8b
# Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
# Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
# Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
#bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0
#  on_boot:
#    priority: 600
#    then:
#      - climate.control:
#          id: climate_heat_mat
#          mode: HEAT

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

#web_server:
#  local: false

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

ota:
  - platform: esphome
    password: ""
    on_begin:
      then:
        - switch.turn_off: ble_client_switch0
#        - switch.turn_off: ble_client_switch1
        - logger.log: "BLE connection suspended for OTA update"

logger:
#  level: DEBUG
  level: INFO

[... 459 lines cut ...]
    id: heat_mat

text_sensor:
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms0
#    errors:
#      name: "${bms0} errors"
#    total_runtime_formatted:
#      name: "${bms0} total runtime formatted"
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms1
#    errors:
#      name: "${bms1} errors"
#    total_runtime_formatted:
#      name: "${bms1} total runtime formatted"
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
```

done, but I took three sensor^^

now we got 130kb free heap and the biggest block is ~86kb.

peff74, 2024-11-02:
[image]
the current status.
I had to activate my thermo function yesterday because it was getting too cold.
Since I was unable to set the thermostat via MQTT, it always reset itself to the value I had set on the website, I also had to activate the web server.
Nevertheless, it seems to be stable.
So it seems to be the number of BMS sensors, or one of them has a problem?

syssi, 2024-11-02:
It's too early to make assumptions but we are on a good track.

peff74, 2024-11-02:
What should I do next to continue testing?
Can you tell me why the thermostat always gets the values that were last set on the website, even if the web server is no longer active?
Do you save the settings on the ESP?

syssi, 2024-11-02:
Let's stick to two sensors (per BMS) but add the second BMS again.

I'm on the road and cannot provide answers to your other questions ATM.

peff74, 2024-11-02:
I didn't get around to expanding it today.
The system as last described is now running 24 hours a day :-)

I think tomorrow I will add the second BMS again

syssi, 2024-11-02:
Please keep in mind, we have changed a lot of parameters and a lot of components are running faster now. It's possible the issue occurs on heavy load only.

peff74, 2024-11-02:
yes I understand, I think we will then add the components back in bit by bit and see what the problem is.
Right?

syssi, 2024-11-02:
Correct!

peff74, 2024-11-04:
This morning I finally got round to adding the second BMS.

[image]

now we got 102kb free heap and the biggest block is ~60kb.

You can now clearly see how the largest block of the heap fluctuates and when the BMS went online, the free memory dropped further.

What I also noticed is that something about ‘cell resistance’ first appeared in the logs, although I didn't even have it active.

peff74, 2024-11-05:
This config has now also been running for 24 hours.
First of all, it contains the most important data:
- Current
- Power
- Voltages of the individual cells
- Delta of the cells 
- Temperature

```
substitutions:
  name: jk-bms
  bms0: "${name}"
  bms1: "${name}1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
# Akku gross
  bms0_mac_address: 20:22:09:27:7c:44
# Akku klein
  bms1_mac_address: 20:22:08:25:0f:a9
# Akku gross_2
# bms1_mac_address: 20:22:08:25:26:8b
# Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
# Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
# Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0
  on_boot:
    priority: 600
    then:
      - climate.control:
          id: climate_heat_mat
          mode: HEAT

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

web_server:
  local: false

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

ota:
  - platform: esphome
    password: ""
    on_begin:
      then:
[... 447 lines cut ...]

  - platform: gpio
    name: "heat_mat"
    pin: GPIO14
    id: heat_mat

text_sensor:
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms0
#    errors:
#      name: "${bms0} errors"
#    total_runtime_formatted:
#      name: "${bms0} total runtime formatted"
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms1
#    errors:
#      name: "${bms1} errors"
#    total_runtime_formatted:
#      name: "${bms1} total runtime formatted"
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
´´´

syssi, 2024-11-05:
Is the used menory still constant or do you see peaks?

peff74, 2024-11-05:
[image]
the line is relatively smooth

[image]
but when I open the website, you can clearly see something

peff74, 2024-11-05:
[image]
I have now added the “cell resistance” to both BMS, i.e. 24 additional sensors.
The memory is shrinking more and more. 

Now we got 97kb free heap and the biggest block is ~55kb.

I wonder why adding the sensors eats up so much memory.
I only know Arduino.
If I take a look at how the data is processed there, then only the data stream that comes via Bluetooth is actually analyzed.
Then it doesn't really matter whether I pull 10 or 50 pieces of data information from the data stream, the data is already there anyway.

syssi, 2024-11-05:
Just a guess: The combination of `sensor` objects and the `mqtt` component is probably heavy because a topic must be assembled and measurements must be transfromed etc. You could remove `mqtt` once just for testing and looking the heat at the log to have a comparison.

peff74, 2024-11-05:
I have tested it out of curiosity.
That makes a clear difference!

[image]

Just a thought from someone who is not familiar with ESP Home...
Would it be possible to run only some sensors/settings through MQTT and make others only visible via the web browser.

syssi, 2024-11-05:
> Would it be possible to run only some sensors/settings through MQTT and make others only visible via the web browser.

Yes. You can declare sensors as `internal: true`. In this case they are passed to third parties. The `web_server` component has a special feature to show interal sensors nevertheless:

```
web_server:
  include_internal: true
```

peff74, 2024-11-05:
I have now set the cell resistance as `internal: true` and for the web server `include_internal: true`.
The result is no difference, except that no data arrives via MQTT^^

peff74, 2024-11-05:
by the way, here is the strange picture of the startup that I mentioned before... 
But this only happens sometimes
[image]

syssi, 2024-11-05:
Okay. In this case the implementation is not as efficient as hoped for.

syssi, 2024-11-05:
The screenshot above shows the log output of the settings frame:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L971-L1006

The frame is retrieved once on established connection and every time if you press the `retrieve settings frame` button.

peff74, 2024-11-05:
Well... what else can we do to get ahead?

syssi, 2024-11-05:
Just add new stuff to your YAML step by step and try to get an idea which piece causes the crash.

peff74, 2024-11-05:
O.K. I'll go on

peff74, 2024-11-05:
A question for understanding.
When I have the switches active.
```
switch:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    charging:
      internal: true
      name: "switch ${bms0} charging"
    discharging:
      internal: true
      name: "switch ${bms0} discharging"
    balancer:
      internal: true
      name: "switch ${bms0} balancing"
```
Then I don't need the binary sensors.
```
#binary_sensor:
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms0
#    balancing:
#      internal: true
#      name: "switch ${bms0} balancing"
#    charging:
#      internal: true
#      name: "switch ${bms0} charging"
#    discharging:
#      internal: true
#      name: "switch ${bms0} discharging"
#    online_status:
#      name: "switch ${bms0} online status"
```
Wouldn't that be double?
Because the switches display On/Off

Or are they linked to each other?

syssi, 2024-11-05:
There is a difference between `switch.balancing` is turned `ON` and the BMS is balancing at the moment indicated by `binary_sensor.balancing`. The same applies to `switch.charging`. If charging is enabled in general this doesn't mean the protection board allows charging at the moment (indicated by `binary_sensor.charging`).

peff74, 2024-11-06:
adding the switches seems to have been too much. or something is wrong with the switches.
I will now leave the switches in and remove the ‘cell resistance’.
[image]
Let's see what happens

What always catches my eye is the message of the long loop time of the esp32_ble ~200ms is really a lot. I know this from my Arduino projects, where I split up functions to run them in different loops to reduce the runtime as much as possible.
An ESP32 is not as sensitive as an ESP8266, but it's just a thought

peff74, 2024-11-06:
Perhaps it makes sense to split up the Bluetooth query.
The important data more frequently and the unimportant data less frequently.

syssi, 2024-11-06:
This is a good idea in general but let's check first at some point blocking is really an issue: We will reduce the number of sensor to three pieces again (per BMS) and add a blocking sleep of ~200ms to the loop.

Please keep in mind: The BMS pushs the complete status frame two times per second. We have to discard the traffic because the interval is fix and cannot be changed.

peff74, 2024-11-06:
O.K. I didn't know that, I thought you had to ask the BMS
I haven't looked at the scripts yet, because I know that if I start I'll build my own Arduino version, or make the one from the ‘Akkudoktor’ forum work ^^

At the moment I'm running a version as written above with `switch`: and `binary_sensor:` and less `sensor:`

syssi, 2024-11-06:
How did you retrieve this log:

```
[16:22:17][D][sensor:094]: 'Avg BMS Temp': Sending state 17.05000  with 1 decimals of accuracy
[16:22:17][D][climate:396]: 'heat_mat' - Sending state:
[16:22:17][D][climate:399]:   Mode: HEAT
-----------------------------REBOOT--------------------------------
[16:22:28][W][component:170]: Component mqtt cleared Warning flag
[16:22:28][I][mqtt:286]: MQTT Connected!
[16:22:29][I][app:062]: setup() finished successfully!
```

Via network? Did you try to retrieve the log using the serial line / the USB connection? You should see the exception + stack trace here which will provide some more insights probably.

peff74, 2024-11-06:
> How did you retrieve this log:
>
> ```
> [16:22:17][D][sensor:094]: 'Avg BMS Temp': Sending state 17.05000  with 1 decimals of accuracy
> [16:22:17][D][climate:396]: 'heat_mat' - Sending state: [...]

I received the logs via the network.
Because the ESP runs outside in a small cabinet, it's a bit difficult to connect it serially.
But if it becomes necessary, I could build it.

To get exception and stack trace logger
  `level: DEBUG` 
must be active?
 I don't know ESP-Home very well.

peff74, 2024-11-06:
> I had a similar issue during this year. ESP restarted several times randomly (?). I started commenting sensors/switches but nothing. At least, I created a custom component "switch" (JkRS485BmsSwitch) and a custom component "number" (JkRS485BmsNumber) and everything went OK (with all sensors uncommented).
> 
> I cannot undestand why the problem dissapeared using custom components...
> 
> Now, I have a similar issue with sensors. I have tried to create a new custom component "sensor" (JkRS485BmsSensor) but I have not got it to know if this way the problem goes away for ever. :-(

`JkRS485BmsSwitch`
Sounds like you are connected over cable, I use BLE.
I am not sure if this is comparable.

peff74, 2024-11-06:
Stable so far
[image]

txubelaxu, 2024-11-06:
> `JkRS485BmsSwitch` Sounds like you are connected over cable, I use BLE. I am not sure if this is comparable.

Yes, of course, they are different communication mechanisms, but the cause of the problem could be the same. 
I read your comments and I thought that it, maybe, could be useful for you.

peff74, 2024-11-06:
> Yes, of course, they are different communication mechanisms, but the cause of the problem could be the same. I read your comments and I thought that it, maybe, could be useful for you.

Sure, of course it could be :-)

peff74, 2024-11-07:
The last version without the 24 `cell_resistance` sensors, but with `binary_sensor: `and  `switch:` runs stable.
No peaks, no crashes. The only peaks come from opening the web interface. 
The theory of the long runtime of the loop can also be ruled out.
In the current version we are also at over 300ms

[image]

[image]

peff74, 2024-11-19:
Yes, I've played a few things back and forth again.
I just don't have the time to document it here.
The version below is at least stable.

As soon as more sensors are added, the system becomes unstable.
Unfortunately, I have no idea how to get this under control :-(

What do you think we could do?

```
substitutions:
  name: jk-bms
  bms0: "${name}"
  bms1: "${name}1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
# Akku gross
  bms0_mac_address: 20:22:09:27:7c:44
# Akku klein
  bms1_mac_address: 20:22:08:25:0f:a9
# Akku gross_2
# bms1_mac_address: 20:22:08:25:26:8b
# Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
# Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
# Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0
  on_boot:
    priority: 600
    then:
      - climate.control:
          id: climate_heat_mat
          mode: HEAT

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

web_server:
  local: false
  include_internal: true

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

ota:
  - platform: esphome
    password: ""
[... 502 lines cut ...]

  - platform: gpio
    name: "heat_mat"
    pin: GPIO14
    id: heat_mat

text_sensor:
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms0
#    errors:
#      name: "${bms0} errors"
#    total_runtime_formatted:
#      name: "${bms0} total runtime formatted"
#  - platform: jk_bms_ble
#    jk_bms_ble_id: bms1
#    errors:
#      name: "${bms1} errors"
#    total_runtime_formatted:
#      name: "${bms1} total runtime formatted"
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
```

peff74, 2024-11-19:
what I have just noticed. 
There was a crash of the ESP when I had to reboot my Linux machine and the MQTT service was not available.

peff74, 2024-11-30:
It really seems to be the case that if the MQTT is not accessible, the system crashes.
Today again, because I booted the MQTT server for updates

syssi, 2024-12-25:
Let's close this issue because the situation cannot be improved here.

closed 2024-12-25 by syssi