syssi/esphome-jk-bms #628: NEEY balancer connetivity via Blutetooth proxy..

jeybee74, 2024-11-14:
Hi,
I have a few Shelly devices that can be use as Bluetooth proxy to HA. IE Shelly smart Plug S plus.

Have anyone used the Shelly devices to access NEEY active balancer or JK-BMS or other devices via this Bluetooth proxy to HA?

Some hints to get this working would be great, thanks.

BR
Jon Braathen

syssi, 2024-11-14:
Is the location of your battery bank fixed? Why are you looking for "roaming"?

jeybee74, 2024-11-14:
> Is the location of your battery bank fixed? Why are you looking for "roaming"?

My batterybank is located in a concrete garage, I want to monitor my NEEY balancer from my phone outside the BLE range.
This should be possible with the BLE proxy function I think. :)

syssi, 2024-11-14:
It looks like your understanding of the component isn't correct. The BLE proxy is used normally if you want to decode BLE traffic using Home Assistant. The proxy forwards the RAW BLE traffic to Home Assistant.

This project decodes the BLE traffic at the ESP32 already and ships the decoded measurements to Home Assistant. In other words: If you use ESPHome on your Shelly, just add this component and decode the traffic there.

closed 2024-11-14 by syssi