syssi/esphome-jk-bms #629: New `JK-B2A8S20P HW: V11A SW: V11.54` not working from `GPS UART-TTL` port

Sleeper85, 2024-11-14:
Hi @syssi,

I have next to me two BMS `JK-B2A8S20P`, the first one works well **(HW: V11.XW SW: V11.37)** but the second one with a different architecture (new?) does not work **(HW: V11A SW: V11.54)**.

There is a `UART` menu that allows you to choose between several protocols but none of them work.

The default address of the device is `1` I changed it to `0` like on the other BMS that works but it does not change anything.

I specify that it is a new BMS but **not the JK-PB inverter BMS**.

Here are some pictures of the options in the JK app.

[image]
[image]
[image]
[image]

Sleeper85, 2024-11-14:
Should I really open the BMS?

[image]
[image]
[image]

syssi, 2024-11-14:
I don't want to encourage you to void the warranty. ;-) Don't break the seal.

Sleeper85, 2024-11-14:
[image]

Sleeper85, 2024-11-14:
The software also behaves differently in the same way as the JK-PB series.

* OVPR should be `<` than SOC 100%
* UVPR should be `>` than SOC 0%

This seems logical but it is different from `V11.37` anyway.

[image]

Sleeper85, 2024-11-14:
[I will try to read the info from the display port](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-display-example.yaml]

Sleeper85, 2024-11-14:
[Looks like this issue](https://github.com/Louisvdw/dbus-serialbattery/issues/1045)

Sleeper85, 2024-11-15:
@syssi I can retrieve the info from the RS485 display port.

Sleeper85, 2024-11-16:
After testing the display port, I did a second try from the `GPS` port and it works now.

I don't know what happened, was it a bad contact on my prototype...

Clarification:

The UART protocol used is **000 - 4G-GPS**
Device Addr. can be **0** or **1** (default) it has no effect in this case

So you can add this new **HW: V11A** and **SW: V11.54** to the list of supported BMS.

closed 2024-11-16 by Sleeper85

JohanAlvedal, 2025-05-02:
I have the same HW and can't get it to work via Bluetooth — is it not supported?