syssi/esphome-jk-bms #631: help wanted

lorenzo6201, 2024-12-03:
We are a Romanian company specializing in assembling LFP batteries, primarily utilizing the B2A24S20P BMS. Currently, we are encountering a challenge in configuring 80 BMS units consecutively. Performing this task via the Android app is both time-consuming and prone to human error.

Our factory management application already stores the required parameters in a database, and we aim to develop an automated solution for configuring the BMS via Bluetooth. Although we have explored available online documentation, our attempts have not been successful.

As a proof of concept, we are using an ESP32 preloaded with the necessary parameters to connect to the BMS and push the configuration. However, the parameters are neither saved nor modified within the BMS, and we do not hear the usual confirmation beep associated with the Android app configuration process.

We would greatly appreciate assistance from an engineer to resolve this issue. For inquiries, please contact us via WhatsApp at +40 726 591 154 or email.

Thank you for your support.

syssi, 2024-12-25:
Do you still need some support?

lorenzo6201, 2024-12-25:
I do

lorenzo6201, 2024-12-25:
We managed to push the settings , but we need few more things. 
First thing would be changing the bms bluetooth name and the settings password.

The password is generated by our management app and is a random set of caracters tied to the serial number given to the product.
Same thing applies to the name , is generated by software and usual contains the last six caracters of the serial number. 

 Currently the name and password are changed manualy and that is taking way to much time. Also having a mixed firmware where we could conect via BLE or serial for the PB series would be fantastic. We could suport this financialy if  necesary.

syssi, 2024-12-25:
Could you make more explicit which features do you miss?

lorenzo6201, 2024-12-25:
Changing bms Bluetooth name and settings password.

lorenzo6201, 2024-12-25:
> Could you make more explicit which features do you miss?

I would apreciate if we discuss it over whatsapp. On github I miss the notifications.

supriik, 2024-12-25:
Review BMS Documentation
Ensure you have the complete technical documentation for the BMS. This typically includes command structures, data formats, and specific Bluetooth services/characteristics used by the device. If the documentation is insufficient, I recommend contacting the BMS manufacturer for additional guidance.

2. Analyze Bluetooth Communication
It’s crucial to understand how the Android app communicates with the BMS. Use tools such as nRF Connect or similar BLE sniffers to observe the communication between the Android app and the BMS. This will help you replicate the sequence of commands and data structure required for successful configuration.

3. Debug the ESP32 Code
Confirm that the ESP32 is correctly configured for BLE communication.
Ensure the commands are formatted and sent in the correct order. Any deviation can cause the BMS to ignore the configuration.
Test with minimal configuration steps first to verify that the connection is working and data is being sent correctly.
4. Check for Command Confirmation
The missing confirmation beep suggests that the configuration process is not being successfully finalized. Verify if the BMS requires a specific acknowledgment (e.g., an “ACK” response or a closing command) after receiving parameters.

5. Enable Logging and Debugging
Log all BLE communication attempts to identify any issues in real time. Compare the log with the communication pattern from the Android app to pinpoint discrepancies.

lorenzo6201, 2024-12-26:
Easier said that done. I have no ideea how to do all that.