syssi/esphome-jk-bms #632: Only 1 bms info in "esp32-ble-v14-multiple.yaml"

trung729, 2024-12-05:
I used above code for my 2bms, but it only has 138 entities and display 1 bms infomation only. please help me to resolve. Thanks

trung729, 2024-12-05:
[image]

trung729, 2024-12-06:
[logs_jk_logs.txt](https://github.com/user-attachments/files/18039624/logs_jk_logs.txt)

syssi, 2024-12-06:
The example configuration needs to be extended for additional devices. What's the modbus address of the second BMS?

trung729, 2024-12-07:
I connect to BMS via bluetooth, all 2 MAC Add I have put to yaml file

syssi, 2024-12-07:
Oh, you are using BLE and you picked the correct examples for multiple devices. It looks like I wasnt awake yesterday.

syssi, 2024-12-07:
I've identified the root cause and will provide a fixed version of the YAML soon!

syssi, 2024-12-07:
I've pushed an updated YAML example. Please start from scratch! :-)