syssi/esphome-jk-bms #636: Settings are changing by itself

informatuss, 2024-12-09:
Few settings are changing by itself after few hours. In log there is no indication why.

[image]
[image]
[image]

syssi, 2024-12-09:
Could you describe your setup in a few words? Which YAML configuration do you use?

janas-karol, 2024-12-09:
> Could you describe your setup in a few words? Which YAML configuration do you use?

I'm using ESP32 via bluetooth. Used esp32-ble-example.yaml. 
Everything is working great (thx for you work btw ;)) but these setting keep changing to default. I'm not connecting phone with app or anything, it looks random to me.

syssi, 2024-12-25:
Could you enable the `VERY_VERBOSE` output of the component by:

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: INFO
    jk_bms_ble: VERY_VERBOSE
    scheduler: INFO
    component: INFO
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

I would like to compare the random value with the RAW payload of the BLE notification.

sunnysingh-84, 2025-05-25:
I had the same issue, in my case, it turned out to be that the cell voltages I was specifying for 0% and 100% soc were being rejected by the BMS even though home assistant was showing them as being successfully updated. 

The reason for the cell voltages for 0% and 100% soc being rejected was due to the following logic:

- **cell voltage undervoltage recovery** needs to be greater than **cell soc0 voltage**
- **cell voltage overvoltage recovery** needs to be less than **cell soc100 voltage**

Not sure if its possible for the esphome code to notify user when they are applying incorrect values?