syssi/esphome-jk-bms #638: Fix CONF_BALANCING (Balancing action) device class.

I-Love-47, 2024-12-16:
Home Assistant (2024.12.3) reported the following warning:

`Entity sensor.dc_ups_jk_bms_balancing (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22`

I can see that the device_class is indeed set to DEVICE_CLASS_CURRENT in [sensor.py:218](https://github.com/syssi/esphome-jk-bms/blob/5b2754590945a2d709d31c37405f96f78ae3106a/components/jk_bms_ble/sensor.py#L218).

I believe it is reporting the "Balancing action" from [this code](https://github.com/syssi/esphome-jk-bms/blob/5b2754590945a2d709d31c37405f96f78ae3106a/components/jk_bms_ble/jk_bms_ble.cpp#L512)?

So, it should be `DEVICE_CLASS_EMPTY`?

Also, there is a related issue opened about renaming the field to `balancer status`: https://github.com/syssi/esphome-jk-bms/issues/579

I-Love-47, 2024-12-16:
Sure, I'll prepare a PR.

syssi, 2024-12-16:
Thanks for your contribution. Please recompile your setup and give it a try!

closed 2024-12-16 by syssi

I-Love-47, 2024-12-16:
I edited my .yaml in esphome and changed the   external_components_source to my branch.
This worked well without any errors in Home Assistant.

I-Love-47, 2024-12-16:
And same for merged changes on the original repo now.