syssi/esphome-jk-bms #640: Update Neey/EnerKey EK-24S4EB balancer support Version ZH-2.x

svergata, 2024-12-18:
Hello, 
I bought some new Versions of NEEY EK-24S4EB
Product Model: HW-3.2.0
Software Version: ZH-2.0.8
Production Date: 20240920

With this Version the detection an Readings from the Balancer  does not nespond correctly and is not working.
No Device is found for quite log time. 
Sometimes the device is found but no connect occurs.

Here some Debug Output.

`[21:51:49][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[21:51:49][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[21:51:49][VV][esp32_ble_tracker:434]: Parse Result:
[21:51:49][VV][esp32_ble_tracker:451]:   Address: E3:36:4C:03:2E:97 (RANDOM)
[21:51:49][VV][esp32_ble_tracker:453]:   RSSI: -76
[21:51:49][VV][esp32_ble_tracker:454]:   Name: ''
[21:51:49][VV][esp32_ble_tracker:462]:   Ad Flag: 6
[21:51:49][VV][esp32_ble_tracker:477]:   Manufacturer ID: 0x02E1, data: 10.02.89.A3.02.C2.22.BD.14.F6.F8.EA.7B.F2.90.2E.10.0B.EF.C9.59.02.DE (23)
[21:51:49][VV][esp32_ble_tracker:486]:   Adv data: 02.01.06.1A.FF.E1.02.10.02.89.A3.02.C2.22.BD.14.F6.F8.EA.7B.F2.90.2E.10.0B.EF.C9.59.02.DE (30)
[21:51:49][VV][esp32_ble_tracker:434]: Parse Result:
[21:51:49][VV][esp32_ble_tracker:451]:   Address: 3C:A5:51:9C:11:0C (PUBLIC)
[21:51:49][VV][esp32_ble_tracker:453]:   RSSI: -67
[21:51:49][VV][esp32_ble_tracker:454]:   Name: ''
[21:51:49][VV][esp32_ble_tracker:462]:   Ad Flag: 6
[21:51:49][VV][esp32_ble_tracker:465]:   Service UUID: 0xFFE0
[21:51:49][VV][esp32_ble_tracker:465]:   Service UUID: 0xFEE7
[21:51:49][VV][esp32_ble_tracker:477]:   Manufacturer ID: 0x301C, data: 88.A0.3C.A5.51.9C.11.0C (8)
[21:51:49][VV][esp32_ble_tracker:486]:   Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.1C.30.88.A0.3C.A5.51.9C.11.0C (21)
[21:51:49][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] Found device
[21:51:49][D][esp32_ble_tracker:674]: Found device 3C:A5:51:9C:11:0C RSSI=-67
[21:51:49][D][esp32_ble_tracker:695]:   Address Type: PUBLIC
[21:51:49][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[21:51:49][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[21:51:49][V][esp32_ble:331]: (BLE) gap_event_handler - 18
[21:51:49][V][esp32_ble:331]: (BLE) gap_event_handler - 18
[21:51:49][I][esp32_ble_client:067]: [0] [3C:A5:51:9C:11:0C] 0x00 Attempting BLE connection
[21:51:53][VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=672886 (now=677886)
[21:51:53][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[21:51:58][V][mqtt.idf:121]: Event dispatched from event loop event_id=7
[21:51:58][V][mqtt.idf:124]: MQTT_EVENT_BEFORE_CONNECT`

would appreciate to help in getting it running.

syssi, 2024-12-19:
Is this your first balancer? Could you provide some details about your ESP32 module?

svergata, 2024-12-19:
Yes my first one with BLE. 

It's an AZ-Delivery ESP32 Wroom-32

syssi, 2024-12-25:
Are you sure `3C:A5:51:9C:11:0C` is the MAC address of your BMS? This MAC address is advertised without a BLE name which is uncommon. Could you try to use this YAML to find out the MAC address of your device?

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-heltec-balancer-ble-scanner.yaml

svergata, 2024-12-26:
Hi, 
yes im pretty sure that the MAC addresses are right. 
Sadly both Balancers only Send some time their name via BLE.
Here the Output of the scanner, and a screenshot from the Neey App.

```
[08:47:17][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495269,"address":"3C:A5:51:9C:14:EB","rssi":-62,"name":""}'               <-------NEEY-B01
[08:47:17][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495269,"address":"E3:FE:1A:B0:0B:5E","rssi":-87,"name":"MPPT 450/100"}'
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495269,"address":"3C:A5:51:9C:11:0C","rssi":-62,"name":""}'               <-------NEEY-B02
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495269,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":""}'
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495269,"address":"E3:FE:1A:B0:0B:5E","rssi":-90,"name":"MPPT 450/100"}'
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495270,"address":"E3:FE:1A:B0:0B:5E","rssi":-91,"name":""}'
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495270,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":"SmartShunt "}'
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495270,"address":"3C:A5:51:9C:14:EB","rssi":-57,"name":"Neey-B02'         <-------NEEY-B02
[08:47:18][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495270,"address":"E3:36:4C:03:2E:97","rssi":-74,"name":"SmartShunt"}'
[08:47:19][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495270,"address":"3C:A5:51:9C:11:0C","rssi":-83,"name":""}'               <-------NEEY-B01
[08:47:19][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495271,"address":"E3:FE:1A:B0:0B:5E","rssi":-85,"name":""}'
[08:47:19][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495271,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":"SmartShunt"}'
[08:47:19][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495271,"address":"67:2C:62:37:CE:C5","rssi":-68,"name":""}'
[08:47:19][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495271,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":"SmartShunt"}'
[08:47:20][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495272,"address":"E3:FE:1A:B0:0B:5E","rssi":-90,"name":"MPPT 450/100"}'
[08:47:20][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495272,"address":"3C:A5:51:9C:14:EB","rssi":-58,"name":""}'               <-------NEEY-B02
[08:47:20][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495272,"address":"E3:FE:1A:B0:0B:5E","rssi":-87,"name":"MPPT 450/100"}'
[08:47:21][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495272,"address":"E3:36:4C:03:2E:97","rssi":-74,"name":"SmartShunt"}'
[08:47:21][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495273,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":"SmartShunt"}'
[08:47:22][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495274,"address":"E3:FE:1A:B0:0B:5E","rssi":-86,"name":""}'
[08:47:22][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495274,"address":"E3:36:4C:03:2E:97","rssi":-75,"name":"SmartShunt"}'
[08:47:22][D][text_sensor:064]: 'BLE Devices Scanner': Sending state '{"timestamp":4578295495274,"address":"3C:A5:51:9C:14:EB","rssi":-58,"name":""}'               <-------NEEY-B02
```

[image]

syssi, 2024-12-26:
Could you provide another `VERY_VERBOSE` log like above using the `esp32-heltec-balancer-ble-example.yaml` again. I'm interested in some log messages indicating the advertised services and characteristics (which gets printed to the log):

https://github.com/syssi/esphome-jk-bms/blob/main/components/heltec_balancer_ble/heltec_balancer_ble.cpp#L181-L192

svergata, 2024-12-26:
Find attached the generated Debug

[logs_jk-bms_run-2.txt](https://github.com/user-attachments/files/18251369/logs_jk-bms_run-2.txt)

svergata, 2025-01-26:
Did it help ?

syssi, 2025-01-27:
I've extracted the important parts from the log for further analysis:

```
[09:52:05][VV][esp-idf:000][mqtt_task]: W (26134) mqtt_client: Connection refused, not authorized
[09:52:05][VV][esp-idf:000][mqtt_task]: E (26137) mqtt_client: MQTT connect failed
[...]
[09:52:07][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[09:52:12][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[09:52:17][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=41 gattc_if=3
[09:52:17][D][esp32_ble_client:171]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_DISCONNECT_EVT, reason 256
[09:52:17][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 2
[09:52:17][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=2 gattc_if=3
[09:52:17][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_OPEN_EVT
[09:52:17][W][esp32_ble_client:142]: [0] [3C:A5:51:9C:11:0C] Connection failed, status=133
[...]
[09:52:21][VV][esp32_ble_tracker:434]: Parse Result:
[09:52:21][VV][esp32_ble_tracker:453]:   Address: 3C:A5:51:9C:11:0C (PUBLIC)
[09:52:21][VV][esp32_ble_tracker:456]:   RSSI: -50
[09:52:21][VV][esp32_ble_tracker:457]:   Name: ''
[09:52:21][VV][esp32_ble_tracker:465]:   Ad Flag: 6
[09:52:21][VV][esp32_ble_tracker:468]:   Service UUID: 0xFFE0
[09:52:21][VV][esp32_ble_tracker:468]:   Service UUID: 0xFEE7
[09:52:21][VV][esp32_ble_tracker:479]:   Manufacturer ID: 0x301C, data: 88.A0.3C.A5.51.9C.11.0C (8)
[09:52:21][VV][esp32_ble_tracker:489]:   Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.1C.30.88.A0.3C.A5.51.9C.11.0C (21)
[09:52:21][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] Found device
[...]
[09:52:23][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[09:52:27][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[09:52:32][W][heltec_balancer_ble:226]: [3C:A5:51:9C:11:0C] Not connected
[...]
[09:52:51][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 41
[09:52:51][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=41 gattc_if=3
[09:52:51][D][esp32_ble_client:171]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_DISCONNECT_EVT, reason 256
[09:52:51][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 2
[09:52:51][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=2 gattc_if=3
[09:52:52][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_OPEN_EVT
[09:52:52][W][esp32_ble_client:142]: [0] [3C:A5:51:9C:11:0C] Connection failed, status=133
[..]
[09:53:24][V][esp32_ble:348]: (BLE) gap_event_handler - 18
[09:53:24][I][esp32_ble_client:066]: [0] [3C:A5:51:9C:11:0C] 0x00 Attempting BLE connection
[09:53:24][V][esp32_ble:348]: (BLE) gap_event_handler - 18
[09:53:25][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 40
[09:53:25][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=40 gattc_if=3
[09:53:25][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_CONNECT_EVT
[09:53:25][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 2
[09:53:25][V][esp32_ble_client:120]: [0] [3C:A5:51:9C:11:0C] gattc_event_handler: event=2 gattc_if=3
[09:53:25][D][esp32_ble_client:110]: [0] [3C:A5:51:9C:11:0C] ESP_GATTC_OPEN_EVT
[09:53:25][D][esp32_ble_tracker:273]: Starting scan...
[09:53:25][VV][scheduler:032]: set_timeout(name='scan', timeout=600000)
[09:53:25][V][esp32_ble:348]: (BLE) gap_event_handler - 2
[09:53:25][V][esp32_ble:348]: (BLE) gap_event_handler - 7
[09:53:25][V][esp32_ble:348]: (BLE) gap_event_handler - 20
[09:53:25][V][esp32_ble:348]: (BLE) gap_event_handler - 21
[09:53:26][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[09:53:26][V][esp32_ble:348]: (BLE) gap_event_handler - 3

```

syssi, 2025-01-27:
Could you try to provde a btsnoop capture of the BLE communication between your Android phone and the Balancer?

See https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

NAP141, 2025-02-01:
I have the same balancer with the same identification and would very much appreciate as well to make it work.  The older NEEY balancers have Device Name "GW-24S4EB", now it is "EK-24S4EB", I ordered mine on Mar 3, 2024
```
[00:06:00][D][esp32_ble_tracker:273]: Starting scan...
[00:06:00][W][heltec_balancer_ble:238]: [2B:80:03:F5:7A:21] Not connected
[00:06:00][D][esp32_ble_client:110]: [0] [2B:80:03:F5:7A:21] Found device
[00:06:00][D][esp32_ble_tracker:677]: Found device 2B:80:03:F5:7A:21 RSSI=-76
[00:06:00][D][esp32_ble_tracker:698]:   Address Type: PUBLIC
[00:06:00][D][esp32_ble_tracker:700]:   Name: 'EK-24S4EB'
[00:06:00][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[00:06:00][I][esp32_ble_client:066]: [0] [2B:80:03:F5:7A:21] 0x00 Attempting BLE connection
[00:06:02][D][esp32_ble_client:171]: [0] [2B:80:03:F5:7A:21] ESP_GATTC_DISCONNECT_EVT, reason 62
[00:06:02][D][esp32_ble_client:110]: [0] [2B:80:03:F5:7A:21] ESP_GATTC_OPEN_EVT
[00:06:02][W][esp32_ble_client:142]: [0] [2B:80:03:F5:7A:21] Connection failed, status=133
```

syssi, 2025-02-01:
I would like to support the device but a btsnoop capture is required here.

NAP141, 2025-02-01:
Thanks for the quick response and your willingness to support. The device I manage is far away from me. Until recently I had an old android tablet next to it which I was remoting in, but the tablet died. This is why I am trying esphome which i have there. I will try to have a friend do it but this could take a while. Hope @svergata can help.

NAP141, 2025-02-01:
I have zero knowledge of BT communication but have seen this been mentioned that a service explorer py script from bleak:
https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py
is helpful. If this helps here is the response: 
```
python service_explore.py --address 2B:80:03:F5:7A:21
2025-02-01 23:30:52,777 __main__ INFO: starting scan...
2025-02-01 23:30:55,831 __main__ INFO: connecting to device...
2025-02-01 23:31:01,554 __main__ INFO: connected
2025-02-01 23:31:01,554 __main__ INFO: [Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
2025-02-01 23:31:01,581 __main__ INFO:   [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read), Value: bytearray(b'EK-24S4EB')
2025-02-01 23:31:01,608 __main__ INFO:   [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: bytearray(b'\x00\x00')
2025-02-01 23:31:01,636 __main__ INFO:   [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters (read), Value: bytearray(b'\x10\x00\x10\x00\x00\x00,\x01')
2025-02-01 23:31:01,636 __main__ INFO: [Service] 0000ffe0-0000-1000-8000-00805f9b34fb (Handle: 8): Vendor specific
2025-02-01 23:31:01,636 __main__ INFO:   [Characteristic] 0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 9): Vendor specific (write-without-response,write,notify), Max write w/o rsp size: 514
2025-02-01 23:31:01,664 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 11): Client Characteristic Configuration, Value: bytearray(b'\x01\x00')
2025-02-01 23:31:01,664 __main__ INFO:   [Characteristic] 0000ffe2-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific (write-without-response,write,notify), Max write w/o rsp size: 514
2025-02-01 23:31:01,691 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 14): Client Characteristic Configuration, Value: bytearray(b'\x01\x00')
2025-02-01 23:31:01,691 __main__ INFO:   [Characteristic] 0000ffe3-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (write-without-response,write), Max write w/o rsp size: 514
2025-02-01 23:31:01,691 __main__ INFO: disconnecting...
2025-02-01 23:31:04,822 __main__ INFO: disconnected
```

svergata, 2025-02-02:
Hi, 
Find attached an BTSNOOP.
Including 2 full cycles connect disconnect. 

[btsnoop_hci.log](https://github.com/user-attachments/files/18631797/btsnoop_hci.log)

syssi, 2025-02-02:
@svergata Thanks for the effort. Your capture contains all required details to get an idea what's going on.

First of all: The traffic hasn't changed and I'm able to decode the communication of the btsnoop capture:

```
[13:05:06][I][heltec_balancer_ble:695]: Device info frame (100 bytes):
[13:05:06][D][heltec_balancer_ble:696]:   55.AA.11.01.01.00.64.00.45.4B.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.33.2E.32.2E.30.5A.48.2D.32.2E.30.2E.38.56.31.2E.32.2E.37.33.00.32.30.32.34.30.39.32.30.03.00.00.00.64.5D.40.00.33.43.41.35.35.31.39.43.31.31.30.43.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.17.FF (100)
[13:05:06][I][heltec_balancer_ble:713]:   Model: EK-24S4EB
[13:05:06][I][heltec_balancer_ble:715]:   Hardware version: HW-3.2.0
[13:05:06][I][heltec_balancer_ble:717]:   Software version: ZH-2.0.8
[13:05:06][I][heltec_balancer_ble:719]:   Protocol version: V1.2.73
[13:05:06][I][heltec_balancer_ble:721]:   Manufacturing date: 20240920
[13:05:06][I][heltec_balancer_ble:723]:   Power on count: 3
[13:05:06][I][heltec_balancer_ble:725]:   Total runtime: 48d 19h (4218212s)
```

I guess the BLE module has changed and the connection setup is probably different. I will have a look soon!

syssi, 2025-02-02:
Bad news: Both hardware/software versions behave in exactly the same way.

NAP141, 2025-02-02:
What do you mean: you are finding no difference and we don't know why the connection fails? 
Any idea what we can do to try and help?

syssi, 2025-02-02:
In my opinion it should work out of the box. If it doesn't work the ESP32 should be replaced. A dual core ESP32 (ESP32-WROOM-32) is recommended.

NAP141, 2025-02-02:
This is what I am using: ESP32-WROOM-32S. At present I have ESPHome BLE-Proxy and HomeAssistant Integration from https://github.com/patman15/BMS_BLE-HA and successfully connecting to Daly BMS.

syssi, 2025-02-02:
@NAP141 Do you have access to the ESP32 and your Neey balancer at the moment?

NAP141, 2025-02-02:
Yes, remotely through ESPHome on Home Assistant running on RPI 4

syssi, 2025-02-02:
Please flash this YAML to your ESP and provide the complete ESPHome log: https://github.com/syssi/esphome-jk-bms/blob/neey-ek-24s4eb-support/esp32-heltec-balancer-ble-example.yaml

NAP141, 2025-02-02:
I am trying but not succeeding:
1. I changed ota and api sections to match my config this shouldn't be the problem
2. Removed the last section of the yaml about a interval / button as it seems you removed the buttons for now
3. I used before board: "esp32dev" you have "wemos_d1_mini32" but now neither compiles. I messed up something because neither works now I get an error while compiling on some api related stuff. I am trying to get to where I was and will try again

syssi, 2025-02-02:
Please try a **clean** build! The should be a "clean build files" button somewhere.

NAP141, 2025-02-02:
I did clean build and flashed it. from the yaml you send I changed:

removed this section:
```
interval:
  - interval: 30min
    then:
      - button.press: retrieve_settings_button0
      
```
replaced ota and api with this: 

```
api:
  encryption:
    key: "========my key=============="

ota:
  - platform: esphome
    password: "====my password==========="

```
same result: 
```
[22:30:09][D][esp32_ble_tracker:273]: Starting scan...
[22:30:09][V][esp32_ble:348]: (BLE) gap_event_handler - 2
[22:30:09][V][esp32_ble:348]: (BLE) gap_event_handler - 7
[22:30:09][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[22:30:09][VV][esp32_ble_tracker:434]: Parse Result:
[22:30:09][VV][esp32_ble_tracker:453]:   Address: 2B:80:03:F5:7A:21 (PUBLIC)
[22:30:09][VV][esp32_ble_tracker:456]:   RSSI: -74
[22:30:09][VV][esp32_ble_tracker:457]:   Name: 'EK-24S4EB'
[22:30:09][VV][esp32_ble_tracker:465]:   Ad Flag: 6
[22:30:09][VV][esp32_ble_tracker:468]:   Service UUID: 0xFFE0
[22:30:09][VV][esp32_ble_tracker:468]:   Service UUID: 0xFEE7
[22:30:09][VV][esp32_ble_tracker:489]:   Adv data: 02.01.06.0A.09.45.4B.2D.32.34.53.34.45.42.05.02.E0.FF.E7.FE.00 (21)
[22:30:09][VV][esp-idf:000][BTU_TASK]: W (85586) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:30:09][VV][esp-idf:000][BTU_TASK]: W (85697) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:30:09][VV][esp-idf:000][BTU_TASK]: W (85799) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:30:09][VV][esp-idf:000][BTU_TASK]: W (85888) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[22:30:09][VV][esp-idf:000][BTU_TASK]: W (85893) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:30:09][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 41
[22:30:09][V][esp32_ble_client:120]: [0] [2B:80:03:F5:7A:21] gattc_event_handler: event=41 gattc_if=3
[22:30:09][D][esp32_ble_client:171]: [0] [2B:80:03:F5:7A:21] ESP_GATTC_DISCONNECT_EVT, reason 62
[22:30:09][V][esp32_ble:386]: (BLE) gattc_event [esp_gatt_if: 3] - 2
[22:30:09][V][esp32_ble_client:120]: [0] [2B:80:03:F5:7A:21] gattc_event_handler: event=2 gattc_if=3
[22:30:09][D][esp32_ble_client:110]: [0] [2B:80:03:F5:7A:21] ESP_GATTC_OPEN_EVT
[22:30:09][W][esp32_ble_client:142]: [0] [2B:80:03:F5:7A:21] Connection failed, status=133
```

syssi, 2025-02-02:
Alright. I have to think about it. It's strange. If you have the chance (some day in future) please try different distances between the ESP32 and the Balancer.

NAP141, 2025-02-02:
Thank you for trying !!!

NAP141, 2025-02-14:
Happy to report, after moving the esp32 in different position (closer), and powering it with a stand alone power supply (was connected to a PC USB before): 
IT'S WORKING now with the original code from the main branch.

closed 2025-02-14 by syssi