syssi/esphome-jk-bms #642: only 2/4 JK BMS display runtime formatted

trung729, 2024-12-23:
I have add 2 more JK BMS into our system, and now I have total 4 JK BMS, but 2 new ones display wrong "runtime formatted" with this code:
 [esp32-ble-v14-multiple.yaml](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-multiple.yaml).
and I have added 2 more code with named "bms2" and "bms3" in your yaml file.
Please give me a help.

syssi, 2024-12-23:
Could you share your configuration instead of the initial one?

trung729, 2024-12-23:
here you are

trung729, 2024-12-23:
[jk.zip](https://github.com/user-attachments/files/18229219/jk.zip)
here you are

trung729, 2024-12-23:
[image]

trung729, 2024-12-23:
> Could you provide a log which indicates the issue? IMO the YAML looks fine. May be the lovelace card wasn't extended properly?

[image]

This information is wrong

trung729, 2024-12-23:
[image]

trung729, 2024-12-23:
[image]
[image]
[image]
[image]
bms2 and BMS3 donnot have "total runtime formatted"

trung729, 2024-12-23:
Cảm biến
_ bms0 average cell voltage
3,294 V
_ bms0 balancing
Tắt
_ bms0 balancing
0
_ bms0 balancing current
0,00 A
_ bms0 capacity remaining
222,823 Ah
_ bms0 cell resistance 1
0,060 Ω
_ bms0 cell resistance 10
0,058 Ω
_ bms0 cell resistance 11
0,059 Ω
_ bms0 cell resistance 12
0,054 Ω
_ bms0 cell resistance 13
0,056 Ω
_ bms0 cell resistance 14
0,052 Ω
_ bms0 cell resistance 15
0,055 Ω
_ bms0 cell resistance 16
0,053 Ω
_ bms0 cell resistance 2
0,056 Ω
_ bms0 cell resistance 3
0,059 Ω
_ bms0 cell resistance 4
0,058 Ω
_ bms0 cell resistance 5
0,065 Ω
_ bms0 cell resistance 6
0,065 Ω
_ bms0 cell resistance 7
0,069 Ω
_ bms0 cell resistance 8
0,066 Ω
_ bms0 cell resistance 9
0,066 Ω
_ bms0 cell voltage 1
3,294 V
_ bms0 cell voltage 10
3,295 V
_ bms0 cell voltage 11
3,294 V
_ bms0 cell voltage 12
3,295 V
_ bms0 cell voltage 13
3,294 V
_ bms0 cell voltage 14
3,295 V
_ bms0 cell voltage 15
3,294 V
_ bms0 cell voltage 16
3,294 V
_ bms0 cell voltage 2
3,293 V
_ bms0 cell voltage 3
3,295 V
_ bms0 cell voltage 4
3,294 V
_ bms0 cell voltage 5
3,295 V
_ bms0 cell voltage 6
3,293 V
_ bms0 cell voltage 7
[... 424 lines cut ...]
9
_ bms3 power
-378,81 W
_ bms3 power tube temperature
24,4 °C
_ bms3 state of charge
81%
_ bms3 temperature sensor 1
23,9 °C
_ bms3 temperature sensor 2
23,3 °C
_ bms3 temperature sensor 3
23,6 °C
_ bms3 temperature sensor 4
24,0 °C
_ bms3 temperature sensor 5
24,4 °C
_ bms3 total battery capacity setting
280 Ah
_ bms3 total charging cycle capacity
515,494 Ah
_ bms3 total runtime
439.314 s
_ bms3 total voltage
52,86 V

trung729, 2024-12-23:
I have shared my info in the topic, please take a look for me!

syssi, 2024-12-23:
Do you use the `api` component?

syssi, 2024-12-23:
Does it work if you use this reduced YAML configuration:

```
substitutions:
  name: "_"
  friendly_name: jk2
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  bms2: "${name} bms2"
  bms3: "${name} bms3"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:80:16:98:65 #51-280Ah-00-00
  bms1_mac_address: C8:47:80:16:90:DE #51-280Ah-01-01
  bms2_mac_address: 98:DA:20:09:E3:2E
  bms3_mac_address: 98:DA:20:06:27:C6
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S
  bms2_protocol_version: JK02_32S
  bms3_protocol_version: JK02_32S
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Allow provisioning Wi-Fi via serial
improv_serial:

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - switch.turn_off: ble_client_switch1
      - switch.turn_off: ble_client_switch2
      - switch.turn_off: ble_client_switch3
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  scan_parameters:
    active: true

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1
  - mac_address: ${bms2_mac_address}
    id: client2
  - mac_address: ${bms3_mac_address}
    id: client3

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    protocol_version: ${bms1_protocol_version}
    throttle: 5s
    id: bms1
  - ble_client_id: client2
    protocol_version: ${bms2_protocol_version}
    throttle: 5s
    id: bms2
  - ble_client_id: client3
    protocol_version: ${bms3_protocol_version}
    throttle: 5s
    id: bms3

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    total_voltage:
      name: "${bms0} total voltage"
    total_runtime:
      name: "${bms0} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    total_voltage:
      name: "${bms1} total voltage"
    total_runtime:
      name: "${bms1} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    total_voltage:
      name: "${bms2} total voltage"
    total_runtime:
      name: "${bms2} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms3
    total_voltage:
      name: "${bms3} total voltage"
    total_runtime:
      name: "${bms3} total runtime"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    errors:
      name: "${bms2} errors"
    total_runtime_formatted:
      name: "${bms2} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms3
    errors:
      name: "${bms3} errors"
    total_runtime_formatted:
      name: "${bms3} total runtime formatted"
```

trung729, 2024-12-24:
ok

> Does it work if you use this reduced YAML configuration:
>
> ```
> substitutions:
>   name: "_"
>   friendly_name: jk2
>   bms0: "${name} bms0"
>   bms1: "${name} bms1"
>   bms2: "${name} bms2"
>   bms3: "${name} [...]

ok let me check this configuration

trung729, 2024-12-27:
[logs_jk_run.txt](https://github.com/user-attachments/files/18261590/logs_jk_run.txt)
it display this error

syssi, 2024-12-28:
My fault. This is the fixed version:

```
substitutions:
  name: "_"
  friendly_name: jk2
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  bms2: "${name} bms2"
  bms3: "${name} bms3"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:80:16:98:65 #51-280Ah-00-00
  bms1_mac_address: C8:47:80:16:90:DE #51-280Ah-01-01
  bms2_mac_address: 98:DA:20:09:E3:2E
  bms3_mac_address: 98:DA:20:06:27:C6
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S
  bms2_protocol_version: JK02_32S
  bms3_protocol_version: JK02_32S
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Allow provisioning Wi-Fi via serial
improv_serial:

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  scan_parameters:
    active: true

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1
  - mac_address: ${bms2_mac_address}
    id: client2
  - mac_address: ${bms3_mac_address}
    id: client3

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    protocol_version: ${bms1_protocol_version}
    throttle: 5s
    id: bms1
  - ble_client_id: client2
    protocol_version: ${bms2_protocol_version}
    throttle: 5s
    id: bms2
  - ble_client_id: client3
    protocol_version: ${bms3_protocol_version}
    throttle: 5s
    id: bms3

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    total_voltage:
      name: "${bms0} total voltage"
    total_runtime:
      name: "${bms0} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    total_voltage:
      name: "${bms1} total voltage"
    total_runtime:
      name: "${bms1} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    total_voltage:
      name: "${bms2} total voltage"
    total_runtime:
      name: "${bms2} total runtime"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms3
    total_voltage:
      name: "${bms3} total voltage"
    total_runtime:
      name: "${bms3} total runtime"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    errors:
      name: "${bms2} errors"
    total_runtime_formatted:
      name: "${bms2} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms3
    errors:
      name: "${bms3} errors"
    total_runtime_formatted:
      name: "${bms3} total runtime formatted"
```

trung729, 2024-12-28:
[logs_jk_run.txt](https://github.com/user-attachments/files/18265610/logs_jk_run.txt)
another error

syssi, 2024-12-28:
Please try a `CLEAN` build. There is a button somewhere at the dashboard. If you use the CLI please try:

```
esphome clean yourconfig.yaml
esphome run yourconfig.yaml
```

trung729, 2024-12-28:
Please show me how to cleab build, I will try

syssi, 2024-12-28:
Do you use the ESPHome Add-on in Home Assistant?

syssi, 2024-12-28:
[image]

trung729, 2024-12-28:
> esphome_dashboard_clean_build_files.png (view on web)

Yes, I use esphome in hass

syssi, 2024-12-28:
Did you find the button?

trung729, 2024-12-28:
the 2 new ones are still unknown "total runtime"

syssi, 2024-12-28:
Could you provide a log please?

trung729, 2024-12-29:
It does not have entities: "bms2 total runtime formated" and "bms3 total runtime formated"

trung729, 2024-12-29:
[logs_jk_logs (1).txt](https://github.com/user-attachments/files/18270368/logs_jk_logs.1.txt)

syssi, 2024-12-29:
Please flash the reduced example above. Your log shows sensors which arent part of my YAML.

trung729, 2024-12-29:
I have flash the reduced code above, and the log fife I have just shared

syssi, 2024-12-29:
Did you upload the wrong log? These are a few lines from the mentioned log above:

```
[00:17:05][D][sensor:094]: '_ bms1 cell resistance 5': Sending state 0.06400 Ω with 3 decimals of accuracy
[00:17:05][D][sensor:094]: '_ bms1 cell voltage 6': Sending state 3.27500 V with 3 decimals of accuracy
[00:17:05][D][sensor:094]: '_ bms1 cell resistance 6': Sending state 0.06400 Ω with 3 decimals of accuracy
[00:17:05][D][sensor:094]: '_ bms1 cell voltage 7': Sending state 3.27900 V with 3 decimals of accuracy
[00:17:05][D][sensor:094]: '_ bms1 cell resistance 7': Sending state 0.06800 Ω with 3 decimals of accuracy
``` 

It includes cell voltages and resistance. 

The reduced version includes the runtimes only (this is important!): https://github.com/syssi/esphome-jk-bms/issues/642#issuecomment-2564269090

trung729, 2025-01-01:
I am surprised, too. I have flash your new code, and it still has that log

syssi, 2025-01-01:
Could you give it another try?

trung729, 2025-02-11:
> Any news?

I have change to YAMBMS which is also using your code, and it's fine
https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/BMS_JK-PB_RS485_solution.md

closed 2025-02-12 by syssi