syssi/esphome-jk-bms #646: jk BMS + inverter on 1 esp32 via uart

fastvd, 2024-12-29:
Greetings. I've already created a similar topic before, but now I can't find it.
the question remained the same, just then I created a topic and after a few days I left the place where the inverter+bms were...
now I have everything under my hands and I hope that with your help we will do it!

Output data:
1) the scheme is working now:
first board esp32 ---> uart ---> jikong bms + github://syssi/esphome-jk-bms@main

second board esp32 ---> uart --->rs232_to_ttl---> PowMr Hybrid Inverter + https://github.com/odya/esphome-powmr-hybrid-inverter

2)want us to use 1 esp32 board for inverter operation and bms operation at the same time.
i merged the code, it even compiled and it even works! but it works ONLY when some device (or inverter or bms) transmits data (I physically turn off one or the other)

the entire config is divided into packages (I took the idea from the project from the inverter, so the code is better read and easier not to get lost)

root file:
[root.txt](https://github.com/user-attachments/files/18269048/root.txt)
main file:
[main.txt](https://github.com/user-attachments/files/18269050/main.txt)
bms conf file:
[bms.txt](https://github.com/user-attachments/files/18269052/bms.txt)
inverter conf file:
[inv.txt](https://github.com/user-attachments/files/18269053/inv.txt)

fastvd, 2024-12-29:
the darts stick out specially so that it is possible to quickly remove/attach some element.
the esp32 board is powered now from the bms with the help of a step-down board 48v--->5v

[image]
[image]

fastvd, 2024-12-29:
I applied 5 volts from the lower board to the VIN input esp32 board
, and I applied 3.3 volts to the rs232_to_ttl board from contact 3V3 of the esp32 board

fastvd, 2024-12-29:
if only bms is connected, then the data comes normally, I see a normal picture in the logs!
[image]

syssi, 2024-12-29:
Could you describe your issue in detail?

fastvd, 2024-12-29:
[image]

fastvd, 2024-12-29:
the problem is that if the BMS + INVERTER are connected at the same time, then I get a Invalid Header error and data does not come from any device.
as soon as I disconnect GPIO1-2 or GPIO16-17, it works separately and so and so
[image]

fastvd, 2024-12-29:
[image]
[image]
[image]

and at this time, when the Invalid header is poured, the data does not enter the home assistant, then a few minutes may pass and some data will come in, and then they are not updated for 5-10 minutes, while they may be updated from the BMS, and not updated for the inverter, or on the contrary...but at the same time it does not work correctly

fastvd, 2024-12-29:
[image]
and it is interesting that SOMETIMES the data comes correctly at the same time, and at this moment the Invalid header error logs are really less... it seems that some buffer is overloaded, or somehow TOO MUCH data is coming from the inverter or from the BMS at one point in time ...somehow it is not clear...but the fact that physically everything is soldered well is 100%...this is either a matter of software or something diode\capacitor)

syssi, 2024-12-29:
Could you enable the `debug` output of the `uart` component and provide another log? It looks like there is a lot of noise on the signal line as soon both devices are connected.

syssi, 2024-12-29:
Hmm... it looks like the decoder can be improved too. Note for myself:

```
# A frame shifted by one cannot be decoded/identified
[13:22:30][W][jk_modbus:063]: Invalid header: 0x2B 0x4E
[13:22:30][W][jk_modbus:063]: Invalid header: 0x57 0x01
...
[13:23:51][W][jk_modbus:063]: Invalid header: 0x3D 0x4E
[13:23:51][W][jk_modbus:063]: Invalid header: 0x57 0x01
```

fastvd, 2024-12-29:
> Could you enable the `debug` output of the `uart` component and provide another log? It looks like there is a lot of noise on the signal line as soon both devices are connected.

how exactly to do this?
I understand that it's not just like that:
logger:
  level: DEBUG

syssi, 2024-12-29:
https://github.com/syssi/esphome-jk-bms?tab=readme-ov-file#debugging

fastvd, 2024-12-29:
[logs_bms-server-room_run.txt](https://github.com/user-attachments/files/18269854/logs_bms-server-room_run.txt)

syssi, 2024-12-29:
As next step pleas replace the content of the `bms.yaml` with these lines:

```
uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${tx_pin_uart_0}
    rx_pin: ${rx_pin_uart_0}
    stop_bits: 1
    debug:
      direction: BOTH
      dummy_receiver: true

sensor:
  - platform: wifi_signal
    name: "WiFi power PMeter"
    update_interval: 10s

  - platform: uptime
    name: "up"
    id: uptime_sec
```

Please provide another log.

fastvd, 2024-12-29:
[logs_bms-server-room_logs (2).txt](https://github.com/user-attachments/files/18269945/logs_bms-server-room_logs.2.txt)

syssi, 2024-12-29:
Okay. This looks good in general.

fastvd, 2024-12-29:
ok, and what should I do now?)

syssi, 2024-12-29:
I will prepare a feature branch to improve the decoder. This needs some minutes.

syssi, 2024-12-29:
Please use your initial `bms.yaml` again. This one:

```
uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${tx_pin_uart_0}
    rx_pin: ${rx_pin_uart_0}
    stop_bits: 1
    debug:
      direction: BOTH
#      dummy_receiver: false

external_components:
  - source: ${external_components_source}
    refresh: 0s
    
jk_modbus:
  - id: modbus0
    uart_id: uart_0
    rx_timeout: 50ms
jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
    update_interval: 10s

# ....
```

And change the `external_components_source` substitution to `improve-preamble-detection` at the root.yaml:

```
substitutions:
  name: powmr-inverter_bms
  device_name: "powmr-inverter_bms"
  device_description: "Monitor a JK-BMS(UART-TTL) + PowMR inverter in server room LANpro"
  
  node_name: "LANpro server room Inverter and BMS"
  node_id: powmr_inverter
  
  external_components_source: github://syssi/esphome-jk-bms@improve-preamble-detection
```

Flash your device another time and provide another log.

fastvd, 2024-12-29:
[logs_bms-server-room_logs (4).txt](https://github.com/user-attachments/files/18270145/logs_bms-server-room_logs.4.txt)

syssi, 2024-12-29:
It looks like the change resolves the issue.

fastvd, 2024-12-29:
yes, looks good. I confirm - the data comes from 2 devices at the same time. what was the problem if not a secret?

well, thank you very much and I think it makes sense to put this solution somewhere in the main section so that other people could find such a solution)

syssi, 2024-12-29:
The implementation reads the buffer of the `uart` and waits for a valid preamble (`0x4E 0x57`). If the buffer doesn't start with `0x4E 0x57` it gets flushed/discarded. In your case an incoming message did look like this:

```
0x2B 0x4E 0x57 0x01...
```

`0x2B 0x4E` isn't a valid preamble and `0x57 0x01` doesn't fit too. The improved implementation discards single bytes now. 

```
0x2B gets discard
0x4E is the first byte of a valid preamble
0x57 is the second byte -> yay! We have detected a valid preamble
```

closed 2024-12-29 by syssi

fastvd, 2025-01-09:
Greetings. I wanted to complete my project, but I see that it gives an error about the lack of new code, which we then managed to optimize for the operation of the multisystem.
[image]

syssi, 2025-01-09:
I've merged the fix into `main`. Please update the YAML from `@improve-preamble-detection` to `@main`.

fastvd, 2025-01-09:
something is wrong... I resoldered my 100% working board into a more presentable version and here I got an error that there is no github://syssi/esphome-jk-bms@improve-preamble-detection

I redid on github://syssi/esphome-jk-bms@main

and I have a problem:
1) if ONLY UART bms is connected, then everything is ok
2) if any other UART is connected, the data from both the BMS and the inverter will disappear after a few seconds
3) similarly - if there is only an inverter - everything is ok. I connect the BMS - there are no data either from there or from there!

log 2 uart:
[logs_bms-server-room_logs_2 UART.txt](https://github.com/user-attachments/files/18364706/logs_bms-server-room_logs_2.UART.txt)
only bms uart:
[logs_bms-server-room_logs._ONLY BMS.txt](https://github.com/user-attachments/files/18364708/logs_bms-server-room_logs._ONLY.BMS.txt)

and my new board:
[image]

syssi, 2025-01-09:
How do you power the ESP?

fastvd, 2025-01-09:
as before, I take power from the BMS itself ---> step-down DC-DC 48-5v

fastvd, 2025-01-09:
[image]

fastvd, 2025-01-09:
if I turn off the power supply + to the DC-DC input and power up the esp32 via type-c, the situation is the same!
[image]

fastvd, 2025-01-09:
any idea?

syssi, 2025-01-09:
My electricity skills are pretty limited. I guess the common ground between the three devices will be the root cause but I don't know how to proof it.

fastvd, 2025-01-10:
you did not understand, when there was the 1st version, where I did not solder everything neatly, then everything worked. bms+inverter worked. then I even added a 3rd UART for the PZEM...all 3 UARTs worked fine. and now that I moved it to +- normal state, not even 2 work! for test - can you return that separate code github://syssi/esphome-jk-bms@improve-preamble-detection ?
because the only thing I changed is the code!

syssi, 2025-01-10:
I've restored the branch but please keep in mind `main` and the feature branch are equal now. Please provide some more logs.

fastvd, 2025-01-10:
this 2 uart , when dont work our scheme
[2 uart.txt](https://github.com/user-attachments/files/18372629/2.uart.txt)
this is 2 uart, but without GND from inverter (I just unsoldered 1 contact on the fly )
[2 uart without gnd inverter.txt](https://github.com/user-attachments/files/18372665/2.uart.without.gnd.inverter.txt)

syssi, 2025-01-10:
Do you agree common ground could be the root cause?

fastvd, 2025-01-10:
I don't think so, since BEFORE I moved it to the new case, 3 common GNDs WORKED!
Now I'm checking again whether I correctly moved all the connections to the new case....I feel that the problem is not at the physical level!

fastvd, 2025-01-10:
and so...on a physical level I checked everything.
1) if I supply power from the BMS, but at the same time I disconnect RX from the BMS side, then I receive all data from the inverter correctly (GND at this moment is common, since we receive power to the esp32 board from the BMS)
[image]
[image]

syssi, 2025-01-10:
Please provide some more plaintext logs and explain what's attached per log. It's super hard to provide support with these few details.

fastvd, 2025-01-10:
2) in order for me to get the data from the BMS I had to disconnect all 4 wires from the inverter.
here is the log when ONLY the BMS is connected and when everything is fine!
[only rx tx bms.txt](https://github.com/user-attachments/files/18375188/only.rx.tx.bms.txt)

fastvd, 2025-01-10:
3) I connected only +3.3v from the esp32 board to the rs232 board (rx,tx,gnd are disconnected) and at this moment the data from the BMS does not come.
here is the LOG:
[bms-rx+tx, inv + wire.txt](https://github.com/user-attachments/files/18375243/bms-rx%2Btx.inv.%2B.wire.txt)

fastvd, 2025-01-10:
4) I connected only GND from the esp32 board to the rs232 board (rx,tx,+3.3v are disconnected) and at this point the data from the BMS ALSO does not come.
here is the LOG:
[bms tx+rx, inv GND.txt](https://github.com/user-attachments/files/18375280/bms.tx%2Brx.inv.GND.txt)

fastvd, 2025-01-10:
5) I connected only TX from the esp32 board to the rs232 board (tx,GND,+3.3v are disconnected) and at this moment the data from the BMS comes, BUT errors like:
[12:18:22][W][modbus:125]: Modbus CRC Check failed! 4040!=FFEF
[12:18:22][W][modbus:125]: Modbus CRC Check failed! B041!=FFFB
[12:18:23][W][modbus:125]: Modbus CRC Check failed! 4083!=FFFF
[12:18:23][W][modbus:125]: Modbus CRC Check failed! 254D!=FFF.
here is the LOG:
[bms tx+rx, inv TX.txt](https://github.com/user-attachments/files/18375349/bms.tx%2Brx.inv.TX.txt)

fastvd, 2025-01-10:
7. and the last option. I brought an additional step-down DC-DC to 3.3v and connected esp32 + rs232 in parallel. all TX,RX,GND are connected. data comes only from the inverter
here is the LOG:
[all rx+tx_3.3 bms.txt](https://github.com/user-attachments/files/18375597/all.rx%2Btx_3.3.bms.txt)

fastvd, 2025-01-10:
What's most interesting is that I didn't see any bad logs, except for option 5, when the inverter's TX was connected.

fastvd, 2025-01-10:
and now ATTENTION! mysticism and questions for ESP32 experts and radio engineers at heart! I looked at the photo of the old version in the old case again to check if I had moved all the wires to their places! and no, I made ONLY one minor rearrangement...in the old case - from DC-DC it was connected to esp32 to GND which is located near the 3.3V input...and on the new board, to make it look good, I connected GND near the VIN input...
I did it like in the old case and voila! there is data from 2 UARTs!!! isn't GND a COMMON contact in esp32 and they have a difference?
[image]
[image]
[image]

LOG:
[2 uart works.txt](https://github.com/user-attachments/files/18375867/2.uart.works.txt)

syssi, 2025-01-10:
You could use a continuity tester to verify a comman ground between all GND pins of your devboard.

fastvd, 2025-01-10:
i see what you're talking about and that's how i checked for a short....all the GND was ringing!

fastvd, 2025-01-10:
just connected the 3rd UART to the PZEM to still read volts, amps, watts at the input of my inverter.
 data from all 3 UARTs works.
I also added wiraguard to the code to remotely connect to my smart home all the data from my server room, where it will all be located, the VPN connects to my microti home router. the main file also contains a description of what to do on the Mikrotik side to raise the wireguard VPN server

bms-server-room.yaml
[bms-server-room.txt](https://github.com/user-attachments/files/18376247/bms-server-room.txt)

packges:
main.yaml
[main.txt](https://github.com/user-attachments/files/18376251/main.txt)

modules:
bms.yaml
[bms.txt](https://github.com/user-attachments/files/18376253/bms.txt)
inverter.yaml
[inverter.txt](https://github.com/user-attachments/files/18376259/inverter.txt)
pzem.yaml
[pzem.txt](https://github.com/user-attachments/files/18376277/pzem.txt)

helpers:
myHelpers.cpp:
[myHelpers.txt](https://github.com/user-attachments/files/18376304/myHelpers.txt)
myHelpers.h
[myHelpers (2).txt](https://github.com/user-attachments/files/18376307/myHelpers.2.txt)

anf final photo in new case:
[image]
[image]