syssi/esphome-jk-bms #652: Flaky UART-TTL communication after a while

syssi, 2025-01-01:
Same here :-)

Have 2 JK-BMSes and the 2 Year older JK-BMS run's great with the identical ESP8266 Hardware Layout. The Second JK-BMS that is about half a Year old shows the same Interruptions.

[image]

[image]

Also it comes and leaves, and when it's no longer sending Data also no Ping from the second BMS. It comes and runs or it don't run.

WiFi Signal is 99-100% and it is 50 cm away.

Strange. Had played with RX Puffer on the 115200 Baud Settings (Doubled it). Same Error.

The other first BMS with exact the same ESP Hardware Layout (DC-DC Converter, Filter Ceramic Capacitors, Fuse and ESP) runs 100% stable without Issues. The second problematic ESP: I also changed the ESP8266 and flashed it for using the second JK-BMS, same behaviour.

I also reduced the 24 Cells Setting to 8 Cells in the Script (So less Work for the ESP), didn't change also the Hickups.

Strange enough, it can run 1-2 Weeks flawlessy and then hicking up like above, or when hard reset (Pull it from the Socket) it worked 1-2 Minutes then Stall and second ESP Pull, then it runs and then above Hickups.

_Originally posted by @Christyria in https://github.com/syssi/esphome-jk-bms/issues/608#issuecomment-2567081230_

syssi, 2025-01-01:
Let's talk about your specific issue here. Could you describe your setup briefly? Is my understand correct: You have two independent battery setups. One is working without issue. The second on another ESP is flaky?

I've pushed an improvements a few days ago. Could you update your (flaky) setup and give it a try?

Christyria, 2025-01-01:
Thank you @syssi for Updating/Recreating my Posting.

Yes, i have two independent battery setups (in Paralell with Fuses) to deliver Power to my Inverter securely..
The first one (BMS 1) working without Problems. The second one (BMS 2) is flaky. Right.

I just update both ESP8266 with your new Improvements and i will see.

Thank you. 

Your ESP Software is absolutely great.

Christyria, 2025-01-01:
This is my older BMS 1:
[image]

This is my newer BMS 2 (About half a Year old):
[image]

Same JK Hardware Version on both but the newer JK BMS has a newer Software Revision with switchable RS485/RS232 Port.

syssi, 2025-01-01:
Could you provide a ESPHome log of the flaky setup?

Christyria, 2025-01-02:
Hi,

i also noticed on that flaky ESP that the Wireless Update is also way slower than the functional BMS...
Maybe a relly annoying Connection issue? Also hanging the ESP in the Air is also slower, ...

Now i test a setting in the WiFi Config Yaml:
output_power: 8.5

And change on my 2.4 GHz 20 MHz Bandwith from Channel 1 to 13 (Channel 1 is here empty , also Channel 13)

Let's see, ...

Because when the ESPHome Over The Air Update tooks way longer on the flaky BMS 2 and is fast on the functional BMS 1, ...

syssi, 2025-01-03:
Could you provide an ESPHome log? I would like to make sure it's an WiFi connection issue.

Christyria, 2025-01-03:
Thank you for your reply @syssi ,

Here is the original Logfile from my flaky second ESP BMS.

[logs_bde-jkbms-02_logs.txt](https://github.com/user-attachments/files/18302358/logs_bde-jkbms-02_logs.txt)

Greetings

syssi, 2025-01-03:
The log shows just 30 seconds. Could you try to record a log of around 15 minutes?

Christyria, 2025-01-03:
This is the flaky BMS ESP Log - But suddendly it has only a few Minutes. The last Minute written and when i downloaded the Log was + 20 Min (Without Log).

[logs_bde-jkbms-02_logs.txt](https://github.com/user-attachments/files/18304584/logs_bde-jkbms-02_logs.txt)

So the flaky ESP got his hickup, ... Will see when i could write more than a few Minutes out of it.

syssi, 2025-01-04:
Are you able to record the log via a serial line / USB connection? This will include alle connection resets or crashs.

Christyria, 2025-01-04:
Hi @syssi 

Your Support here (Response) is very great. Thanks for that.

I just connect the "flaky" BMS 2 ESP8266 to an Serial Line Output and that says that it suddendly stopped any output like an dead ESP with an Hardware error - like when the Power Supply was cutted out (And it's not)

So i soldered (ESD) a third new ESP8266 and flashed my ESPHome .BIN via Serial USB, delete the old ESP HA and added that new ESP to the HA and let's see, ..

[image]

I am thinking that the last two swapped ESD8266 from the Package are DOA so now the third next is for testing, ...

The joke is:

With the third ESP8266 - the OTA Update is as fast as with the ESP on the BMS 1 and the last flaky ESP was slower by Factor 1-2x - I used an Socket System for the ESP8266 to Hot Swap it out so the Hardware underneath (Power and Serial Connection / DC-DC Converter with Decoupling Caps) ans also the Place is the same (So that is identical for the running BMS 1)

Let's see, ...

syssi, 2025-01-04:
Looks good. Is the antenna part of the ESP module? A photo would be nice.

Christyria, 2025-01-05:
Hi @syssi ,

lets see. Now the Time will tell if the ESP is now 95% Stable.

The antenna is part of the ESP module. Yes.

**Here is BMS 1 (The fully without Problems working one):**
[image]

**And here is BMS 2 (The third (TM):**
[image]

Greetings

Christyria, 2025-01-06:
Hi,

look at this old Graph with the "flaky" BMS 2 ESP:
[image]

And this with the third BMS 2 ESP:
[image]

Let's see if the third ESP is now HW Stable :-)

Christyria, 2025-01-08:
[image]

looks very good now, ...

So the conclusion is: Two "New" out of the Foil Fail ESPs and the Third runs now ,...

Christyria, 2025-01-08:
So i think you can Close the Issue now.

How can i support your with a little coin for your Work ?
Thanks.

syssi, 2025-01-08:
Thanks for your support!

https://buymeacoffee.com/syssi
https://paypal.me/syssi84

closed 2025-01-08 by syssi