syssi/esphome-jk-bms #655: Incorrect data esp32 UART with JK v11.54

nazemnow, 2025-01-08:
I've got three jkbms connected to Home assistant.
Two of them are based on ESP8266 with a bit older HV/SV.
But the last one HV = V11U, SV = V11.54.
I tried to connect ESP32 to it, but I've got strange output.
I thought that there are not enough RAM and cut some not important data for me, but it didn't solve the problem. 
It's my first ESP32 based jk-bms device. 
I've used ESP32 as I need additional web interface (I've added web_server to yaml).
`RAM:   [=         ]  10.0% (used 32708 bytes from 327680 bytes)
Flash: [======    ]  55.4% (used 1016733 bytes from 1835008 bytes)`
I'm receiving next data using UART

```
[10:59:26][I][jk_bms:071]: Status frame received
[10:59:27][D][sensor:093]: 'jk-bms-esp32-template min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[10:59:27][D][sensor:093]: 'jk-bms-esp32-template max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template min voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template max voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template average cell voltage': Sending state nan V with 3 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 1': Sending state 1317.00000 °C with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 2': Sending state -4504.00000 °C with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template total voltage': Sending state 432.16998 V with 2 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template current': Sending state 0.00000 A with 2 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template power': Sending state 0.00000 W with 2 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template charging power': Sending state 0.00000 W with 2 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template capacity remaining': Sending state 0.00000 % with 0 decimals of accuracy
[10:59:28][D][sensor:093]: 'jk-bms-esp32-template temperature sensors': Sending state 0.00000  with 0 decimals of accuracy

[11:00:00][I][jk_bms:071]: Status frame received
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template min voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template max voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template average cell voltage': Sending state nan V with 3 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 1': Sending state 1317.00000 °C with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 2': Sending state 4200.00000 °C with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template total voltage': Sending state 532.47998 V with 2 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template current': Sending state 0.00000 A with 2 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template power': Sending state 0.00000 W with 2 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template charging power': Sending state 0.00000 W with 2 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template capacity remaining': Sending state 64.00000 % with 0 decimals of accuracy
[11:00:00][D][sensor:093]: 'jk-bms-esp32-template temperature sensors': Sending state 16.00000  with 0 decimals of accuracy
```

And the next one I could see in Home Assistant:
[image]

My yaml file:
```

substitutions:
  name: jk-bms-esp32-template
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: "JK BMS Remote Monitoring"
  friendly_name: jk-bms-esp32-template

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  use_address: 192.168.56.81

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jk-Bms-Esp32-Template"
    password: "hTkoxGsB0ltJ"

captive_portal:

#web_server:
#  port: 80
  
uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

jk_modbus:
  - id: modbus0
    uart_id: uart_0
    rx_timeout: 50ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
    update_interval: 15s

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "${name} balancing"
    balancing_switch:
      name: "${name} balancing switch"
    charging:
      name: "${name} charging"
    charging_switch:
      name: "${name} charging switch"
    discharging:
      name: "${name} discharging"
    discharging_switch:
      name: "${name} discharging switch"
    dedicated_charger_switch:
      name: "${name} dedicated charger switch"
    online_status:
      name: "${name} online status"

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"

    power_tube_temperature:
      name: "${name} power tube temperature"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    capacity_remaining:
      name: "${name} capacity remaining"
    capacity_remaining_derived:
      name: "${name} capacity remaining derived"
    temperature_sensors:
      name: "${name} temperature sensors"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    battery_strings:
      name: "${name} battery strings"
    errors_bitmask:
      name: "${name} errors bitmask"
    operation_mode_bitmask:
      name: "${name} operation mode bitmask"

    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    current_calibration:
      name: "${name} current calibration"
    device_address:
      name: "${name} device address"
    sleep_wait_time:
      name: "${name} sleep wait time"
    alarm_low_volume:
      name: "${name} alarm low volume"
    manufacturing_date:
      name: "${name} manufacturing date"
    total_runtime:
      name: "${name} total runtime"
#    start_current_calibration:
#      name: "${name} start current calibration"
    actual_battery_capacity:
      name: "${name} actual battery capacity"
#    protocol_version:
#      name: "${name} protocol version"

text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    battery_type:
      name: "${name} battery type"
    password:
      name: "${name} password"
    device_type:
      name: "${name} device type"
    software_version:
      name: "${name} software version"
    manufacturer:
      name: "${name} manufacturer"
    total_runtime_formatted:
      name: "${name} total runtime formatted" 
#  - platform: template
#    name: "TCP/IP Address"
#    lambda: 'return {WiFi.localIP().toString().c_str()};'
#    id: tcp_ip_address 
```

syssi, 2025-01-08:
Could you enable the `debug` output of the `uart` component and provide some raw traffic (from the log). I would like to reproduce the issue locally using the traffic captures.

nazemnow, 2025-01-08:
> Could you enable the `debug` output of the `uart` component and provide some raw traffic (from the log). I would like to reproduce the issue locally using the traffic captures.
Yes, sure. Is this log correct?
I've added `debug:` to the `-id: uart_0` section

```
[13:13:31][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:13:31][I][jk_bms:071]: Status frame received
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template min voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template max voltage cell': Sending state 0.00000  with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template average cell voltage': Sending state nan V with 3 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 1': Sending state 1317.00000 °C with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template temperature sensor 2': Sending state 15976.00000 °C with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template total voltage': Sending state 30.72000 V with 2 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template current': Sending state 0.00000 A with 2 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template power': Sending state 0.00000 W with 2 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template charging power': Sending state 0.00000 W with 2 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template capacity remaining': Sending state 252.00000 % with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template temperature sensors': Sending state 0.00000  with 0 decimals of accuracy
[13:13:31][D][sensor:093]: 'jk-bms-esp32-template charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template total charging cycle capacity': Sending state 3254533888.00000 Ah with 0 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template battery strings': Sending state 0.00000  with 0 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template errors bitmask': Sending state 65018.00000  with 0 decimals of accuracy
[13:13:32][D][text_sensor:064]: 'jk-bms-esp32-template errors': Sending state 'Power tube overtemperature;Discharging undervoltage;Battery over temperature;Charging overcurrent;Discharging overcurrent;Cell pressure difference;Overtemperature alarm in the battery box;Cell overvoltage;Cell undervoltage;309_A protection;309_A protection'
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template operation mode bitmask': Sending state 4248.00000  with 0 decimals of accuracy
[13:13:32][D][text_sensor:064]: 'jk-bms-esp32-template operation mode': Sending state 'Battery dropped'
[13:13:32][D][binary_sensor:036]: 'jk-bms-esp32-template balancing switch': Sending state ON
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template total battery capacity setting': Sending state 1523920.00000 Ah with 0 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template capacity remaining derived': Sending state 3840278.25000 Ah with 1 decimals of accuracy
[13:13:32][D][binary_sensor:036]: 'jk-bms-esp32-template charging switch': Sending state ON
[13:13:32][D][binary_sensor:036]: 'jk-bms-esp32-template discharging switch': Sending state ON
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template current calibration': Sending state 0.00000 A with 3 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template device address': Sending state 253.00000  with 0 decimals of accuracy
[13:13:32][D][text_sensor:064]: 'jk-bms-esp32-template battery type': Sending state 'Unknown'
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template sleep wait time': Sending state 39419.00000 s with 0 decimals of accuracy
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template alarm low volume': Sending state 168.00000 % with 0 decimals of accuracy
[13:13:32][D][text_sensor:064]: 'jk-bms-esp32-template password': Sending state '@\xd0'
[13:13:32][D][binary_sensor:036]: 'jk-bms-esp32-template dedicated charger switch': Sending state ON
[13:13:32][D][text_sensor:064]: 'jk-bms-esp32-template device type': Sending state '?ؙ\xfb?\xa8'
[13:13:32][D][sensor:093]: 'jk-bms-esp32-template total runtime': Sending state 3.26667 h with 0 decimals of accuracy
[13:13:45][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:13:46][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:60:00:00
[13:13:48][W][api.connection:103]: Home Assistant 2025.1.1 (192.168.56.11): Connection reset
[13:13:48][D][api:103]: Accepted 192.168.56.11
[13:13:48][W][component:237]: Component api took a long time for an operation (58 ms).
[13:13:48][W][component:238]: Components should block for at most 30 ms.
[13:13:48][D][api.connection:1446]: Home Assistant 2025.1.1 (192.168.56.11): Connected successfully
[13:13:50][W][api.connection:105]: Home Assistant 2025.1.1 (192.168.56.11): Connection closed
[13:13:50][D][api:103]: Accepted 192.168.56.11

```

syssi, 2025-01-08:
The response doesn't look like the GPS port / UART-TTL protocol. Which BMS model do you own?

nazemnow, 2025-01-08:
> The response doesn't look like the GPS port / UART-TTL protocol. Which BMS model do you own?

I got three B1A24S15P, last one HV v11u, SV v11.54

as I remember, 4 Pin JST connector couldn't be connected to another port. 

One thing, I could mix up Tx with Rx. When I swapped it in yaml, I got next  log output:

```
[14:20:28][W][jk_modbus:065]: Invalid header: 0x4E 0xDF
[14:20:28][D][uart_debug:114]: <<< FF:DB:DA:FF:FF:DF:FE:FF:DF:FB:FF:FF:DB:FF:FF:FF:FF:FE:FF:EF:FB:FD:DF:FF:FF:EF:DF:BF:DF:9F:DF:FF:FF:DF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:BF:DF:EF:DD:DF:9E:BF:BE:BF:DF:DF:DF:DF:9F:FF:DE:FF:DF:FB:DF:DF:FF:FF:FF:5B:FB:FF:FF:FB:DF:FF:FF:DF:DB:DB:DF:ED:9F:DA:DF:DF:FF:FF:BF:DD:DB:DF:DF:BF:DF:DF:FF:F7:F7:FD:FB:FF:FF:FB:FF:FF:FF:FF:FF:FB:FB:FF:95:D7:4F:DF:DF:9E:9E:FF:DF:DE:FF:EE:DD:FF:9E:DE:FF:DF:FF:EF:DF:FF:FF:FF:FF:FF:FF:FF:FB:BF:DB:DF:DB:5F:4E:DF:DF:DE:DF
[14:20:28][D][uart_debug:114]: <<< DF:DF:59:5F:DF:9F:DF:FF:DF:DF:F7:FF:FF:FF:DF:FF:FF:FF:FF:FF:DF:FA:FF:FF:DF:FF:9B:DA:FF:B6:FF:FF:FE:DF:AF:FB:FF:DF:DF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FB:FF:FF:59:FB:DF:BF:FB:FF:FF:FB:DF:FF:FF:FF:DB:FF:FF:FF:FF:FB:FF:FF:B2:FF:FF:FF:FF:FB:FA:FF:4F:FB:FF:FF:FF:5F:FA:FF:FF:FF:DB:DB:FF:FF:FF:FF:FB:FB:FF:FE:DF:AE:FD:FB:9F:FF:9F:FF:BE:FF:DF:DF:BF:FF:FF:FF:DB:FF:FF:FF:FF:FF:4D:DF:DA:DF:7F:FE:5F:DD:DB:DF:9F:DF:CF:FD:5F:F7:FD:DF:9F:DF:FF:FF:4B:FF:FF
[14:20:28][D][uart_debug:114]: <<< DB:FF:DF:F7:BF:FF:ED:DF:FF:FF:FF:FF:FF:DF:DF:DF:DF:DF:FF:FF:FF:FF:FD:DF:FF:FF:FB:6D:DB:BE:BF:FF:DF:CB:9F:BF:DF:9F:EF:BF:DF:7D:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:D6:EF:9F:DE:CF:DF:DF:DF:EF:DF:FF:9E:DE:BE:FF:DF:DD:DF:FF:FF:FF:FF:FB:FF:FF:FD:FF:FF:FB:DB:DF:92:DF:DB:DD:DF:BE:DF:BF:DF:59:5F:9F:9A:EE:FF:FF:FB:F7:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FB:FB:FF:FF:DB:DB:DB:FD:BA:FF:FB:FE:DB:FF:B7:FF:FF:FF:FF:FF:FF:FB:FE:DB:B6:DF:FF:BF:FB:FF:FF:FB:DB:DF:FF:BF
[14:20:28][D][uart_debug:114]: <<< FF:FF:FF:FF:B7:FF:FF:FF:FF:FB:4F:FF:FF:F3:9A:DF:BA:6D:DF:5B:DF:DF:DB:FE:DE:DF:DF:DB:FB:FF:FF:FF:FF:FF:FB:FF:FF:FF:DB:DB:FF:FF:D7:DD:FF:FF:FF:B2:DF:BF:9F:FA:DF:FF:9B:DF:DF:DF:9F:FF:FF:FF:FF:FF:FB:FB:FF:FF:FF:DF:FB:FB:FF:FF:DF:EF:DF:FF:D7:5B:FB:9A:FF:DF:FE:FF:F7:7F:DF:DF:9F:DF:FF:FF:FF:DF:FF:FF:C9:FF:FF:FF:FF:FF:FD:DB:FF:BA:9E:DB:FF:FB:DF:DF:CF:EF:FF:DF:FF:DF:DF:DF:FE:FF:DB:FF:FF:DF:B6:FA:DB:FB:5F:FF:FB:FB:FB:7B:FF:FF:FF:DF:FF:FF:DF:FF:BF:FE:FF:FF
[14:20:29][D][uart_debug:114]: <<< B6:FF:FF:FF:FF:FF:FF:FF:F2:B6:9E:D7:FF:DF:FF:FF:9E:FF:DB:DF:DE:FF:DD:FB:FF:FF:FF:FB:FF:FF:FF:FE:FF:FF:DD:DB:FB:E8:ED:4B:DB:FF:DF:6D:FF:FB:FF:DB:FF:FF:FF:FF:FB:FF:FF:6F:ED:DF:59:F9:FF:DF:DF:DF:FF:FF:7F:CF:DF:EF:DF:FF:DF:FF:FF:FF:F3:DF:FB:FF:FF:FB:F7:FF:BF:DF:BF:DE:9E:FF:DF:B7:F7:DF:DB:DF:9E:9E:FF:FF:FF:FF:DF:DF:DF:FF:FF:FF:DF:FF:FF:FF:CB:BF:FF:BF:FF:FF:B7:FF:FF:FF:DB:F7:FF:CF:FF:DE:DF:EF:FF:FF:DF:F7:DF:DF:DF:FF:FF:FF:B6:FF:FF:FF:FF:FF:FF:FD:FF:EF
[14:20:29][D][uart_debug:114]: <<< FF:4B:FB:DF:DF:9F:DF:DF:DF:7F:FF:DD:DB:DF:DF:FF:FF:FF:DB:FF:FF:FF:FF:DF:FF:FF:FF:D3:92:FF:FF:FF:FB:7F:FF:FB:CF:FF:DF:DB:DF:FF:FF:FF:B2:FB:FF:DF:FF:FF:FF:FF:FF:FF:EF:FF:FB:FD:EF:D7:FF:EF:FD:FF:DF:DF:DB:DF:DF:EF:EF:DF:FF:DB:FF:FB:FF:FF:ED:FF:37:FF:FF:DF:FF:F7:FF:FF:B2:DF:FD:FB:FB:DF:FF:FF:FF:FF:DB:DF:FF:FF:FF:FF:DB:BF:FF:DF:5D:FF:DA:DB:FE:DF:FD:5D:DF:DF:DF:DE:FF:FF:FF:FF:FF:DF:FF:FF:DF:FF:FF:DF:FB:FB:FF:FF:FF:FB:FF:FF:FF:92:DF:BE:DF:FF:FF:BD:ED:FF
[14:20:29][D][uart_debug:114]: <<< B2:FF:DF:DF:DF:FF:FF:DB:FF:FF:FF:DB:FB:FF:FB:DF:FE:BF:F9:FF:7D:BF:FF:DB:DF:9E:FF:DF:FF:7F:DF:9B:DE:DF:9F:DF:EF:FF:DF:DF:CF:FB:FF:FF:FF:B6:FF:FF:9B:DB:DF:BF:BE:BD:DF:45:CF:9E:DF:FF:DF:DF:FF:DB:FF:9F:D7:DF:FF:FD:DF:FF:FF:FF:DF:DB:FF:FF:FF:FF:FF:DB:DF:FF:FF:CF:FF:7D:59:5B:CF:9E:9E:AE:FF:DF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FB:FF:FF:DF:FB:FF:FF:FE:FF:C9:FB:9F:BF:BF:49:5F:DE:9E:CF:FF:BE:DF:FF:DF:9F:FF:DF:FF:DB:FF:DB:FF:FF:FB:FB:DF:FF:AF:FF:DF:DB:DF:DF
[14:20:29][D][uart_debug:114]: <<< DF:F7:5F:5F:FF:DF:DF:FF:FD:DF:DB:FB:FF:FF:FF:FB:FF:DF:9F:9F:FE:DF:FF:4D:FF:DF:FF:FF:DF:DF:FE:DF:DF:FF:FF:FF:FB:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:9F:9B:FF:BB:DA:EF:DE:9E:9E:CF:C7:D9:DD:9F:DF:BF:FF:DF:DF:FF:FF:9B:FF:FE:FF:FF:DF:DF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FB:FF:DB:FF:DD:F7:DF:FF:FF:DB:FF:FF:FF:FF:FF:FF:FE:ED:FB:6F:9B:FF:DF:D3:CD:FF:FF:DF:FF:FD:FF:7F:FF:DB:FF:FD:DB:FF:FF:FF:FF:FF:FF:FF:FF:FB:F2:FF:FF:DF:7F:FF:5F:FF:DF:DF:DF:DB:FF:FF:FF:D7:FF:FF
[14:20:29][D][uart_debug:114]: <<< FB:F9:FF:FF:FF:FF:DB:FF:FF:FF:FF:FF:B7:FF:DF:DF:FB:BE:DF:FF:DF:DF:F9:FF:DB:FF:FF:BF:FF:FB:FF:FF:FF:FF:FF:DB:FF:FF:FF:FF:FF:DB:7F:FF:DF:BB:FD:FF:BF:3D:BF:DF:D7:B7:D3:FF:DF:FF:FF:7F:DB:FF:DF:FF:FF:DB:DB:BF:DB:FF:FF:FF:FF:FF:FD:FD:6F:BF:FF:DF:FF:DD:FF:9E:9F:FF:DF:DF:FF:EC:DF:BF:9E:FF:FF:DB:FF:FF:FF:DB:FF:DB:FF:FF:FB:DA:F2:DB:DF:97:FB:DD:EF:DF:DF:FF:FF:FF:D9:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:DF:FF:B7:CB:FF:FF:93:DF:FA:DF:9E:9F:DF:F7:DD:DB:BE:BF:DF:DF:FF
[14:20:29][D][uart_debug:114]: <<< DF:FF:DF:FF:FB:DB:FF:FB:FF:DF:FF:FF:FF:DB:BF:DF:FB:F3:9F:5C:BF:DF:96:FB:DB:BE:9E:FF:BF:5D:FF:BF:FF:FF:FF:DF:FF:FF:DB:FF:FF:DE:FF:FF:DF:97:B6:5F:DB:DF:DF:DF:DF:DF:BF:5F:DF:FF:DF:DF:9C:FF:FF:FB:FF:7F:FF:FF:DF:FF:FB:B6:DB:B6:9B:9A:2F:BE:BF:D7:DF:FF:DE:DA:DF:DF:FF:DD:FE:DF:DF:FF:FF:DB:FF:DF:FF:FF:FF:FB:FF:FE:FF:BF:FF:9F:DB:BE:F7:FF:DF:DF:DD:EF:DE:FF:DF:BE:FF:FF:FF:DB:FF:FF:FF:FF:6D:FA:DF:FF:DF:9F:DF:FE:FF:7D:DF:DD:DF:DB:9F:FF:DF:DF:EF:FF:FF:F7:FF:FF
[14:20:29][D][uart_debug:114]: <<< DB:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:DF:FF:9A:FF:EF:FF:DF:D7:7F:A7:DF:BF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:ED:BF:FF:EF:F6:DF:DE:9E:FF:BE:EF:FF:FF:5F:EF:BE:9F:DF:DF:FF:FF:FF:DF:DF:FF:FF:FF:FF:FB:FF:FF:FF:FF:6F:FF:FF:FB:FF:FF:DF:0F:D [... cut ...] DB:BF:FF:FF
[14:21:24][D][uart_debug:114]: <<< DB:DB:FB:93:FF:BF:FF:DF:FF:6F:DB:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BB:BF:FB:7D:BD:BE:FF:DF:DF:BF:FF:DC:FF:FF:DD:DF:DF:FE:DF:FF:DF:FF:DF:FF:FF:DB:FB:FF:5B:FF:DB:FF:2D:FF:FF:FF:9A:9F:FE:BF:FB:BE:DB:DF:FF:DF:FF:D3:FF:FF:FF:FF:FB:FF:FF:FF:F7:6D:DF:DB:BF:BF:FF:FF:79:59:D7:DF:FF:BE:FF:DF:FF:FF:DD:FF:BF:FF:FB:DB:F7:FF:FF:DF:FF:FF:FB:DF:7D:FF:FF:96:FD:DF:9F:9F:BF:DF:FF:DF:7F:DF:DB:DB:FF:FF:FF:FF:FF:FF:FF:DA:FF:FF:DF:FF:FF:FB:BF:3F:FF:FF:B7
```

syssi, 2025-01-08:
Could you verify all three wires are connected properly? The second capture with swapped pins looks like a missing GND connection.

nazemnow, 2025-01-08:
> Could you verify all three wires are connected properly? The second capture with swapped pins looks like a missing GND connection.

I'll check it at evening, thank you

nazemnow, 2025-01-09:
I've checked wiring, everything correct. 
And I've checked this ESP32 with my old BMS's - it could read data as it should be. 
So the problem between ESP32 - JK BMS HV v11u, SV v11.54 .
Now I got a lot of incoming spam packets, they are looks similar to correct ones. 
Adding some log from UART debugging.

```
[15:50:53][D][uart_debug:114]: <<< FB:FF:BF:DF:9F:DF:DF:EF:FF:FF:EF:DB:FF:FF:FF:FF:FF:FF:FF:FE:FF:DB:DF:4D:DF:BE:BF:9F:DC:DF:FF:97:9F:DB:BF:DE:FF:BF:FF:5B:DF:DF:DF:FF:FF:FF:FF:FF:FF:FB:DB:FF:FF:BF:FF:FF:CF:5F:DE:DF:BE:FF:FF:FF:7D:FD:FB:F7:FF:FF:EB:FF:FF:FF:FF:FF:DF:DF:DB:FF:FF:FF:FF:DB:DF:FF:D3:FD:FF:9F:8F:DF:BE:FF:DF:7D:DF:DF:9E:DF:FF:FF:FF:D6:FF:FF:FF:FF:FF:FF:FF:DB:DF:FF:BE:FF:DB:FF:FF:FD:F9:79:BE:9B:BF:FB:DB:9E:DF:1D:5E:EF:FF:5F:DF:FF:DE:FF:FF:FF:FF:FB:DB:FF:D3:FF:FF:FF:B6:FF
[15:50:53][W][jk_modbus:065]: Invalid header: 0x4E 0xDF
[15:50:53][D][uart_debug:114]: <<< FF:CF:DF:DF:5D:EF:CD:BE:BE:EF:CF:DF:D7:DF:FF:9F:DB:FF:FF:FB:DB:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:DF:9F:EF:96:D7:DF:DF:FF:EF:DF:BF:DF:D7:FF:DF:BF:FF:FE:FF:FF:DF:FF:DB:DB:FF:FB:B7:FF:FF:DB:DB:DB:DF:BF:EF:DF:DE:59:9E:FF:DF:3C:DF:DF:59:D9:DF:FF:FF:FF:FF:FF:DB:FB:FF:DB:DB:FF:FB:DB:FF:DB:DE:9E:9E:FF:DF:4E:DF:9E:9E:DF:9D:BE:B7:DF:DF:FF:FF:FF:FF:FF:FF:FB:FB:DB:FF:FF:FF:FB:5B:FF:DF:96:7D:FF:9B:DF:BE:BF:9F:FF:B7:FF:FF:FF:5B:FF:F7:FF:FD:FF:FF:BF:49:9E:9F:2D:5E
[15:50:53][D][uart_debug:114]: <<< ED:BA:DB:DF:DA:DF:ED:6F:FF:FF:DB:BB:BF:9B:FF:FF:FB:FF:FF:FF:FF:FB:FB:FF:6D:FF:FB:D7:FF:DF:DF:9E:FF:AE:4D:DF:BE:FE:DF:9F:BF:DE:3F:FD:FF:FF:DF:FF:BF:FF:FF:FF:FF:FB:FF:3B:FF:BF:BF:BF:BF:BB:DF:9F:DB:FF:7F:FF:FB:FF:FF:FF:FB:FF:B7:B7:FF:FF:FD:C9:FF:BF:FF:6D:BF:BB:DB:FB:DB:5E:FB:FB:DF:FF:FF:FF:DB:FF:FB:B7:FF:FF:FF:FF:B6:EF:FF:FF:96:DF:DB:FB:EF:DB:93:FD:DB:FE:FF:FF:FF:FF:BF:DB:FB:FF:FF:FF:FF:FF:FF:FF:FB:DF:DB:DF:FD:5B:ED:DA:DF:DF:DF:D7:5F:DF:DF:9E:BF:FF
[15:50:54][D][uart_debug:114]: <<< FF:FF:FF:FB:FF:FD:FB:FF:DF:BF:FF:FF:9F:DA:FF:DF:FB:FB:9A:4F:DB:FF:BF:B7:FF:BA:BF:DB:BF:FF:FB:7B:DB:FF:FF:FF:FF:FF:B7:DF:FF:FF:FB:DB:BF:DB:DB:DF:FE:FF:DB:FF:DE:DF:FF:DF:FB:BF:FF:FF:FF:79:FF:FF:FF:FB:FB:FB:BF:EF:FF:FA:FF:FB:DB:FF:6D:FF:DF:FB:7D:DF:ED:DF:FF:FF:FF:FF:FF:FF:FF:DB:FF:7D:BF:FB:FF:DB:DF:FF:7D:D6:5F:DD:9F:9F:DE:9F:BF:9E:EF:FF:DF:DD:DF:DE:FF:DF:DF:FF:FB:FB:FF:FF:FB:FF:B6:FF:FB:FB:F7:6E:FF:EF:ED:DA:DF:DF:DF:DF:FF:DF:DF:FF:DF:FF:FF:9F:FF:FA
[15:50:54][D][uart_debug:114]: <<< D3:9E:FD:BE:FF:FF:59:DF:DF:9F:FF:BE:BE:5D:DF:9F:FF:FF:DF:FB:5F:FB:FF:FF:FB:DF:EB:DF:9B:BF:FF:EE:BE:BF:FD:DD:DF:9E:9F:9E:DF:BE:DF:FD:DD:FF:DF:DE:FF:FF:FB:FF:FF:FF:FB:DF:FF:FF:FF:FF:FF:FB:FF:DF:9A:9F:BE:EF:DF:DF:DD:FB:7F:DF:BE:BE:BF:DF:DF:F7:FD:FF:DF:FB:FF:FB:FF:FF:FF:92:DB:DF:F3:FF:FF:EC:99:DB:DF:DF:6F:BF:DB:DF:EF:FF:FF:EB:F6:FF:FF:FF:DF:FB:FF:FD:DB:DB:FF:FA:FF:FF:FF:9A:FA:BF:BB:DF:DF:FF:BF:FD:BF:EF:FF:FF:7F:DB:FF:FF:FF:FF:FF:FF:EF:FB:FF:FF:97:EF
[15:50:54][D][uart_debug:114]: <<< FF:DF:B6:FD:9B:DB:BF:9F:FF:BF:FF:DE:FF:FF:B7:DB:FF:FF:FF:DB:FF:FF:FF:FF:FF:6B:DF:FF:FF:FF:FD:FB:6D:5B:9E:DA:BE:DF:0E:FF:D7:FF:DF:DB:BF:DE:9E:9F:EE:FF:FF:5F:DF:FF:FF:FF:B7:BF:FF:FF:EB:FF:B6:FE:FF:FF:DF:B6:92:DF:9A:BE:8F:DF:AF:AF:DF:DF:DF:9F:DE:EF:FF:FF:F7:FF:FF:DF:FF:FF:DB:DB:FF:FF:FF:FF:DF:FE:FD:FE:FF:FF:7E:FB:FA:6D:DF:9B:2D:FF:CF:DF:FF:DF:DD:6F:DF:D7:FF:FF:FF:FE:FF:DF:FF:DB:FF:FF:BF:FF:DF:FF:FF:FF:FF:4D:FF:EF:DF:DF:FF:FD:DF:DF:DB:FF:FE:DE:DF:9F
[15:50:54][D][uart_debug:114]: <<< DB:DF:FF:EF:BF:FF:FF:FF:FB:FF:DB:FF:FF:FF:FF:FF:6F:FF:DF:FF:BF:7D:9F:DF:DF:7D:FF:DB:DF:FF:FF:BF:9F:EF:FB:FB:FB:DF:FF:FE:FF:B7:FF:FF:BA:FB:F7:69:7F:69:FF:FF:7D:BA:DB:FB:FF:DF:DF:FB:DF:FF:FF:FF:DF:FF:DB:DB:FF:FF:FB:FB:DB:69:B6:DF:FF:FB:7F:FF:FF:DF:FF:B2:FF:DF:7F:FF:7F:FF:FF:ED:DF:DF:BA:DE:9E:BE:BF:FF:9D:FD:DF:9E:BF:FF:DF:BD:FF:D9:7F:DF:EF:BF:FF:FF:DF:FF:FF:DF:FF:FB:FF:DB:97:DE:97:1C:9F:DF:DF:FF:FB:9E:BF:EF:BC:DF:FF:DF:FF:DA:FF:FB:FF:DB:DB:FF:FF:FF
[15:50:54][D][uart_debug:114]: <<< DF:FD:FF:FD:FF:EE:DF:FF:7F:FF:9F:FF:FF:FB:7F:DF:FF:FB:FF:FF:FF:FB:4B:B6:FB:FF:FB:BE:9B:9B:FF:3E:DF:FE:7D:DF:DB:BF:9A:DF:F7:DF:DF:D9:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:CD:9F:FF:CF:DF:FF:FD:DF:DF:9F:DF:DE:FF:FF:FF:DF:DF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:5F:FF:DB:FB:FF:EF:DF:DB:DF:9F:9F:DA:DF:AE:DF:FF:DB:DF:DF:DE:FF:FF:FF:DF:DF:FF:FF:DF:FF:FF:FF:DF:FF:FB:DF:FF:FF:FF:FB:69:BF:DF:FF:D7:6F:BF:BF:DF:DF:D7:B7:FD:DF:9F:9F:ED:FF:FF:DF:DB:FF:FE:FF:FF:FB:B7:FF
[15:50:54][D][uart_debug:114]: <<< FF:FF:FF:FB:FF:FF:FF:FB:FB:E9:9A:FB:BF:DF:DF:FB:FF:FF:FB:FB:FF:FF:FF:FB:DB:FF:DB:DB:FE:FF:FF:FF:FB:FF:6D:9B:FF:EE:FF:CF:FF:FF:DF:FF:FF:DF:FF:FF:FF:DB:FF:DB:FF:FF:FF:FF:FB:FF:FF:FF:DB:FF:E5:FF:FF:FB:B7:9F:DB:FB:DF:FF:7D:FF:FF:DF:FB:FF:FF:FB:FF:FF:BF:DB:FF:FF:BF:7F:CD:4D:FF:BF:DF:DE:DE:EF:BF:BB:9F:FF:7E:FF:DF:FF:DF:FF:FF:FB:FD:F9:FF:FF:FF:FF:B6:FF:FA:FB:DB:FF:ED:DF:DE:6F:DB:FF:FD:FF:FF:FF:DB:FB:FF:FF:FF:FF:FF:FF:FF:DF:FF:7D:DF:FF:3D:DF:9F:9E:FF:9D
[15:50:54][D][uart_debug:114]: <<< BF:FF:D7:7D:7F:DF:FF:FF:FF:FF:FF:FF:DB:FF:FF:FF:FF:FF:FB:9E:D6:BD:DF:DF:FF:DF:BE:9F:9F:DF:DE:DF:FF:AE:D9:FD:DF:DF:DF:FF:DF:FF:BF:DB:FF:DF:FF:FF:FF:B6:DF:DF:FF:BE:DF:DF:7D:FF:DF:DF:9F:DF:DF:DF:DE:DF:DF:5D:DB:FF:DF:7D:FF:FB:FF:FA:F2:DB:B7:D3:DF:DF:BB:6F:6B:D3:B3:DF:FD:DB:DF:FF:FF:FF:DF:FF:FF:ED:FF:DB:FF:FF:FF:FF:FF:FB:DB:DF:D7:FD:9F:BF:BF:DF:7F:FF:FF:9F:FF:FE:9F:9F:DF:DF:AF:FF:BF:FF:DF:FF:FF:FB:FB:FF:FF:FF:DB:FF:FF:FF:6F:FF:FF:FF:D7:FD:FD:9F:9E:BF
[15:50:54][D][uart_debug:114]: <<< DF:DF:D6:F7:DF:9F:9E:BE:FE:BF:FF:7F:DB:DB:FF:FF:FF:FF:FB:DB:FF:FE:D2:9F:7D:FF:FD:9F:9B:9F:9E:BE:5F:DD:DD:DF:9F:DF:DF:FF:FF:DF:DF:DF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:DA:DB:BF:BE:7D:5D:FF:DA:9A:BE:1D:EF:FF:D6:59:7F:DF:9F:DE:FF:FF:FF:5F:FF:FB:BF [... cut ...]
[15:51:01][D][uart_debug:114]: <<< DD:4B:DF:DF:DF:FF:FF:FF:DF:FF:BB:FF:FF:FF:FF:B7:FF:FF:DB:FF:FB:FB:DB:FF:FF:DF:FF:EF:DF:DF:D7:CB:DF:7F:FF:DB:7F:FF:FF:FB:BF:FF:FF:FF:FF:FB:FF:FF:DB:D3:4D:EF:DB:B7:DB:BF:FB:FF:BF:BF:DF:FD:DF:F7:FF:FF:DB:FB:FF:BF:FB:DB:FF:49:FF:FB:9A:CF:CF:DD:FB:FF:5B:DD:DB:9E:DF:1D:F7:FF:F7:DF:EF:DF:DF:7D:FF:DD:FF:DF:FF:DB:BF:EF:FF:FF:FB:FF:FF:FB:BE:DE:DF:E7:59:FF:BA:EF:FF:BE:DE:BC:DF:DF:BF:BE:DF:DE:9F:FF:DF:F7:FF:DF:6D:FF:DB:FF:FF:DF:6D:FB:DB:69:7F:FB:F7:FF:FF:DF
[15:51:01][W][jk_modbus:065]: Invalid header: 0x4E 0xDF
```

And a screenshot of the frame, when I connected this ESP to my old BMS.
[image]
And I thied to change UART protocol Number - looks like nothing changed.

syssi, 2025-02-14:
Long story short: This isn't a software issue. There is a lot of noise on the wire.

closed 2025-02-15 by syssi