syssi/esphome-jk-bms #656: Хто може допомогти отримати дані з Jk-BMS по BLE + RPi5. Please HELP ME.

RondaYummy, 2025-01-08:
I get services and some only characteristics (About the BMS itself) but do not get data on battery power and the rest.
I am writing an application on nestjs + dbus-next.

Here is the code I wrote:
`import { Injectable, OnModuleInit, Logger } from '@nestjs/common';
import * as dbus from 'dbus-next';

@Injectable()
export class BluetoothService implements OnModuleInit {
  private readonly logger = new Logger(BluetoothService.name);
  private systemBus;
  private bluez;

  constructor() {
    try {
      this.systemBus = dbus.systemBus();
      if (!this.systemBus) {
        throw new Error('Failed to initialize systemBus');
      }
    } catch (error) {
      console.error('DBus initialization error:', error);
    }
  }

  async onModuleInit() {
    console.log('Initializing BlueZ interface...');
    try {
      const bluez = await this.systemBus.getProxyObject('org.bluez', '/');
      this.bluez = bluez.getInterface('org.freedesktop.DBus.ObjectManager');
      console.log('BlueZ interface initialized successfully');

      await this.connectToFirstDevice();
    } catch (error) {
      console.error('Failed to initialize BlueZ interface:', error);
    }
  }

  async connectToFirstDevice() {
    console.log('Listing devices...');
    try {
      const objects = await this.bluez.GetManagedObjects();
      const devices = Object.keys(objects).filter((path) =>
        path.includes('/org/bluez/hci0/dev_')
      );
      console.log('Discovered devices:', devices);

      if (devices.length === 0) {
        console.warn('No devices found.');
        return;
      }

      const devicePath = devices[0];
      this.log('Attempting to connect to the first device:', devicePath);

      const deviceProxy = await this.systemBus.getProxyObject('org.bluez', devicePath);
      const deviceInterface = deviceProxy.getInterface('org.bluez.Device1');

      this.log('Connecting to device using retry logic...', devicePath);
      await this.connectWithRetry(deviceInterface, devicePath, 3, 2000);
      this.log('Device connected successfully.', devicePath);
      await new Promise(resolve => setTimeout(resolve, 1000));

      this.log('Reading characteristics...', devicePath);
      await this.readDeviceCharacteristics(deviceProxy, objects);

      await this.readBatterySOC(deviceProxy, devicePath);
    } catch (error) {
      console.error('Failed to connect to device:', error);
    }
  }
[... 283 lines cut ...]
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service0014/char0015 value (UUID: 00002a29-0000-1000-8000-00805f9b34fb) | HEX: 42454B454E20534153, Int: 66, UTF-8: BEKEN SAS
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Service /org/bluez/hci0/dev_C8_47_80_12_41_99/service000e UUID: 0000ffe0-0000-1000-8000-00805f9b34fb
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Discovered characteristics for service /org/bluez/hci0/dev_C8_47_80_12_41_99/service000e: [
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19:   '/org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char0011',
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19:   '/org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char0011/desc0013',
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19:   '/org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char000f'
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: ]
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Inspecting characteristic: /org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char0011, UUID: 0000ffe1-0000-1000-8000-00805f9b34fb, Flags: write-without-response,write,notify
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Subscribed to notifications for characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char0011
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Inspecting characteristic: /org/bluez/hci0/dev_C8_47_80_12_41_99/service000e/char000f, UUID: 0000ffe2-0000-1000-8000-00805f9b34fb, Flags: write-without-response
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Service /org/bluez/hci0/dev_C8_47_80_12_41_99/service000a UUID: 00001801-0000-1000-8000-00805f9b34fb
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Discovered characteristics for service /org/bluez/hci0/dev_C8_47_80_12_41_99/service000a: [
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19:   '/org/bluez/hci0/dev_C8_47_80_12_41_99/service000a/char000b',
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19:   '/org/bluez/hci0/dev_C8_47_80_12_41_99/service000a/char000b/desc000d'
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: ]
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Inspecting characteristic: /org/bluez/hci0/dev_C8_47_80_12_41_99/service000a/char000b, UUID: 00002a05-0000-1000-8000-00805f9b34fb, Flags: read,indicate
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service000a/char000b value (UUID: 00002a05-0000-1000-8000-00805f9b34fb) | HEX: 0100FFFF, Int: 1, UTF-8: ��
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Skipping characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service002b/char0030/desc0033 as it lacks GattCharacteristic1 interface.
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Skipping characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service002b/char0030/desc0032 as it lacks GattCharacteristic1 interface.
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Skipping characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service002b/char002c/desc002f as it lacks GattCharacteristic1 interface.
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: Skipping characteristic /org/bluez/hci0/dev_C8_47_80_12_41_99/service002b/char002c/desc002e as it lacks GattCharacteristic1 interface.
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: [Nest] 26061  - 08/01/2025, 12:35:19     LOG [BluetoothService] [Andrii 3] Reading battery level...
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: [Nest] 26061  - 08/01/2025, 12:35:19     LOG [NestApplication] Nest application successfully started +22ms
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: [Nest] 26061  - 08/01/2025, 12:35:19     LOG [Bootstrap] 🟢 Application is running on: http://[::1]:3000
0|⛅️ Solar Monitor  | 2025-01-08T12:35:19: [Nest] 26061  - 08/01/2025, 12:35:19     LOG [BluetoothService] [Andrii 3] Battery SOC: 0%`

syssi, 2025-01-08:
Could you provide some details about your BMS model? The implemented request (`0xDD, 0xA5, 0x03, 0x00, 0xFF, 0xFD, 0x77`) is from a pretty outdated protocol version / one of the early JK-BMS models (v4?).

RondaYummy, 2025-01-08:
> Could you provide some details about your BMS model? The implemented request (`0xDD, 0xA5, 0x03, 0x00, 0xFF, 0xFD, 0x77`) is from a pretty outdated protocol version / one of the early JK-BMS models (v4?).

How can I do this?
The data I am receiving now about the BMS from the characteristic 00002a25-0000-1000-8000-00805f9b34fb:
BEKEN SAS
BK-BLE-1.0
1.0.0.0.0-LE
1.0.0
6.1.2
6.3.0

I don't really understand what is related to what.
I'm sorry for asking such questions, I've just started to get interested in this topic and it's a bit complicated for me now.

syssi, 2025-01-08:
Are we talking about anm BMS or Active Balancer? Which Android app do you use to talk to your device?

RondaYummy, 2025-01-08:
> Ми говоримо про anm BMS чи Active Balancer? Яку програму Android ви використовуєте для спілкування зі своїм пристроєм?
It is simply called "JK BMS".

https://play.google.com/store/search?q=jk-bms&c=apps&hl=uk

RondaYummy, 2025-01-08:
I would also be grateful if there is some kind of manual on the Internet on how to work with these JK-BMS, receive notifications, and know where and what data it provides. I've searched for a long time and haven't found anything that could help me understand this concept.
If it's helpful, here's a gitHub, I'm using the dbuss.service.ts file because I couldn't get any characteristics at all through @abandonware/noble. Now I get at least the developer information. I will be sincerely grateful for any help, I want to make a status notification on the DBMS for my own use.

syssi, 2025-01-08:
There is no official documentation or protocol description available. It looks like you have implemented the wrong protocol / an unsupported request. Please have a look at the `jk_bms_ble/jk_bms_ble.cpp`.

RondaYummy, 2025-01-08:
> Офіційної документації чи опису протоколу немає. Схоже, ви запровадили неправильний протокол/непідтримуваний запит. Будь ласка, перегляньте `jk_bms_ble/jk_bms_ble.cpp`.

Unfortunately, I am not familiar with C++.
It is difficult to understand how everything is implemented there. And what activation code is used in new versions of JK-BMS?

syssi, 2025-01-08:
I will summarize some protocol details tomorrow.

syssi, 2025-01-09:
# JK BMS BLE Protocol Documentation

## BLE Service & Characteristic

The BMS is accessible via the following BLE service:
- Service UUID: `0xFFE0`
- Write Characteristic UUID: `0xFFE1` (handle `0x03`) - Used to send commands
- Notification Characteristic UUID: `0xFFE1` (handle `0x05`) - Used to receive notifications

**Note:** While both characteristics share the same UUID (`0xFFE1`), they have different handles and purposes:
- Commands are written to handle `0x03`
- Notifications are received on handle `0x05`

## Communication Flow

1. Establish BLE connection
2. Register for notifications on characteristic `0xFFE1` (handle `0x05`)
3. Send device info command on characteristic `0xFFE1` (handle `0x03`)
4. Send cell info command if no status notification is received

## Command Structure

Commands are sent as 20-byte frames with the following format:

| Offset | Length | Description |
|--------|---------|-------------|
| 0-3 | 4 | Header (0xAA 0x55 0x90 0xEB) |
| 4 | 1 | Command byte |
| 5 | 1 | Length byte |
| 6-9 | 4 | Value (32-bit) |
| 10-18 | 9 | Zero padding |
| 19 | 1 | CRC |

### Command Types

- Device Info: `0x97`
- Cell Info: `0x96`

### Example Command Payloads

#### Device Info Command (0x97)
```
AA 55 90 EB     # Header
97              # Command
00              # Length
00 00 00 00     # Value
00 00 00 00     # Padding
00 00 00 00     
00 00 00 00
00              # CRC (calculated across bytes 0-18)
```

#### Cell Info Command (0x96)
```
AA 55 90 EB     # Header
96              # Command  
00              # Length
00 00 00 00     # Value
00 00 00 00     # Padding
00 00 00 00     
00 00 00 00
00              # CRC (calculated across bytes 0-18)
```

## Response Handling

### Frame Assembly
- Complete responses range from 300-320 bytes
- Responses are fragmented due to BLE MTU limitations
- Each response frame starts with sequence: `0x55 0xAA 0xEB 0x90`
- When start sequence is detected, current frame buffer is cleared
- Frame validation uses CRC (sum of all bytes except CRC byte)
- All responses are received as notifications on characteristic `0xFFE1` (handle `0x05`)

### Response Frame Types
Response type is identified by byte[4]:
- `0x01`: Settings frame
- `0x02`: Cell info frame  
- `0x03`: Device info frame

## CRC Calculation

The CRC is calculated by summing all bytes in the frame except the CRC byte itself:

```c
uint8_t crc(const uint8_t data[], const uint16_t len) {
  uint8_t crc = 0;
  for (uint16_t i = 0; i < len; i++) {
    crc = crc + data[i];
  }
  return crc;
}
```

## Error Handling

- Maximum allowed frame size is 320 bytes
- Frames exceeding this size are dropped
- CRC validation is performed on assembled frames
- Frames with invalid CRC are discarded

RondaYummy, 2025-01-14:
@syssi I don't know, I tried to write this in Python using the data given, but I failed, maybe someone can tell me what I'm doing wrong?

`Connected to Andrii4 (C8:47:80:21:34:82) and notification started
Device Info command sent to Andrii4: aa5590eb97000000000000000000000000000011
Notification stopped for Andrii4 (C8:47:80:21:34:82)
Connected to Andrii 3 (C8:47:80:12:41:99) and notification started
Device Info command sent to Andrii 3: aa5590eb97000000000000000000000000000011
Notification received: 55aaeb90036d4a4b5f4244364131375336500000000031312e584100000031312e3438000000983c3a0001000000416e647269692033000000000000000031323334000000000000000000000000323431313330000034303331353335303131320030303000496e707574205573657264617461000031323334353600000000
Device Info Parsed:
device_name: mJK_BD6A17S6P11.XA11.48
serial_number: <:Andrii 3
firmware_version: 1234
hardware_version: 241130403153501
other_info: bytearray(b'12\x00000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00')
Invalid CRC: 70 != 0`

And code:
`import asyncio
from bleak import BleakClient, BleakScanner

SERVICE_UUID = "0000FFE0-0000-1000-8000-00805f9b34fb"
CHARACTERISTIC_UUID = "0000FFE1-0000-1000-8000-00805f9b34fb"

CMD_HEADER = bytes([0xAA, 0x55, 0x90, 0xEB])
CMD_TYPE_DEVICE_INFO = 0x97
CMD_TYPE_CELL_INFO = 0x96

def calculate_crc(data):
    return sum(data) & 0xFF

def create_command(command_type):
    frame = bytearray(20)
    frame[:4] = CMD_HEADER
    frame[4] = command_type
    frame[19] = calculate_crc(frame[:19])
    return frame

def parse_device_info(data):
    if data[0:4] != b'\x55\xAA\xEB\x90':
        print("Invalid start sequence")
        return None

    frame_type = data[4]
    if frame_type != 0x03:
        print(f"Unexpected frame type: {frame_type}")
        return None

    # Device Info Frame
    device_info = {
        "device_name": data[5:35].decode('utf-8', errors='ignore').strip(),
        "serial_number": data[35:55].decode('utf-8', errors='ignore').strip(),
        "firmware_version": data[55:75].decode('utf-8', errors='ignore').strip(),
        "hardware_version": data[75:95].decode('utf-8', errors='ignore').strip(),
        "other_info": data[95:],
    }

    print("Device Info Parsed:")
    for key, value in device_info.items():
        print(f"{key}: {value}")

    # Verify CRC (just as an example, the CRC should match the sum of the frame bytes)
    crc_calculated = calculate_crc(data[:-1])
    crc_received = data[-1]

    if crc_calculated != crc_received:
        print(f"Invalid CRC: {crc_calculated} != {crc_received}")
    else:
        print("CRC Valid")

    return device_info

async def notification_handler(sender, data):
    if data[:4] == b'\x55\xAA\xEB\x90':
        print(f"Notification received: {data.hex()}")
        # Parse the device info response
        parse_device_info(data)

async def connect_and_run(device):
    try:
        async with BleakClient(device.address) as client:
            await client.start_notify(CHARACTERISTIC_UUID, notification_handler)
            print(f"Connected to {device.name} ({device.address}) and notification started")

            command = create_command(CMD_TYPE_DEVICE_INFO)
            await client.write_gatt_char(CHARACTERISTIC_UUID, command)
            print(f"Device Info command sent to {device.name}: {command.hex()}")

            await asyncio.sleep(30)
            await client.stop_notify(CHARACTERISTIC_UUID)
            print(f"Notification stopped for {device.name} ({device.address})")
    except Exception as e:
        print(f"Error with {device.name} ({device.address}): {str(e)}")

async def main():
    devices = await BleakScanner.discover()
    if not devices:
        print("No BLE devices found.")
        return

    # Підключення по черзі
    for device in devices:
        await connect_and_run(device)

if __name__ == "__main__":
    asyncio.run(main())`

syssi, 2025-01-14:
Could you be more explicit? There is a log above the code block with the implementation which looks good in general and a second one after the code block which doesn't look good.

RondaYummy, 2025-01-14:
> Не могли б ви бути більш чіткими? Над блоком коду є журнал із реалізацією, яка загалом виглядає добре, і другий журнал після блоку коду, який виглядає недобре.

I'm sorry, I'm having trouble with the CRC calculation, I don't understand how to do it correctly and I'm getting only the following logs ( Looking for help with CRC checking, data integrity checking, and data chunking ):

<details><summary>Details</summary>
<p>

[Andrii 1] Connected and notification started
[Andrii 1] Device Info command sent: aa5590eb97000000000000000000000000000011
[Andrii 1] Notification received: 55aaeb9003de4a4b5f4244364131375336500000000031312e584100000031312e3438000000e8cd3a0001000000416e647269692031000000000000000031323334000000000000000000000000323431313330000034303331353335303338350030303000496e707574205573657264617461000031323334353600000000
[Andrii 1] Parsing Device Info Frame...
[Andrii 1] Raw data: bytearray(b'U\xaa\xeb\x90\x03\xdeJK_BD6A17S6P\x00\x00\x00\x0011.XA\x00\x00\x0011.48\x00\x00\x00\xe8\xcd:\x00\x01\x00\x00\x00Andrii 1\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00241130\x00\x0040315350385\x00000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00')
[Andrii 1] Device Info Parsed:
[Andrii 1] device_name: JK_BD6A17S6P
[Andrii 1] firmware_version: 11.XA
[Andrii 1] serial_number: 11.48
[Andrii 1] hardware_version: :
[Andrii 1] other_info: ['\x01', 'Andrii 1', '1234', '241130', '40315350385', '000', 'Input Userdata', '123456']
[Andrii 1] Invalid CRC: 162 != 0
[Andrii 1] Cell Info command sent: aa5590eb96000000000000000000000000000010
[Andrii 1] Notification received: 55aaeb9001de36100000040b0000220b000068100000541000000a0000005e100000180b00000000000000000000f00a0000409c00001e0000003c00000060ea00002c0100003c0000003c00000058020000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300000c00000001000000010000000100
[Andrii 1] Unknown frame type: 1

</p>
</details> 

```python
import asyncio
from bleak import BleakClient, BleakScanner
from colors import *

SERVICE_UUID = "0000FFE0-0000-1000-8000-00805f9b34fb"
CHARACTERISTIC_UUID = "0000FFE1-0000-1000-8000-00805f9b34fb"

CMD_HEADER = bytes([0xAA, 0x55, 0x90, 0xEB])
CMD_TYPE_DEVICE_INFO = 0x97
CMD_TYPE_CELL_INFO = 0x96

def calculate_crc(data):
    return sum(data) & 0xFF

def create_command(command_type):
    frame = bytearray(20)
    frame[:4] = CMD_HEADER
    frame[4] = command_type
    frame[19] = calculate_crc(frame[:19])
    return frame

def log(device_name, message):
    """Додає назву пристрою до логів."""
    print(f"{BLUE}[{device_name}]{RESET} {message}")

def parse_device_info(data, device_name):
    """Парсинг Device Info Frame (0x03)."""
    log(device_name, "Parsing Device Info Frame...")

    try:
        log(device_name, f"Raw data: {data}")
        # Використання структури протоколу для визначення довжин
        device_info = parse_device_data(data)

        log(device_name, "Device Info Parsed:")
        for key, value in device_info.items():
            log(device_name, f"{key}: {value}")

        # CRC Validation
        crc_calculated = calculate_crc(data[:-1])
        crc_received = data[-1]
        if crc_calculated != crc_received:
            log(device_name, f"Invalid CRC: {crc_calculated} != {crc_received}")
[... 100 lines cut ...]

            await asyncio.sleep(1)  # Очікування між командами

            await client.write_gatt_char(CHARACTERISTIC_UUID, cell_info_command)
            log(device.name, f"Cell Info command sent: {cell_info_command.hex()}")

            await asyncio.sleep(30)  # Час для отримання даних
            await client.stop_notify(CHARACTERISTIC_UUID)
            log(device.name, "Notification stopped")
    except Exception as e:
        log(device.name, f"Error: {str(e)}")

async def main():
    devices = await BleakScanner.discover()
    if not devices:
        print("No BLE devices found.")
        return

    for device in devices:
        await connect_and_run(device)

if __name__ == "__main__":
    asyncio.run(main())

```

syssi, 2025-01-14:
Please keep in mind the size of a valid / complete JK-BMS response frame is between 300 and 320 bytes. The received responses above are around 128 bytes. You have to concatenate multiple BLE notifications to a complete frame. At the moment the last byte of the notification isn't the CRC value because it's not the end of the frame.

RondaYummy, 2025-01-14:
> Please keep in mind the size of a valid / complete JK-BMS response frame is between 300 and 320 bytes. The received responses above are around 128 bytes. You have to concatenate multiple BLE [...]

I think we managed to get a whole one. Where can I find information on how to disassemble it? I'm showing it here in HEX. I also have a question about Cell data, I get data like in the second block of parts, I don't understand why there are so many of them. My battery has 12 cells, and there are so many of them. I will be very grateful for the tips...
I hope I haven't bored you yet.

<details><summary>Details</summary>
<p>

[Andrii 2] Connected and notification started
[Andrii 2] Device Info command sent: aa5590eb97000000000000000000000000000011
[Andrii 2] Full frame received: 55aaeb9003344a4b5f42443641323053365000000000313755000000000031372e3032000000d42c3b0001000000416e6472696920320000000000000000313233340000000000000000000000003234313133300000343038303933383230383800303030004a4b2d424d5300000000000000000000313233343536000000000000000000004a4b2d424d5300000000000000000000feffffff9fb91d0200000000901f00000000c0d8e7f73c00000100000000000000000000ff670000000000000000000000000000ff0f000000000000000000000000000003ff670000000000000000000000000009080001640000005f0000003c000000320000000000000000000000100e00003232011e0000000000000000000000000000000000fe9f699f0f0000000000000004
[Andrii 2] Parsing Device Info Frame...
[Andrii 2] Raw data: bytearray(b'U\xaa\xeb\x90\x034JK_BD6A20S6P\x00\x00\x00\x0017U\x00\x00\x00\x00\x0017.02\x00\x00\x00\xd4,;\x00\x01\x00\x00\x00Andrii 2\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00241130\x00\x0040809382088\x00000\x00JK-BMS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00JK-BMS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\xff\xff\x9f\xb9\x1d\x02\x00\x00\x00\x00\x90\x1f\x00\x00\x00\x00\xc0\xd8\xe7\xf7<\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xffg\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\xffg\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t\x08\x00\x01d\x00\x00\x00_\x00\x00\x00<\x00\x00\x002\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10\x0e\x00\x0022\x01\x1e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x9fi\x9f\x0f\x00\x00\x00\x00\x00\x00\x00\x04')
[Andrii 2] Device Info Parsed:
[Andrii 2] device_name: 4JK_BD6A20S6P
[Andrii 2] firmware_version: 17U
[Andrii 2] serial_number: 17.02
[Andrii 2] hardware_version: ,;
[Andrii 2] other_info: ['\x01', 'Andrii 2', '1234', '241130', '40809382088', '000', 'JK-BMS', '123456', 'JK-BMS', '\x1d\x02', '<', '\x01', 'g', '\x0f', '\x03g', '\t\x08', '\x01d', '_', '<', '2', '\x10\x0e', '22\x01\x1e', 'i\x0f']
[Andrii 2] CRC Valid
[Andrii 2] Full frame received: 55aaeb9003344a4b5f42443641323053365000000000313755000000000031372e3032000000d42c3b0001000000416e6472696920320000000000000000313233340000000000000000000000003234313133300000343038303933383230383800303030004a4b2d424d5300000000000000000000313233343536000000000000000000004a4b2d424d5300000000000000000000feffffff9fb91d0200000000901f00000000c0d8e7f73c00000100000000000000000000ff670000000000000000000000000000ff0f000000000000000000000000000003ff670000000000000000000000000009080001640000005f0000003c000000320000000000000000000000100e00003232011e0000000000000000000000000000000000fe9f699f0f0000000000000004aa5590ebc801010000000000000000000000004441540d0a
[Andrii 2] Invalid CRC: 50 != 10
[Andrii 2] Full frame received: 55aaeb9003344a4b5f42443641323053365000000000313755000000000031372e3032000000d42c3b0001000000416e6472696920320000000000000000313233340000000000000000000000003234313133300000343038303933383230383800303030004a4b2d424d5300000000000000000000313233343536000000000000000000004a4b2d424d5300000000000000000000feffffff9fb91d0200000000901f00000000c0d8e7f73c00000100000000000000000000ff670000000000000000000000000000ff0f000000000000000000000000000003ff670000000000000000000000000009080001640000005f0000003c000000320000000000000000000000100e00003232011e0000000000000000000000000000000000fe9f699f0f0000000000000004aa5590ebc801010000000000000000000000004441540d0a41540d0a
[Andrii 2] Invalid CRC: 222 != 10
[Andrii 2] Full frame received: 55aaeb9003344a4b5f42443641323053365000000000313755000000000031372e3032000000d42c3b0001000000416e6472696920320000000000000000313233340000000000000000000000003234313133300000343038303933383230383800303030004a4b2d424d5300000000000000000000313233343536000000000000000000004a4b2d424d5300000000000000000000feffffff9fb91d0200000000901f00000000c0d8e7f73c00000100000000000000000000ff670000000000000000000000000000ff0f000000000000000000000000000003ff670000000000000000000000000009080001640000005f0000003c000000320000000000000000000000100e00003232011e0000000000000000000000000000000000fe9f699f0f0000000000000004aa5590ebc801010000000000000000000000004441540d0a41540d0a41540d0a
[Andrii 2] Invalid CRC: 138 != 10
[Andrii 2] Full frame received: 55aaeb9003344a4b5f42443641323053365000000000313755000000000031372e3032000000d42c3b0001000000416e6472696920320000000000000000313233340000000000000000000000003234313133300000343038303933383230383800303030004a4b2d424d5300000000000000000000313233343536000000000000000000004a4b2d424d5300000000000000000000feffffff9fb91d0200000000901f00000000c0d8e7f73c00000100000000000000000000ff670000000000000000000000000000ff0f000000000000000000000000000003ff670000000000000000000000000009080001 [... cut ...]
[Andrii 2] Cells Wire Resistance 44: 0.145 V
[Andrii 2] Cells Wire Resistance 45: 0.144 V
[Andrii 2] Cells Wire Resistance 46: 0.146 V
[Andrii 2] Cells Wire Resistance 47: 0.145 V
[Andrii 2] Cells Wire Resistance 48: 0.143 V
[Andrii 2] Cells Wire Resistance 49: 0.145 V

</p>
</details> 

I displayed only those with a value greater than 0 and divided them by their values, if greater than 4, I displayed them in a log as Cell and if less than 4, as Cells Wire Resistance. I just have no idea how to find out what exactly is here and where :(

syssi, 2025-01-14:
This is the frame structure of a cell info frame. Please be aware. You are requesting the `device info frame` at the moment. If the frame starts with `55aaeb9003` it's a device info frame. `55aaeb9002` is a cell info frame.

# JK-BMS Cell Info Frame Specification (JK02-32S)

| Position (Bytes) | Length | Description | Format | Coefficient | Unit | Notes |
|-----------------|--------|-------------|---------|-------------|------|-------|
| 0-3 | 4 | Header | Raw | - | - | Fixed value: 0x55 0xAA 0xEB 0x90 |
| 4 | 1 | Record type | Raw | - | - | Fixed value: 0x02 |
| 5 | 1 | Frame counter | Raw | - | - | - |
| 6-69 | 64 | Cell voltages 1-32 | 16-bit | 0.001 | V | 2 bytes per cell |
| 54-57 | 4 | Enabled cells bitmask | Raw | - | - | 0xFF 0xFF 0xFF 0xFF = 32 cells |
| 64-127 | 64 | Cell resistances 1-32 | 16-bit | 0.001 | Ohm | 2 bytes per cell |
| 112-113 | 2 | Power tube temperature | 16-bit signed | 0.1 | °C | - |
| 114-117 | 4 | Wire resistance warning bitmask | Raw | - | - | 1 bit per cell/wire |
| 118-121 | 4 | Battery voltage | 32-bit | 0.001 | V | - |
| 122-125 | 4 | Battery power | 32-bit | 0.001 | W | - |
| 126-129 | 4 | Charge current | 32-bit signed | 0.001 | A | - |
| 130-131 | 2 | Temperature Sensor 1 | 16-bit signed | 0.1 | °C | - |
| 132-133 | 2 | Temperature Sensor 2 | 16-bit signed | 0.1 | °C | - |
| 134-135 | 2 | Errors bitmask | 16-bit | - | - | System error flags |
| 138-139 | 2 | Balance current | 16-bit signed | 0.001 | A | - |
| 140 | 1 | Balancing action | Raw | - | - | 0=Off, 1=Charging, 2=Discharging |
| 141 | 1 | State of charge | 8-bit | 1 | % | - |
| 142-145 | 4 | Remaining capacity | 32-bit | 0.001 | Ah | - |
| 146-149 | 4 | Nominal capacity | 32-bit | 0.001 | Ah | - |
| 150-153 | 4 | Cycle count | 32-bit | 1 | - | - |
| 154-157 | 4 | Total cycle capacity | 32-bit | 0.001 | Ah | - |
| 158 | 1 | State of Health (SOH) | 8-bit | 1 | % | - |
| 159 | 1 | Precharge status | Raw | - | - | 0=Off, 1=On |
| 162-165 | 4 | Total runtime | 32-bit | 1 | s | - |
| 166 | 1 | Charging MOSFET status | Raw | - | - | 0=Off, 1=On |
| 167 | 1 | Discharging MOSFET status | Raw | - | - | 0=Off, 1=On |
| 168 | 1 | Precharging status | Raw | - | - | 0=Off, 1=On |
| 222-223 | 2 | Temperature Sensor 5 | 16-bit signed | 0.1 | °C | - |
| 224-225 | 2 | Temperature Sensor 4 | 16-bit signed | 0.1 | °C | - |
| 226-227 | 2 | Temperature Sensor 3 | 16-bit signed | 0.1 | °C | - |
| 186-187 | 2 | Emergency time countdown | 16-bit | 1 | s | >0 = Emergency active |
| 299 | 1 | CRC checksum | Raw | - | - | - |

syssi, 2025-01-15:
# JK02_32S Device Info Frame Field Table

| Position (Bytes) | Length | Description | Format | Coefficient | Unit | Notes |
|-----------------|--------|-------------|---------|-------------|------|-------|
| 0-3 | 4 | Frame header | hex | - | - | Fixed value: 0x55 0xAA 0xEB 0x90 |
| 4 | 1 | Frame type | uint8 | - | - | Fixed value: 0x03 |
| 5 | 1 | Frame counter | uint8 | - | - | Increments with each frame |
| 6-21 | 16 | Vendor ID | string | - | - | Null-terminated string |
| 22-29 | 8 | Hardware version | string | - | - | Null-terminated string |
| 30-37 | 8 | Software version | string | - | - | Null-terminated string |
| 38-41 | 4 | Device uptime | uint32 | 1 | seconds | Little-endian |
| 42-45 | 4 | Power on count | uint32 | 1 | count | Little-endian |
| 46-61 | 16 | Device name | string | - | - | Null-terminated string |
| 62-77 | 16 | Device passcode | string | - | - | Null-terminated string |
| 78-85 | 8 | Manufacturing date | string | - | - | Null-terminated string |
| 86-96 | 11 | Serial number | string | - | - | Null-terminated string |
| 97-101 | 5 | Passcode | string | - | - | Null-terminated string |
| 102-117 | 16 | User data | string | - | - | Null-terminated string |
| 118-133 | 16 | Setup passcode | string | - | - | Null-terminated string |
| 184 | 1 | UART1M Protocol number | uint8 | - | - | Protocol version |
| 185 | 1 | CAN Protocol number | uint8 | - | - | Protocol version |
| 218 | 1 | UART2M Protocol number | uint8 | - | - | Protocol version |
| 219 | 1 | UART2M Protocol enable | uint8 | - | - | Enable flag |
| 234 | 1 | LCD buzzer trigger | uint8 | - | - | Trigger setting |
| 235 | 1 | DRY1 Trigger | uint8 | - | - | Trigger setting |
| 236 | 1 | DRY2 Trigger | uint8 | - | - | Trigger setting |
| 237 | 1 | UART protocol library version | uint8 | - | - | Protocol version |
| 238-241 | 4 | LCD Buzzer Trigger Value | uint32 | - | - | Little-endian |
| 242-245 | 4 | LCD Buzzer Release Value | uint32 | - | - | Little-endian |
| 246-249 | 4 | DRY1 Trigger Value | uint32 | - | - | Little-endian |
| 250-253 | 4 | DRY1 Release Value | uint32 | - | - | Little-endian |
| 254-257 | 4 | DRY2 Trigger Value | uint32 | - | - | Little-endian |
| 258-261 | 4 | DRY2 Release Value | uint32 | - | - | Little-endian |
| 262-265 | 4 | Data Stored Period | uint32 | - | - | Little-endian |
| 266 | 1 | RCV Time | uint8 | 0.1 | hours | Request Charge Voltage Time |
| 267 | 1 | RFV Time | uint8 | 0.1 | hours | Request Float Voltage Time |
| 268 | 1 | CAN protocol library version | uint8 | - | - | Protocol version |
| 299 | 1 | CRC | uint8 | - | - | Frame checksum |

RondaYummy, 2025-01-15:
@syssi Thank you very much for the data provided, I managed to do everything with your help in one go in Python. I couldn't do it in JS for two months. I don't know much about Python or how to work with bytes at this level because I'm more of a front-end developer. But with your help, everything worked out and everything is working, I am sincerely grateful for all the tips. I'm still afraid to tinker with the settings so as not to harm the BMS.
I was beginning to think that I would not succeed.

**_You're the best_**

closed 2025-01-15 by syssi

RondaYummy, 2025-01-19:
@syssi Hi, do you have a description like the one you posted above for other characteristics, for 0x01?

syssi, 2025-01-19:
| Position (Bytes) | Length | Description | Format | Coefficient | Unit | Notes |
| --------------- | ------ | ----------- | ------ | ----------- | ---- | ----- |
| 0-3 | 4 | Header | HEX | - | - | 0x55 0xAA 0xEB 0x90 |
| 4 | 1 | Frame type | HEX | - | - | 0x01 |
| 5 | 1 | Frame counter | HEX | - | - | 0x4F |
| 6-9 | 4 | Smart sleep voltage | uint32 | 0.001 | V | - |
| 10-13 | 4 | Cell UVP (undervoltage protection) | uint32 | 0.001 | V | - |
| 14-17 | 4 | Cell UVPR (undervoltage protection recovery) | uint32 | 0.001 | V | - |
| 18-21 | 4 | Cell OVP (overvoltage protection) | uint32 | 0.001 | V | - |
| 22-25 | 4 | Cell OVPR (overvoltage protection recovery) | uint32 | 0.001 | V | - |
| 26-29 | 4 | Balance trigger voltage | uint32 | 0.001 | V | - |
| 30-33 | 4 | SOC 100% voltage | uint32 | 0.001 | V | - |
| 34-37 | 4 | SOC 0% voltage | uint32 | 0.001 | V | - |
| 38-41 | 4 | Cell request charge voltage (RCV) | uint32 | 0.001 | V | - |
| 42-45 | 4 | Cell request float voltage (RFV) | uint32 | 0.001 | V | - |
| 46-49 | 4 | Power off voltage | uint32 | 0.001 | V | - |
| 50-53 | 4 | Max charge current | uint32 | 0.001 | A | - |
| 54-57 | 4 | Charge OCP delay | uint32 | 1.0 | s | - |
| 58-61 | 4 | Charge OCP recovery time | uint32 | 1.0 | s | - |
| 62-65 | 4 | Max discharge current | uint32 | 0.001 | A | - |
| 66-69 | 4 | Discharge OCP delay | uint32 | 1.0 | s | - |
| 70-73 | 4 | Discharge OCP recovery time | uint32 | 1.0 | s | - |
| 74-77 | 4 | Short circuit protection recovery time | uint32 | 1.0 | s | - |
| 78-81 | 4 | Max balance current | uint32 | 0.001 | A | - |
| 82-85 | 4 | Charge OTP | uint32 | 0.1 | °C | - |
| 86-89 | 4 | Charge OTP recovery | uint32 | 0.1 | °C | - |
| 90-93 | 4 | Discharge OTP | uint32 | 0.1 | °C | - |
| 94-97 | 4 | Discharge OTP recovery | uint32 | 0.1 | °C | - |
| 98-101 | 4 | Charge UTP | int32 | 0.1 | °C | - |
| 102-105 | 4 | Charge UTP recovery | int32 | 0.1 | °C | - |
| 106-109 | 4 | MOS OTP | int32 | 0.1 | °C | - |
| 110-113 | 4 | MOS OTP recovery | int32 | 0.1 | °C | - |
| 114-117 | 4 | Cell count | uint8 | 1.0 | - | Only first byte used |
| 118-121 | 4 | Charge switch | bool | - | - | Only first byte used |
| 122-125 | 4 | Discharge switch | bool | - | - | Only first byte used |
| 126-129 | 4 | Balancer switch | bool | - | - | Only first byte used |
| 130-133 | 4 | Nominal battery capacity | uint32 | 0.001 | Ah | - |
| 134-137 | 4 | Short circuit protection delay | uint32 | 1.0 | μs | - |
| 138-141 | 4 | Start balance voltage | uint32 | 0.001 | V | - |
| 142-269 | 128 | Connection wire resistance (1-32) | uint32 | 0.001 | Ω | 32 x 4-byte values |
| 270 | 1 | Device address | uint8 | - | - | - |
| 274 | 1 | Precharge time | uint8 | - | s | - |
| 282-283 | 2 | Controls bitmask | uint16 | - | - | See bitmask notes |
| 286 | 1 | Smart sleep | uint8 | - | h | - |
| 287 | 1 | Data field enable control 0 | uint8 | - | - | - |
| 299 | 1 | CRC | uint8 | - | - | 0x40 |

Bitmask notes for bytes 282-283:
- bit0: Heating enabled (1)
- bit1: Disable temperature sensors (2)
- bit2: GPS Heartbeat (4)
- bit3: Port switch (1: RS485, 0: CAN) (8)
- bit4: Display always on (16)
- bit5: Special charger (32)
- bit6: Smart sleep (64)
- bit7: Disable PCL module (128)
- bit8: Timed stored data (256)
- bit9: Charging float mode (512)
- bit10-15: Reserved

RondaYummy, 2025-01-20:
> # JK02_32S Device Info Frame Field Table
> Position (Bytes)	Length	Description	Format	Coefficient	Unit	Notes
> 0-3	4	Frame header	hex	-	-	Fixed value: 0x55 0xAA 0xEB 0x90
> 4	1	Frame type	uint8	-	-	Fixed [...]

Hi, I have a question about “Cell resistances 1-32”. It says that it is in the range of 64-127, but there are other data that are in this range, is this a mistake?
For example, “Power tube temperature” is in the range of 112-113. I get a little invalid data after Cell resistances, but if I add +32 to the indices, it looks better.

syssi, 2025-01-24:
Ups. I will provide a fixed version later the day!

RondaYummy, 2025-01-28:
> Ups. I will provide a fixed version later the day!

If I haven't bored you yet and you have a chance to answer, I have one more small question.
I send commands to get Device info and Cell info, but sometimes one of the devices does not send Cell info, instead I see that it sends Setting headers, and until I get these headers I do not get Cell info. It helps to restart the application and reconnect to all devices, but after a while, this may happen again. Have you experienced this? Because in most cases, everything works correctly and I get the frames I need, but at a random moment it happens, and I don't know how to deal with it.
I also tried to send commands to get Cell info again in this case, but it doesn't help, it will send only Setting info all the time. :(

soeter01, 2025-08-15:
> @soeter01 <https://github.com/soeter01> Are you aware of the different
> btsnoop (bluetooth traffic) captures here?
>
> https://github.com/syssi/esphome-jk-bms/tree/main/docs
>
> You can open the capture using wireshark to get an better idea about the
> communication.
> btsnoop.png (view on web)

Thanks. Will look into this - not much experience as yet with wireshark
though! Purchased a BLE sniffer and will see what the JK app actually sends
to the BMS.

syssi, 2025-08-15:
You don't need to prepare a capture on your own. Just pick an existing one. The protocol is pretty the same for all BMS models.

soeter01, 2025-08-15:
> This is correct. Did you subscribe to the notification stream beforehand
> and do you issue the two commands in the same order as the btsnoop capture?

Yes first subscribe & then command issue. BUT: 1 send direct 96 command -
not after 97 command... I'll try to subsequent commands starting with 97
and then 96. Will also try improving the timing in the code when reading
the frames once notification is received.

soeter01, 2025-08-15:
It seems to work: key was to issue 97 command first and then 96 command to
get the bms continuously sending data incl the cell data! Many thanks for
this key hint!

soeter01, 2025-08-15:
Did you ever manage to change the parameters such as the balance trigger
voltage or switching the charging off?

syssi, 2025-08-15:
Yes. Most registers are supported by this project.

syssi, 2025-08-15:
Please provide a complete raw frame + your intention. I will verify your "guess".

soeter01, 2025-08-17:
I plan to set the balancing start value at 3.42V by sending command over
BLE to my JK 2A20A20P. I propose to send the following 20 byte frame:
AA 55 90 EB                           the header
8A                                           the address of the balancing
start register (138 decimal following your table with the settings obtained
when 0x97 command is used)
04                                            indicating 4 bytes sent
5C 0D 00 00                            the value 3420 in little endian
00 00 00 00 00 00 00 00 00   9 padding 0 bytes
0x71                                        CRC

would this work?