syssi/esphome-jk-bms #664: JK BMS beep while polling

ouinouin, 2025-01-16:
Hi, just gone from daly (after one DOA and one not working after 2 month.) to JK bms, 

i m using the bluetooth communication, the issue is that from times to times there is a beep coming frome the MS, this is the same beep we hear while changin the settings in the officiall application, i found nowere in the application for this beep to be removed.
so i m guessing there are categories , and its while ths bluetooth is polling from the values to the settings that the beep occurs. (i couldnt find a specific time when it occurs, i was thinking it was tied to my polling time but i have difficulties to see a scheme.
i sent a message to JK in parallel to ask if they can add in their firmware a toggle for this useless beep (at least in stationnary setups)

```
substitutions:
  name: "lacoume-bt-devices"
  device_description: "full monitor of lacoume"
  mac_address: C8:47:80:26:29:62
  protocol_version: JK02_32S
esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-daly-bms"
    version: 1.0.0

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf

external_components:
  - source: github://Fabian-Schmidt/esphome-victron_ble
  - source: github://syssi/esphome-jk-bms@main
  - source: github://syssi/esphome-daly-bms@main

wifi:
  networks:
    - ssid: mifimiji
      password: paswoooooooord
      manual_ip:
        static_ip: 192.168.7.253
        gateway: 192.168.7.254
        subnet: 255.255.255.0
    - ssid: ouinouin_Nicol
      password: paswoooooooord
ota:
  platform: esphome

web_server:
  port: 80
  version: 1

logger:
#  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client
mqtt:
  broker: 192.168.7.254
api:
  #encryption:
  #  key: "ofaI6e0mYZ28/PNNKv09lBuRQmwDU1+hoy8o9ztNyWg="
  reboot_timeout: 0s
esp32_ble_tracker:
  scan_parameters:
    active: false
    interval: 2s

ble_client:
  - mac_address: ${mac_address}
[... 521 lines cut ...]
    discharge_overtemperature_protection_recovery:
      name: "${name} discharge overtemperature protection recovery"
    charge_undertemperature_protection:
      name: "${name} charge undertemperature protection"
    charge_undertemperature_protection_recovery:
      name: "${name} charge undertemperature protection recovery"
    power_tube_overtemperature_protection:
      name: "${name} power tube overtemperature protection"
    power_tube_overtemperature_protection_recovery:
      name: "${name} power tube overtemperature protection recovery"

switch:
  - platform: jk_bms_ble
  #  balancer:
  #   name: "BMS balancer"
    charging:
      name: "BMS charging"
    discharging:
      name: "BMS discharging"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "enable esp32 bluetooth connection" 
```

ouinouin, 2025-01-16:
after trial and errors , it looks that this portion of code is the culrpit :  
 - interval: 90s
    then:
      - button.press: retrieve_BMS_settings
#      - lambda: |-
#          id(MPPTstate).publish_state(id(MPPTstate).state);
indeed in case i get some infos about JK bms suppport to gat rid of this noise in any case, ill publish it.

closed 2025-01-16 by ouinouin

syssi, 2025-01-16:
Every write instruction / command let the BMS beep once. On an established BLE connection the implementation issues two commands (so on every BLE re-connect the BMS beeps once) and on every change of a read/write entity (number, select, button, switch) the BMS will beep. This cannot be avoided.