syssi/esphome-jk-bms #665: switch JKBMS using BLE

humpataa, 2025-01-16:
So I am trying to switch Charge / Discharge mode via Bluetooth - esphome-jk-bms claims it can do it. Since I cannot install the full thing (just have an ESP32 at hand), I am trying to use code from this repository to implement the switch into my Arduino IDE project. Unfortunately, it doesn't work. First, can someone confirm that this actually really works?
(I am not talking about serial communication or RS485 but Bluetooth!)

Here's what I have tried based on this info:
https://github.com/syssi/esphome-jk-bms/blob/main/docs/protocol-design-ble.md

With excerpts of the code here:
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/switch/__init__.py#L17-L22
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/switch/jk_switch.cpp#L11
https://github.com/syssi/esphome-jk-bms/blob/824f7d5a5b2a4638301c0765b0a14c74fc79b112/components/jk_bms_ble/jk_bms_ble.cpp#L1440

My sketch makes a connection (using NimBLE) to the JKBMS and sends two commands (getInfo and getdeviceInfo) to get status data, which works good, repeatedly (the BMS sends info every 500ms I believe). I have added a third command similar to the above code that shall set Discharge to ON. In detail:

```
uint8_t frame[20];
  frame[0] = 0xAA;     // start sequence
  frame[1] = 0x55;     // start sequence
  frame[2] = 0x90;     // start sequence
  frame[3] = 0xEB;     // start sequence
  frame[4] = address;  // holding register
  frame[5] = length;   // size of the value in byte
  frame[6] = value >> 0;
  frame[7] = value >> 8;
  frame[8] = value >> 16;
  frame[9] = value >> 24;
  frame[10] = 0x00;
  frame[11] = 0x00;
  frame[12] = 0x00;
  frame[13] = 0x00;
  frame[14] = 0x00;
  frame[15] = 0x00;
  frame[16] = 0x00;
  frame[17] = 0x00;
  frame[18] = 0x00;
  frame[19] = crc(frame, sizeof(frame) - 1);
```
Where address is 0xAC (for Discharge register), length is 0x01 or 0x04 (tried both because I am not sure which protocol my BMS uses), value is 0x00000001 (switch ON). I am using the crc-function of the above mentioned protocol specs.

Unfortunately nothing happens to the "switches". The BMS sends a response that is different from the Info blocks, but I don't know what it means - probably some kind of error message:

`AA 55 90 EB C8 1 0 0 0 0 0 0 0 0 0 0 0 0 0 43`

Any thoughts? I am our of ideas.
A log of working command / reply traffic would be great ... any help is appreciated.

syssi, 2025-01-16:
>  Since I cannot install the full thing (just have an ESP32 at hand), I am trying to use code from this repository to implement the switch into my Arduino IDE project. Unfortunately, it doesn't work. First, can someone confirm that this actually really works?

If you are able to flash the ESP using the Arduino IDE it will be pretty simple to move to ESPHome. Do you like to give it a try?

Do you have access to a linux host?

humpataa, 2025-01-16:
Thanks Sebastian, I am on a Mac ... it has a unix-like prompt, no?

syssi, 2025-01-16:
Is the python package manager available on Mac per default? See https://esphome.io/guides/installing_esphome.html#mac

Please install the `esphome` package by `pip3 install esphome.`. As soon as the package is installed there should be a new CLI command available called `esphome`. The command can be used to flash the ESP32:

```
# Download a example YAML
$ wget https://raw.githubusercontent.com/syssi/esphome-jk-bms/refs/heads/main/esp32-ble-v14-example.yaml

# Change the MAC address to the one of your BMS: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml#L5
$ vim esp32-ble-v14-example.yaml

# Compile and flash the YAML configuration
esphome run esp32-ble-v14-example.yaml
```

That's all!

syssi, 2025-01-16:
I've collected the RAW payloads from an ESPHome log if you still want to implement the protocol on your own:

```
Turn charging on: AA.55.90.EB.1D.04.01.00.00.00.00.00.00.00.00.00.00.00.00.9C
Turn charging off: AA.55.90.EB.1D.04.00.00.00.00.00.00.00.00.00.00.00.00.00.9B

Turn discharging on: AA.55.90.EB.1E.04.01.00.00.00.00.00.00.00.00.00.00.00.00.9D
Turn discharging off: AA.55.90.EB.1E.04.00.00.00.00.00.00.00.00.00.00.00.00.00.9C
```

Make sure to write into the correct handle (it's not the same as the BLE notification handle):

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L150-L203

humpataa, 2025-01-16:
Thank you will try. I am not sure about the "handle" because ... there is simply no option in NimBLE's write function for this. My sketch sends cell info command and receives notification via callback (split packages coming in):

```
pSvc = pClient->getService(NimBLEUUID(serviceUUID));
  if (pSvc) {
    pChr = pSvc->getCharacteristic(NimBLEUUID(charUUID));
    if (pChr) {

      // activate notify and set callback
      if (pChr->canNotify()) {
          pChr->subscribe(true, notifyCallback);
      }
...
```
...
```
pChr->writeValue(getInfo, sizeof(getInfo), false);
...
pChr->writeValue(getdeviceInfo, sizeof(getdeviceInfo), false);
```

Guess the code sends using handle 3 and receives using handle 5 by its own?

humpataa, 2025-01-16:
> ```
> Turn charging on: AA.55.90.EB.1D.04.01.00.00.00.00.00.00.00.00.00.00.00.00.9C
> Turn charging off: AA.55.90.EB.1D.04.00.00.00.00.00.00.00.00.00.00.00.00.00.9B
>
> Turn discharging on: [...]

Now this is strange: the RAW payloads use 1D and 1E as commands.
But your code here (and iirc the chinenglish JKBMS pdf  as well) say AB (charging register) and AC (discharging register) instead?!
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/switch/__init__.py#L17-L22

syssi, 2025-01-16:
You are still mixing UART vs. BLE communication. These are the BLE registers:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/switch/__init__.py#L37-L50

humpataa, 2025-01-16:
Because you pointed me to it:
https://github.com/syssi/esphome-jk-bms/issues/20#issuecomment-2564723475

Well, my fault, I should have looked more carefully. :disappointed_relieved:
Thanks alot for your help!

syssi, 2025-01-16:
Because issue #20 is about "Control the BMS via serial interface". ;-)