syssi/esphome-jk-bms #671: [W][component:237]: Component esp32_ble took a long time for an operation (1972 ms)

abc2006, 2025-01-25:
Hi, my post is possibly related to #659 or #608. Please note that I am not a native English speaker, so my statements may not be 100% accurately translated. (Translation provided by ChatGPT)

I am using two ESP32s to connect two JK-BMS:

JK-BMS <-> Bluetooth <-> ESP32 <-> WiFi <-> MQTT Server (Mosquitto, not related to HomeAssistant or anything else)

This setup is duplicated, see the configurations jk00[12]-esp32-ble-v14.yaml.orig:

[jk002-esp32-ble-v14.yaml.orig.txt](https://github.com/user-attachments/files/18546478/jk002-esp32-ble-v14.yaml.orig.txt)
[jk001-esp32-ble-v14.yaml.orig.txt](https://github.com/user-attachments/files/18546481/jk001-esp32-ble-v14.yaml.orig.txt)

Sometimes, one or both of the connections stop working, so I don't receive any new values in my MQTT server. A power cycle of the ESP32 fixes it, and then it works again.

Today, I did some research and found the following message:

[W][component:237]: Component esp32_ble took a long time for an operation (965 ms).
[W][component:238]: Components should block for at most 30 ms.

To reduce the load, I changed the configurations and removed any unused datapoints (e.g., I have 16 cells, so I removed the voltage and resistance datapoints for cells 17 to 24). As a result, the "long time" dropped to around 700 ms:

[jk002-esp32-ble-v14.yaml.txt](https://github.com/user-attachments/files/18546517/jk002-esp32-ble-v14.yaml.txt)
[jk001-esp32-ble-v14.yaml.txt](https://github.com/user-attachments/files/18546518/jk001-esp32-ble-v14.yaml.txt)

[W][component:237]: Component esp32_ble took a long time for an operation (696 ms).
[W][component:238]: Components should block for at most 30 ms.

I also pulled the updates from GitHub and updated the Python files on my computer, before recompiling and updating with the new configurations.

Just now, I was watching the log outputs a bit, and I saw this message:

[14:24:58][D][jk_bms_ble:599]: Temperature sensor 4: 0.0 °C
[14:24:58][D][jk_bms_ble:600]: Temperature sensor 5: 0.0 °C
[14:24:58][D][jk_bms_ble:601]: Time enter sleep: 0 s
[14:24:58][D][jk_bms_ble:602]: PCL Module State: OFF
[14:24:58][W][component:237]: Component esp32_ble took a long time for an operation (695 ms).
[14:24:58][W][component:238]: Components should block for at most 30 ms.
[14:24:59][D][esphome.ota:117]: Starting update from 192.168.243.253...
[14:24:59][W][component:157]: Component esphome.ota set Warning flag: unspecified
[14:24:59][D][switch:016]: 'jk-bms-001 enable bluetooth connection' Turning OFF.
[14:24:59][I][ble_client:043]: [C8:47:8C:E8:3E:CE] Disabling BLE client.
[14:24:59][I][esp32_ble_client:084]: [0] [C8:47:8C:E8:3E:CE] Disconnecting.
[14:24:59][D][switch:055]: 'jk-bms-001 enable bluetooth connection': Sending state OFF
[14:24:59][D][main:2757]: BLE connection suspended for OTA update
[14:24:59][W][esphome.ota:129]: Magic bytes do not match! 0x47-0x45-0x54-0x20-0x2F
[14:24:59][E][component:164]: Component esphome.ota set Error flag: unspecified
[14:24:59][D][esp-idf:000][BTU_TASK]: W (5637211) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
[14:24:59][D][esp-idf:000][BTC_TASK]: E (5637222) BT_APPL: BTA_GATTC_DeregisterForNotifications registration not found bd_addr:c8:47:8c:e8:3e:ce
[14:25:16][W][component:170]: Component mqtt cleared Warning flag
[14:25:16][I][mqtt:291]: MQTT Connected!
[14:25:18][I][app:062]: setup() finished successfully!
[14:25:18][I][app:100]: ESPHome version 2024.12.4 compiled on Jan 25 2025, 12:50:41
[14:25:18][I][app:102]: Project syssi.esphome-jk-bms version 2.1.0
[14:25:18][C][wifi:600]: WiFi:

However, I didn't start an update, I was just watching the OTA log output. Is this error known? If so, what can I do? If not, how can I provide more information? I will keep an eye on it to see if this error occurs more frequently.

This is my Hardware and software-versions: 
Processing jk-bms-001 (board: wemos_d1_mini32; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208

Thanks for your help, 
Stephan 

Thanks for your help, 
Stephan

Wiki591, 2025-02-09:
Similar problem with my JK BMS. Worked perfectly for about two months, meanwhile receiving the data is not reliable anymore. 
I am monitoring the battery from my RV, using deep sleep for one hour. Run time before was 30 secs without any problems. To get at least some data I raised the run time to two minutes. Every now and then I get the required informations.

Having a look on the logs looks like that:

[image]

syssi, 2025-02-09:
Did the issue start after updating the ESP or out of the blue without touching the setup? Could you provide some more details what was changed recently?

Wiki591, 2025-02-09:
Before I didn't change anything. Data transmission was stable around two months and stopped last week as you call it "out of the blue". The last update I made was 2.1.2025, worked until last week flawlessly. In January I only changed the parameters for deep sleep.
Today I've freshly compiled with clean build, flashed on a brandnew ESP32 D1 Mini. Now I get the data but during the first startups I recognized the posted errors.

Wiki591, 2025-02-09:
```
substitutions:
  name: bms-murphy
  friendly_name: JK BMS 12V Murphy
  device_description: "Monitoring 12V JK-BMS BLE"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: AA:BB:CC:XX:YY:ZZ
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "Wiki.JK-BMS Murphy basierend auf github://syssi/esphome-jk-bms"
    version: 2.0.4
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

deep_sleep:
  run_duration: 120s
  sleep_duration: 
    minutes: 60

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid_joshua1
  password: !secret wifi_password_joshua1

ota:
  platform: esphome
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: INFO

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: ESP32_BLE_JKBMS
  birth_message:
    topic: binda/topic
    payload: online
  will_message:
    topic: binnichtda/topic
    payload: offline

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0 || x.get_name().rfind("JK_", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 10s
    id: bms24v

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "Balancieren"
    charging:
      name: "Laden"
    discharging:
      name: "Entladen"
    heating:
      name: "Heizen"
    online_status:
      name: "Online Status"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "Einstellungen empfangen"
    retrieve_device_info:
      name: "Infos empfangen"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms24v
    balance_trigger_voltage:
      name: "Start Balancieren"
    cell_count:
      name: "Zellenzahl"
    total_battery_capacity:
      name: "Gesamtkapazität"
    cell_voltage_overvoltage_protection:
      name: "Überspannungsschutz Zellen"
    cell_voltage_overvoltage_recovery:
      name: "Überspannungsschutz Zellen Recovery"
    cell_voltage_undervoltage_protection:
      name: "Unterspannungsschutz Zellen"
    cell_voltage_undervoltage_recovery:
      name: "Unterspannungsschutz Zellen Recovery"
    balance_starting_voltage:
      name: "Startspannung Balancieren"
    voltage_calibration:
      name: "Spannung kalibrieren"
    current_calibration:
      name: "Strom kalibrieren"
    power_off_voltage:
      name: "Abschaltspannung"
    max_balance_current:
      name: "Balancieren max Strom"
    max_charge_current:
      name: "Max Ladestrom"
    max_discharge_current:
      name: "Max Entladestrom"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms24v
    min_cell_voltage:
      name: "Min Zellspannung"
    max_cell_voltage:
      name: "Max Zellspannung"
    min_voltage_cell:
      name: "Min Spannung Zelle"
    max_voltage_cell:
      name: "Max Spannung Zelle"
    delta_cell_voltage:
      name: "Differenzspannung"
    average_cell_voltage:
      name: "Durchschnitt Zellspannung"
    cell_voltage_1:
      name: "Spannung Zelle 1"
    cell_voltage_2:
      name: "Spannung Zelle 2"
    cell_voltage_3:
      name: "Spannung Zelle 3"
    cell_voltage_4:
      name: "Spannung Zelle 4"
    total_voltage:
      name: "Gesamtspannung"
    current:
      name: "Strom"
    power:
      name: "Leistung"
    charging_power:
      name: "Ladeleistung"
    discharging_power:
      name: "Entladeleistung"
    temperature_sensor_1:
      name: "Temperatur 1"
    temperature_sensor_2:
      name: "Temperatur 2"
    power_tube_temperature:
      name: "Temperatur MOSFET"
    state_of_charge:
      name: "SOC"
    capacity_remaining:
      name: "Verbleibende Kapazität"
    total_battery_capacity_setting:
      name: "Gesamtkapazität"
    charging_cycles:
      name: "Zyklenzahl"
    total_charging_cycle_capacity:
      name: "Ges Zyklenkapazität"
    total_runtime:
      name: "Laufzeit"
    balancing_current:
      name: "Balance Strom"
    errors_bitmask:
      name: "Fehler Bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "Laden"
    discharging:
      name: "Entladen"
    balancer:
      name: "Balancer"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "Bluetooth Abfrage"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "Fehler"
    total_runtime_formatted:
      name: "Gesamtlaufzeit"

```

syssi, 2025-02-09:
How often do you see this error?

Wiki591, 2025-02-09:
Hhmm, hard to say.
The device is up once an hour for two minutes, throttle 10s. I would say, every 30s it appears, but not in every session. but it takes 5-6 attempts to connect the bms.
If you are interested, I could setup a logging over a while using the mqtt messages.

Wiki591, 2025-02-09:
Distance 3m to bms, bms located in a tin case (EEL 12V case)

syssi, 2025-02-09:
Please try to collect as most details as possible.

Wiki591, 2025-02-09:
Hhmmm, loglevel is set to "INFO". If needed, tomorrow I could change it to "DEBUG" or whatever you desire. Today its too late for me, the device in the RV is not OTA reachable due to deep sleep. And at the moment, I'm too lazy to step outside...

syssi, 2025-02-10:
@abc2006 Sorry for the late response and thanks for your issue with all the important details. Unfortunately it doesn't show any critical things.

> [W][component:237]: Component esp32_ble took a long time for an operation (965 ms).
> [W][component:238]: Components should block for at most 30 ms.

It's true: This implementation does block the loop but it shouldn't harm. As you have already found out: The load can be reduced by removing unneeded entities. Also reducing the log level will help here.

Nevertheless the warning can be ignored in general. Every user does see this warning.

> [14:24:59][W][esphome.ota:129]: Magic bytes do not match! 0x47-0x45-0x54-0x20-0x2F
> [14:24:59][E][component:164]: Component esphome.ota set Error flag: unspecified

I've never seen this issue before. It looks like an ESPHome core issue: https://github.com/esphome/issues/issues/6221

Unfortunately I cannot reproduce the issue. Do you see the issue on every OTA update?

Could you provide the (serial) log output of the time period where no data can be passed to the MQTT broker?

abc2006, 2025-02-11:
Hi, 
thanky for your answer. i flashed the devices again and until now i have not discovered any problems. 
However i have found out that the Charging/Discharging-Values (ON/OFF) are not updated in MQTT accordingly - the BMS changed to DISCHARGE OFF, but i found out only after connecting with my bluetooth Handy app (and killing Power of the esp, of course, to be able to connect via bluetooth). 

I can keep this in mind - can you tell me how exactly i can capture the serial log output? 
i just know the "esphome logs" command ... 

Thank you so far!

closed 2025-02-11 by syssi

Wiki591, 2025-02-15:
> (and killing Power of the esp, of course, to be able to connect via bluetooth).
> 

There is no need to power off the esp. In HA the device has a button named "enable bluetooth connection". Setting to "Off" stops the bluetooth connection of the esp to the bms and you are able to connect your smartphone. 

@syssi Having set the configuration to log level "DEBUG" (changed nothing else) and newly flashed: No errors anymore, receiving the data hourly without problems. 

??? mysterious, can't understand, maybe hickup of the bms firmware before ???