syssi/esphome-jk-bms #672: new "Running parameters" in V15.35 Upgrade

holle75, 2025-01-26:
Described by Jikong as:

4.Running parameters add the current charging status (Bulk, Absorption, Float), RCV Timer, RFV Timer display (mobile phone APP needs to be updated).

any chance to implement those readings/parameters?

whole upgrade log:

V15.35 Upgrade logs
1.Optimized ampere-hour integration algorithm to improve SOC accuracy.
2.Optimized SOC reset logic to 100%. After the absorption timer ends, if BATVOL > VOLSOC100%*CELLCOUNT , the SOC is set to 100%.
3.Added logic to enter float mode.
After the absorption timer ends, if BATVOL > VOLSOC100%*CELLCOUNT,
the BMS enters floating mode. Otherwise, the device enters bulk mode again.
4.Running parameters add the current charging status (Bulk, Absorption, Float), RCV Timer, RFV Timer display (mobile phone APP needs to be updated).

V15.36 Upgrade logs

    PBxx series parallel communication optimization.
    Add the status of LCD buzzer ,dry node 1 ,Dry node 2 on the APP(mobile phone APP needs to be updated).

V15.37 Upgrade logs

    Fixed bugs in the PACE RS485 communication protocol.

V15.38 Upgrade logs

    Fixed bugs in the PYLON communication protocol.

downset, 2025-02-15:
Hello, I provided some logs of the different new states in Patman15 linked above repository,
maybe it helps 

best regards
downset

testpaul999, 2025-03-28:
Hi, follow our diskussion here my first try give you log files.
the image shows "BULK"

[image]

this are the esp32 log for "BULK"

[logs_esp32-ble-jk-v14_logs.txt](https://github.com/user-attachments/files/19503054/logs_esp32-ble-jk-v14_logs.txt)

I'll try this for Absorbtion if the time is there.

regards

testpaul999, 2025-04-12:
Any news by implementier this new parameter?
Regards

syssi, 2025-04-12:
I wasn't able to identify the new details at the payload yet. More raw data including the details what's included exactly would speed things up / allows some comparison / better guesses.

testpaul999, 2025-04-12:
Hi, ok I’ll try.
But I have within the problem with the debug and very_verbose.
Deleting all sensors and switches dos not help the connection will not come up reliable.
Is there on other way with the esp32 to debug the ble connection?
Regards

Galusz, 2025-04-28:
Hi,

I added the following line:
 ESP_LOGI(TAG, "Hex_All  %s", format_hex_pretty(&data.front() + 249, data.size() - 249).c_str()); to JkBmsBle::decode_jk02_cell_info_

[11:54:02][I][jk_bms_ble:634]: Hex_All 
00.00.00.00.00.CF.00.C3.00.BF.00.BD.03.4F.7D.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.CC.00.02.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.15 (51) Float Time 207s??

[11:57:00][I][jk_bms_ble:634]: Hex_All  
00.00.00.00.00.CF.00.C3.00.BF.00.BD.03.02.7E.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.7F.01.02.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.3F (51) Float Time 383s??
[12:01:53][I][jk_bms_ble:634]: Hex_All  

00.00.00.00.00.CD.00.C3.00.BF.00.BD.03.28.7F.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.A5.02.02.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.14 (51) Float

[12:03:13][I][jk_bms_ble:634]: Hex_All
00.00.00.00.00.D1.00.C4.00.BF.00.BD.03.77.7F.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.00.00.00.02.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.3A (51) Bulk Time 0

12:06:58][I][jk_bms_ble:634]: Hex_All  
00.00.00.00.00.CE.00.C5.00.C0.00.BD.03.59.80.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.73.00.01.02.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.49 (51) Absorption time 117?

[12:11:14][I][jk_bms_ble:634]: Hex_All  
00.00.00.00.00.CF.00.C5.00.BF.00.BD.03.59.81.03.0A.1D.00.00.00.80.51.01.00.00.00.01.00.0A.00.02.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.CF.07.00.00.D3 (51) Float Time 10s?

As you can see, the data matches, as much as I was able to accurately capture the time.

syssi, 2025-05-26:
@Galusz Good job! I've picked up your work and will finish the feature soon. Thanks for your support!

syssi, 2025-05-27:
I've merged the new feature into main. Please give it a try by extending your YAML configuration:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v15-example.yaml#L321-L324
https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v15-example.yaml#L362-L363