syssi/esphome-jk-bms #676: Use Led on ESP01 (esp8266) module to inform about connection status.

agaletski, 2025-01-27:
I have successfully implemented my JK-BMS monitoring and happy to thank project developers for good work.
Also I have a question about LED existing on ESP01 module which seems remain unused. I guess it would be nice to use this LED for inform about system status  WiFi or HA connection or fashing when data transfer. Possibly this option isn’t related to current project and can be implementad independently in YML config. Unfortunatelly I’m rather newbe in the ESPHOME software configuration and don’t have enough knowledge to implement status reporting by LED. Possybly some one could point me some examples?

syssi, 2025-01-27:
Take a look at the `led_status` component: https://esphome.io/components/status_led.html

meccip, 2025-02-04:
I use this for my d1 mini in every cell reading platform. you will have a flash on every reading. try to adapt for esp01, but keep in mind that esp01 have low memory and could run into trouble if is to much code.

`on_value:
      then:
        - switch.toggle: led`

and add into code:

`switch:
  - platform: gpio
    pin:
      number: D4    <<< change for esp01 led pin
      mode: output
    id: led`

example code: - is for jkpb modbus, just for reference

` # 0x1210   16    UINT16    2    R    CellVol8    mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "cell voltage 09"
    address: 0x1210
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 3
    id: cell09
    on_value: 
      then:
        - switch.toggle: led
    filters:
      - multiply: 0.001

  # 0x1212   18    UINT16    2    R    CellVol9    mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "cell voltage 10"
    address: 0x1212
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 3
    id: cell10
    on_value:
      then:
        - switch.toggle: led
    filters:
      - multiply: 0.001`

syssi, 2025-02-04:
Another but very similar solution:

```
sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
      on_data:
        then:
          - output.turn_on: rx_led
          - delay: 100ms
          - output.turn_off: rx_led

output:
  - platform: gpio
    pin: GPIO2
    id: rx_led
```